OpenCPN/OpenCPN #3611: Both toolbars should have window type ‘utility’

mgrouch, 2024-01-28:
Currently left top toolbar is of type ‘utility’
and right buttom one is of type ‘normal’

that inconsistency create issues with stylings applied by window managers on Linux

bdbcat, 2024-01-29:
And where do these definitions (utility, normal, dialog, etc.) come from?  They are no part of wxWidgets, I think.

mgrouch, 2024-01-29:
xprop is the command utility which shows those properties.

Thanks

bdbcat, 2024-01-29:
Gonna need more info.  I guess this is a GTK3 issue?
Link, please?

mgrouch, 2024-01-29:
In Linux system you can run

xprop

then click on one toolbar to get report about its window 

then run it again and click on another toolbar

you will see that left one has type ‘utility’
and the other one has type ‘normal’

So there is some difference in the way (might be flags) how they are created in the code of OpenCPN

nohal, 2024-01-29:
What exactly is the problem resulting from this? I don't see any on any of my Linux systems on these windows that should not be styled by the WM in any way
 OpenCPN does not set any of the low level X properties on it's windows and while wxWidgets does some some of it internally, there is no public API to do so. So excuse my usual ignorance, but before we start to try to figure out what it is, show us what we are fixing here.

opaezosuna, 2024-02-14:
I do not know if this is somewhat related, but when using a touchscreen, the behavior of the two toolbars are different: the bottom functions "normally" (as it should) the one on to the left (the main) ignores the tapping (simulated mouse click). While hovering the mouse pointer over an icon of the left toolbar using the touchscreen, I can see the related labels being displayed, but the clicks do not work at all.

opaezosuna, 2024-02-14:
I just built a version on Fedora with the new style for the main toolbar:
`long wstyle = wxNO_BORDER | wxFRAME_NO_TASKBAR | wxFRAME_SHAPED | wxFRAME_FLOAT_ON_PARENT`

I have not noticed any difference in behavior between the two. I will test on the boat's pi once I install all the developer tools in it. Are there any provisions in the project to prepare a version for the pi4 from my desktop?

nohal, 2024-02-14:
Thanks for looking into it

I do this

```sh
podman run --arch arm64 -v /home/nohal/source/OpenCPN:/src:Z -it debian:bookworm-slim sh
cd /src
apt update
apt install devscripts equivs
mk-build-deps -i -r ci/control
apt-get --allow-unauthenticated install -f
mkdir build
cd build
cmake -D OCPN_BUILD_TEST=OFF ..
make -j 16
make package
```

The build in the container takes about half an hour (about 40x slower than the native x86_64 build) on my Ryzen 9 machine, but it is impossible to build on the Pi3 at all and I suppose even on the newer models it would take longer.

mgrouch, 2024-02-14:
> Thanks for looking into it
> 
> I do this
> 
> ```shell
> podman run --arch arm64 -v /home/nohal/source/OpenCPN:/src:Z -it debian:bookworm-slim sh
> cd /src
> apt update
> apt install devscripts equivs
> mk-build-deps -i -r ci/control
> apt-get --allow-unauthenticated install -f
> mkdir build
> cd build
> cmake -D OCPN_BUILD_TEST=OFF ..
> make -j 16
> make package
> ```
> 
> The build in the container takes about half an hour (about 40x slower than the native x86_64 build) on my Ryzen 9 machine, but it is impossible to build on the Pi3 at all and I suppose even on the newer models it would take longer.

Build on arm natively in CircleCi. It's super fast

leamas, 2024-02-14:
Sidenote: I  use an Orange Pi 5 Plus,  16G , ssd disk. Total build time from scratch is about 10 minutes which I find acceptable

opaezosuna, 2024-02-14:
Thanks for all the suggestions for the Pi build!

opaezosuna, 2024-02-19:
I was able to test the new build (only changed the window style) on the boat's Pi, the tool bar is still not responding to the touch clicks, the bottom one works correctly.

Any ideas on where to look for this? Where is the mouse events callback method?

opaezosuna, 2024-02-20:
Ok, the issue I had with the touchscreen and the main toolbar was related to the g_btouch global boolean. It is required to be set to true for the touch clicks to work. Setting it to true makes the clicking in the toolbar buttons work, internally it is set via the pMobile member of the options class ("Enable Touchscreen interface" checkbox), of course I had it unchecked, darn.

leamas, 2024-02-20:
Perhaps it's time to close this. The original issue is basically NOTABUG (see [Pavel's comment](https://github.com/OpenCPN/OpenCPN/issues/3611#issuecomment-1914485833)) and the same applies the issue brought forward by @opaezosuna

Will close unless there is more input.

leamas, 2024-02-22:
No more input. Closing.

closed 2024-02-22 by leamas