OpenCPN/OpenCPN #3613: RFE: Dashboard  Humidity%  (Relative Humidity) Instrument

rgleason, 2024-01-29:
**Is your feature request related to a problem? Please describe.**
Implement an Relative Humidity instrument Dial and Value.using Humidity% and "P" pressure

**Describe the solution you'd like**
add this to the nmea0183 XDR definitions and make a Dashboard instrument

**Describe alternatives you've considered**
Watchdog has this instrument partially done, but it also needs to be made functional.
We should drop Absolute Humidity which can be derived from AirTemp and RH.

**Additional context**
See this Watchdog Issue  https://github.com/rgleason/watchdog_pi/issues/103

Hakansv, 2024-02-01:
A personal, not specifically a OCPN common, view. Let's see if I can express it understandable:
I'm not sure humidity would fit in the Dashboard service.
Historical and maybe a future philosophy for the inbound Dashboard is to be an aid for navigation. That's to show data on instruments related to navigation and to maneuver the ship.
Dashboard would not be used for engine(s) supervision, weather forecasts/reports or similar functions. For that I think external plugins are more suitable and in some cases already present.

(OCPN's handle and possible extension of the XDR sentence to serve external plugins is another topic?)

closed 2024-02-01 by rgleason

rgleason, 2024-02-01:
Hakan re "aid for navigation. That's to show data on instruments related to navigation and to maneuver the ship."

Then why does Dashboard have:
Air Temp
Altitude
Altitude Trace
Barometric Pressure

Those are environmental and so is Relative Humidity.

Hakansv, 2024-02-01:
Good question. Historical reasons I'm not aware of.
And, my view is not the final answer. Here are more people to set the philosophy standard.
Again my view; Air pressure fall may have an impact to a possible gybe coming night. But Humidity?

rgleason, 2024-02-01:
Weather fronts

Hakansv, 2024-02-06:
A quick closure wo further discussion?
"Weather fronts" - why not.
And since inbound plugin grib can handle humidity and this discussion #3447:
I could try to implement a Dashboard single instrument. Data from XDR, MDA, SignalK and PGN130313.
Thoughts?