OpenCPN/OpenCPN #3614: Crash with Documentation (associated with navobj)

rgleason, 2024-01-29:
**Describe the bug**
@pa2wlt  Werner Toonk reports 
crash upon startup, that I could solve by removing navobj.xml.changes. I reckon the particular xml can be handy for developers, to figure out why it crashed. I also have the crash logs of course. I can share it all in some place,

**To Reproduce**
Steps to reproduce the behavior:
202401291130 OpenCPN immediately crashes after start up, also in SafeMode.

All was working fine.
I am not sure if the following is related, though an hour or so before the first crash I:
1. Imported a GPX file with quite some Patagonian Waypoints, deleted them.
2. Created a a layer of this GPX file (so that I am able to hide them all in one go, since all these waypoint labels, make it hard to see the chart).
3. Changed some settings in the ux control (font-sizes)

After the crahs, I tried the following to resolve:
1. Just restart OpenCPN (in SafeMode as well)
2 In opencpn.ini changed OpenGL=1 to OpenGL=0 and add line DisableOpenGL=1 bellow, save the file and start OpenCPN as usually.
3 Rebooted the Windows 10 computer.
4 Renamed the charts folders for oeCL and oeAR
5 Renamed complete Charts folder
6 Renamed opencpn.ini
7 Removed the layer
8 Just waited 15 minutes

Though none of that worked and the behaviour remained unchanged. (crashing upon startup)

Then I:
1. Moved navobj.xml navobj.xml.1., - .5 and .changes from program data and OpenCPN did not crash  upon startup.
2. Copied navobj.xml ( incl l. - 5) back to program data:  all fine.
3. Copied navobj.xml.changes back to program data, then OpenCPN crashes upon startup.

So, there seems to be an issue with this particular navobj.xml.changes, that makes OpenCPN crash on start up.

Not sure what that is. I kept the navobj.xml.changes in a separated folder as navobj.xml.changes.202401291130

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: [e.g. macOS]
 - Version [e.g. 22]

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: [e.g. Galaxy S20]
 - Andorid version: [e.g. 10]

**Additional context**
Werner will try to upload the files. here. He is sailing in Patogonia and is somewhat occupied navigating, so it may take awhile.

pa2wlt, 2024-01-29:
On Windows 10. Open CPN 5.8.4.0
[navobj.xml.changes.202401291130.zip](https://github.com/OpenCPN/OpenCPN/files/14088391/navobj.xml.changes.202401291130.zip)
[15130b7c-f6ff-46b1-9d66-49fb9453270e.7z.zip](https://github.com/OpenCPN/OpenCPN/files/14088397/15130b7c-f6ff-46b1-9d66-49fb9453270e.7z.zip)

nohal, 2024-01-30:
I can reproduce the crash, will look into it.

transmitterdan, 2024-01-30:
I am no expert on the changes scheme, but the navobj.xml.changes file in question appears to try to first update a route then it tries to delete the same route multiple times.  The second delete causes the crash.

transmitterdan, 2024-01-30:
There are 2 of these double-deletes in the changes file.  If I delete the redundant ones then it doesn't "crash" as before. But I get a runtime error in the debug build with "heap corruption". So, there is more to the story than the double-delete issue.

nohal, 2024-01-30:
Yes, there are multiple problems here that need to be fixed.

nohal, 2024-01-30:
I have fixed everything I was able to find in https://github.com/OpenCPN/OpenCPN/pull/3585
Would appreciate some more testing from others

transmitterdan, 2024-01-31:
It doesn't any more crash for me. Nothing was imported from the "bad" changes xml file.

nohal, 2024-01-31:
That is exactly what is expected, that file does not contain any additions and you do not have the base data. I am much more interested in testing other scenarios you can think of with actual data you have. As you can see from the PR, the number of things fixed that are not directly related with the crash, but immediately popped up is pretty high. Higher than I would like.

closed 2024-02-25 by nohal