OpenCPN/OpenCPN #3615: OpenCPN requires root priviledge to install despite "CMAKE_INSTALL_PREFIX" due to udev rules

Ryanf55, 2024-01-29:
**Describe the bug**

I would like to compile, install, and run OpenCPN without root permissions.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the steps to compile on linux: https://opencpn-manuals.github.io/main/opencpn-dev/linux.html
2.  Set the install prefix to a local directory `cmake -DCMAKE_INSTALL_PREFIX=../install  ..`
3. See error

**Expected behavior**
All files will be installed in `../install`

**Actual behavior**
The installation attempts to write udev rules to `/lib/udev/rules.d/98-sglock-ocpn.rules`

**Screenshots**

```
friedman@AutonomyML01:~/Development/OpenCPN$ cmake --install build --prefix install
...
-- Installing: /lib/udev/rules.d/98-sglock-ocpn.rules
CMake Error at build/cmake_install.cmake:367 (file):
  file INSTALL cannot copy file
  "/home/friedman/Development/OpenCPN/buildlinux/98-sglock-ocpn.rules" to
  "/lib/udev/rules.d/98-sglock-ocpn.rules": Permission denied.
  ```

**Desktop (please complete the following information if applicable):**
 - Ubuntu Linux
 - Version 22.04

**Additional context**
* Seems like a regression in https://github.com/OpenCPN/OpenCPN/issues/1373

leamas, 2024-01-29:
Well, bug, I don't know. Anyway, I assume this untested patch should fix it.
```
iff --git a/CMakeLists.txt b/CMakeLists.txt
index 876b4c12d..24c87ac43 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -1979,11 +1979,20 @@ if (NOT APPLE)
   endif ()
 
   if (PACKAGE_FORMAT MATCHES "DEB")
-    install(
-      FILES "buildlinux/98-sglock-ocpn.rules"
-      PERMISSIONS OWNER_READ OWNER_WRITE GROUP_READ WORLD_READ
-      DESTINATION "/lib/udev/rules.d"
-    )
+    if (${CMAKE_VERSION} VERSION_GREATER_EQUAL 3.21)
+      # Best effort to write into /lib/udev, errors silently ignored.
+      file(
+        COPY_FILE "buildlinux/98-sglock-ocpn.rules"
+                  "/lib/udev/rules.d/98-sglock-ocpn.rules"
+        RESULT discard
+      )
+    else ()
+      install(
+        FILES "buildlinux/98-sglock-ocpn.rules"
+        PERMISSIONS OWNER_READ OWNER_WRITE GROUP_READ WORLD_READ
+        DESTINATION "/lib/udev/rules.d"
+      )
+    endif ()

```

leamas, 2024-02-02:
Closed by 92cfacf

closed 2024-02-02 by leamas