OpenCPN/OpenCPN #3616: signalK net driver : Crash when restarting network connection

leamas, 2024-01-30:
**Describe the bug**
When using using the signalK network driver system crashes when a network connection fails and later is restored.

**To Reproduce**
Steps to reproduce the behavior:
1. Start opencpn
2. Configure with one single  enabled driver using signalK protocol from signalk.stupan.se:3000
3. In a separate CLI window run `nmcli connection show`  which lists the network connections. Note the one currently giving access not the outside world i. e., the one reading from stupan.se. For me, this is `Wired connection 1`,  YMMV.
4. With openpn running cut the network connection using `sudo nmcli conn down "Wired connection 1"`. This simulates a failing netwotk connection.
5. Wait for at least 1 minute
6. Restore connection using `nmcli conn up  "Wired connection 1"`
7. Wait some time until ship gets valid data from the signalK connection (15 seconds?)
8. Wait a few seconds
9. See crash

**Expected behaviour**
Well...

**Desktop (please complete the following information if applicable):**
 - Fedora 38, native build

**Additional context**
backtrack: 
```
#0  0x00007ffff50b0884 in __pthread_kill_implementation () at /lib64/libc.so.6
#1  0x00007ffff505fafe in raise () at /lib64/libc.so.6
#2  0x00007ffff504887f in abort () at /lib64/libc.so.6
#3  0x00007ffff504879b in _nl_load_domain.cold () at /lib64/libc.so.6
#4  0x00007ffff5058187 in  () at /lib64/libc.so.6
#5  0x0000000000c47280 in rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::MemoryPoolAllocator<rapidjson::CrtAllocator> >::MemberEnd() const
    (this=0x7fffffffd080) at /usr/include/rapidjson/document.h:1078
#6  0x0000000000c45158 in rapidjson::GenericValue<rapidjson::UTF8<char>, rapidjson::MemoryPoolAllocator<rapidjson::CrtAllocator> >::HasMember(char const*) const (this=0x7fffffffd080, name=0xfd79b4 "version") at /usr/include/rapidjson/document.h:1094
#7  0x0000000000cc0c03 in CommDriverSignalKNet::handle_SK_sentence(CommDriverSignalKNetEvent&) (this=0x209e130, event=...)
    at /home/mk/src/OpenCPN/model/src/comm_drv_signalk_net.cpp:368
#8  0x00007ffff6dc09e2 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
    at /lib64/libwx_baseu-3.2.so.0
#9  0x00007ffff6dc2dad in wxEvtHandler::SearchDynamicEventTable(wxEvent&) () at /lib64/libwx_baseu-3.2.so.0
#10 0x00007ffff6dc310d in wxEvtHandler::TryHereOnly(wxEvent&) () at /lib64/libwx_baseu-3.2.so.0
#11 0x00007ffff6dc31bf in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /lib64/libwx_baseu-3.2.so.0
#12 0x00007ffff6dc32d2 in wxEvtHandler::ProcessEvent(wxEvent&) () at /lib64/libwx_baseu-3.2.so.0
#13 0x00007ffff6dc3ed5 in wxEvtHandler::ProcessPendingEvents() () at /lib64/libwx_baseu-3.2.so.0
#14 0x00007ffff6cb0a4a in wxAppConsoleBase::ProcessPendingEvents() [clone .part.0] () at /lib64/libwx_baseu-3.2.so.0
#15 0x00007ffff7372aea in wxApp::DoIdle() () at /lib64/libwx_gtk3u_core-3.2.so.0
#16 0x00007ffff7372bc7 in wxapp_idle_callback () at /lib64/libwx_gtk3u_core-3.2.so.0
#17 0x00007ffff5ead4fd in g_idle_dispatch () at /lib64/libglib-2.0.so.0
#18 0x00007ffff5eb14fc in g_main_context_dispatch () at /lib64/libglib-2.0.so.0
--Type <RET> for more, q to quit, c to continue without paging--
#19 0x00007ffff5f0f6b8 in g_main_context_iterate.isra () at /lib64/libglib-2.0.so.0
#20 0x00007ffff5eb0aff in g_main_loop_run () at /lib64/libglib-2.0.so.0
#21 0x00007ffff6606975 in gtk_main () at ../gtk/gtkmain.c:1329
#22 0x00007ffff7385ac5 in wxGUIEventLoop::DoRun() () at /lib64/libwx_gtk3u_core-3.2.so.0
#23 0x00007ffff6cd709a in wxEventLoopBase::Run() () at /lib64/libwx_baseu-3.2.so.0
#24 0x00007ffff6cb2bcd in wxAppConsoleBase::MainLoop() () at /lib64/libwx_baseu-3.2.so.0
#25 0x00000000009aa8f7 in MyApp::OnRun() (this=0x1375ad0) at /home/mk/src/OpenCPN/gui/src/ocpn_app.cpp:922
#26 0x00007ffff6d132c0 in wxEntry(int&, wchar_t**) () at /lib64/libwx_baseu-3.2.so.0
#27 0x00000000009a82c2 in main(int, char**) (argc=1, argv=0x7fffffffda08) at /home/mk/src/OpenCPN/gui/src/ocpn_app.cpp:725

```

leamas, 2024-01-30:
Native build,   current master at 5334661c1

nohal, 2024-01-30:
Can you reproduce it with https://github.com/OpenCPN/OpenCPN/pull/3207 ?

leamas, 2024-01-30:
> Can you reproduce it with https://github.com/OpenCPN/OpenCPN/pull/3207 ?

No.

nohal, 2024-01-30:
Ok, then I think it is time to merge it, should generally be better than the current master

Hakansv, 2024-01-30:
On windows:
I've now tested with #3207.
Connection to SK works as before.
If I close the network, wifi, and start it again after some minutes the SK connection is reestablished. Good.
But:
If I restart the SK server, on a RPi, the OCPN SK connection is not reestablished when SK server is up and running again.
If I set Win to suspend and up again after some minutes the SK connection is not reestablished.
The only method to reestablish for the two last cases is to disable connection and enable it again. Or of course restart O.

leamas, 2024-01-30:
Closed by #3207. Fast, indeed.

closed 2024-01-30 by leamas