OpenCPN/OpenCPN #3625: Boost dependency makes OpenCPN again unusable on legacy macOS

nohal, 2024-02-05:
**Describe the bug**

https://www.cruisersforum.com/forums/f134/list-of-macos-versions-and-compatible-versions-of-opencpn-283354.html#post3867439

We will probably have to add boost to the dependency bundle for the legacy build as the Homebrew version that is now being pulled in does not seem to work. Or do some other voodoo once again.
I now officially hate this.

bdbcat, 2024-02-05:
Where does the boost dependency originate?
I officially hate Boost.

nohal, 2024-02-05:
From `std::filesystem` not available with old clang on old macOS. I like C++17 and newer, we do want it, so it's probably better to somehow handle this until all the old machines finally die...

nohal, 2024-02-05:
There seem to be some ways to solve this without actually pulling in `boost::filesystem` at all. Maybe the single header from https://github.com/gulrak/filesystem will be the easiest, avoiding another rebuild of the dependency bundle. I will somehow take care of it...

PR #3627 (2024-02-05): Replace boost::filesysten dependency on legacy macOS with ghc::filesystem
Boost dependency causes the usual trouble at runtime on machines older than the builder. As we need only `filesystem` and for that we have a trivial alternative in `ghc::filesystem`, it is not worth the trouble to build Boost and include it in the dependency bundle.

Closes #3625.
closed 2024-02-05 by nohal