OpenCPN/OpenCPN #3634: AIS target track recording still broken

nohal, 2024-02-07:
**Describe the bug**

If the "Show AIS target tracks" option in the Toolbox is not enabled, tracks for targets are not persisted even when explicitly requested by the button in the target query and confirming it should stay on.

The recording of AIS target track, once activated in the target properties dialog, should simply persist the track until it is explicitly disabled, regardless of anything else.

bdbcat, 2024-02-08:
Turning this around a bit:
As I have tested, the track is always persisted or stopped, according to the toggle button in the Query dialog.
However, it is not displayed unless the  "Show AIS target tracks" option in the Toolbox is enabled.
I suggest that we force display of a "persistently tracked" AIS target, regardless of the global "Show AIS target tracks" option.
Possible GUI enhancement might be to change the default color of persistently tracked targets, as well.
Thoughts?

bdbcat, 2024-02-08:
Ooops...
Belay my last.
There is indeed some logic problem here.  As you say, there is some interference with the global "ShowAIS tracks" button.
Digging in.

nohal, 2024-02-08:
I did some testing and it actually seems more broken than that. Making it visually apparent is sure a good idea. There are also the Show all/Hide all settings in the target list and "buddy" settings involved.

bdbcat, 2024-02-08:
And one more:
global "Show..." off.  Query->persist toggle on.  A new Track appears in RouteManagerDialog, length 0 NMi.
Show Properties on that track--> crash.

bdbcat, 2024-02-08:
P.S.
Bothers me a little that we consider an AIS Target track to be nearly the same thing as an ownship track, at least in the RouteMenager Dialog. Should we think of another Tab in RMD, for persisted AIS tracks?  Or is this too much muchness..
Hmmmm.

nohal, 2024-02-08:
As long as we only persist and show in the RM the ones the user explicitly wants that is IMHO fine. Introducing another type of navobject would just make what people in https://www.cruisersforum.com/forums/f121/best-chart-navigation-software-283446.html speak worse.
What I think is a good idea is to make the live AIS target tracks distinct from persistent tracks on the chart canvas.
And when at it, the "highlight track" feature is confusing people a lot, but that probably deserves own issue.

bdbcat, 2024-02-08:
Fair enough.
Lets re-think "highlite" as a separate issue.
Anyway, I got this one, I'll see what I can come up with.

bdbcat, 2024-02-08:
A first cut at a more rational GUI.
Nothing done on AIS target list, or Buddy Boat tracking, yet.

Hakansv, 2024-02-08:
The new blue color is fine.
I agree about further discussion on the highlite. User frustration received also here.

bdbcat, 2024-02-08:
Allow user-specified persistent track color fixed.
I think I am done here.  Close when ready.

Hakansv, 2024-02-08:
Confirmed. Persistent track color can now be user defined.
Thanks

nohal, 2024-02-11:
I still think there are some corner cases that are not quite right, mainly the behavior of persistent AIS target tracks while general AIS target track visibility is changed, will take care of it and close then.

nohal, 2024-02-13:
Did some testing, a couple of things still need our attention...
- Pretty much very trackpoint is recorded twice for AIS targets when track recording is enabled from the target query dialog when using the stupan.se SignalK data feed
- The recorded track of AIS target is not selectable in it's whole length, only the part that already existed in memory when persistent tracking was turned on
- Similar to the above, last segment of the ownship track between the last persisted point and the boat is not selectable (very well testable with stupan.se feed where the boat goes straight for long stretches)
- AIS target with persistent track enabled -> Hide track from the context menu -> Already persisted track changes thickness, new track segments are not recorded or drawn until track is again shown using the context menu. Then the persistent track continues, connecting the last point before it was hidden and the current position (For persistently tracked targets, I would probably completely remove this option from the menu, what to expect is too confusing and it will be consistent with the target list where showing/hiding all skips the ones with persistent tracks)

bdbcat, 2024-03-02:
"every trackpoint is recorded twice "
Cannot reproduce with current master.  Hints?

bdbcat, 2024-03-02:
(2)  "track of AIS target is not selectable"
(4)  "completely remove this option from the menu"

Completed by commits referenced above.

(3)  Does not refer to AIS tracking, by core ownship tracking?
(1)  Sometime reproduced.  May be due to signalK update logic, as read from a VDR recording at the server"?

Hakansv, 2024-03-02:
With yesterday's patches:
Now the persistent and normal track are able to select together as one. Good.
Side effect: When using right click canvas menu and delete such a track also the normal track is disabled for that target. It's though possible to enable it again from the canvas menu.

re: "_as read from a VDR recording at the server_" 
Just to clarify: The own ship movement and some sensor's data are from a VDR alike reply but all AIS targets are "real" from a up to date TCP stream.

bdbcat, 2024-03-05:
With latest "duplicate filter" patch, I think we are done here.
Close?

bdbcat, 2024-03-27:
Close, please?

closed 2024-03-27 by nohal