OpenCPN/OpenCPN #3638: Multiple 0 bytes files after installing big o-charts packages in Linux.

sailoog, 2024-02-08:
This issue has been reproduced in Raspberry OS Bookworm in Raspberry Pi 4-5 and also vanilla Debian Bookworm installations in laptop/desktop running OpenCPN no matters the source (Debian repos, PPA or Flatpak).

When the o-charts package is big (UK, Norway...), lots of files have 0 bytes after the plugin unzip them. The file in DownloadCache is correct and can be unzipped manually without errors. There are also no errors when the installation is done offline. These failing cells can be identified validating the installation because all of them will result in error 10.

There is a workaround to fix this editing /etc/systemd/logind.conf and replacing:

#RuntimeDirectorySize=10% 

by

RuntimeDirectorySize=2000M 

After rebooting the run time directory looks like this and the issue is fixed:

tmpfs           2,0G  304K  2,0G   1% /run/user/1000

Some questions...

Why does no this happen when the files are unzipped using the file manager? Is the runtime directory not used in these cases? Could the plugin use another directory instead of the runtime?

bdbcat, 2024-02-08:
Roberto...
The issue here is that o-charts unzips the files to a "tmp" file system, which is RAM resident for performance.  The size of the "tmp" FS is set by "RuntimeDirectorySize", a percentage of available system RAM.
This is probably not a great idea on memory limited systems, like RPi.
I'll take a look at the code, see if I can refactor the "relocating charts" logic step.  Seems maybe unnecessary, after all.  Why not unzip to the proper final location, and be done with it once?

Thanks
Dave

sailoog, 2024-02-09:
Ops I did not know that it was RAM. I will check the amount of RAM of the systems that fail.

There is an observed pattern. All the failing systems are SSD, I have been able to reproduce the issue in an SD installation some time ago but after some systems updates I can not. The SSD installations keep failing. Just in case it helps.

Thanks

bdbcat, 2024-02-09:
I think the simplest, most reliable solution will be to use real disk file for unzip, instead of tmp (RAM-based) unzip location.
This will remove RAM stress, while at the same time verifying that the actual disk space is sufficient to contain the entire charts set.  We can arrange it so that the original unzipped files are deleted after relocation, so there should never be a need for more than 1X of the full chart set on disk.
Your thoughts?

sailoog, 2024-02-09:
Of course that sounds the better way.

Do not take my last contribution too seriously (SD/SSD) because now I think it may depend on the RAM usage at that time...

bdbcat, 2024-02-09:
Did a little reading.
Seems that some linux systems implement /tmp on a physical SSD, if present.  Others use pure RAM, virtualized to physical disk as necessary.  Rather complicated, and not easy to figure out exactly what any particular system is doing.
Lets try the "physical" solution describe above.
Do you have a reliably failing test system?

sailoog, 2024-02-09:
No here, but we have several users at o-charts using the RuntimeDirectorySize=2000M fix and I could ask them.

sailoog, 2024-02-09:
I have already a user with Raspberry Pi 5 and Raspberry OS Bookworm affected and willing to help.

leamas, 2025-07-18:
This should be fixed in 5.12 and related plugin updates.

@sailoog : Could you please test this in 5.12 so we can --------------------------+