OpenCPN/OpenCPN #3639: Scrolling the Options User Interface window screws up the sliders on MacOS

antipole2, 2024-02-08:
**Describe the bug**
Any attempt to scroll the User Interface General Options panel screws up the sliders.
No way to recover use of the sliders outside quitting OCPN and starting again.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Options > User Interface Genetal Options
2. Lower part of window contains sliders for User Interface scale factor - ENC Sounding factor etc.
3. In my case, not all the panel is visible so there is a scroll bar at the side
4. Scroll the window and the sliders narrow, whether scrolling up or down.
5. There is no way to make them usably wider because they narrow whether you scroll up or down.
6. The same effect if you scroll with two finders on the trackpad.

**Expected behavior**
Slider with should not change.

**Screenshots**
Screenshot below

**Desktop (please complete the following information if applicable):**
MacOS Sonoma 14.0
OCPN  5.9.0-0-db1a12d

[image]

nohal, 2024-02-08:
The patch from https://github.com/OpenCPN/OpenCPN/issues/3019

```diff
From aa6141fc1c6c27186c4a108f386b9f69149f894e Mon Sep 17 00:00:00 2001
From: dsr <bdbcat@yahoo.com>
Date: Wed, 1 Mar 2023 21:08:33 -0500
Subject: [PATCH] Correct slider distortion when used in sizer, MacOS

---
 src/osx/slider_osx.cpp | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/src/osx/slider_osx.cpp b/src/osx/slider_osx.cpp
index 08b4477a71..dd72c9f924 100644
--- a/src/osx/slider_osx.cpp
+++ b/src/osx/slider_osx.cpp
@@ -435,7 +435,7 @@ void wxSlider::DoSetSize(int x, int y, int w, int h, int sizeFlags)
 
         if (ht > textheight)
             textheight = ht;
-
+/*
         if (GetWindowStyle() & wxSL_HORIZONTAL)
         {
             if ( m_macMinimumStatic )
@@ -447,7 +447,7 @@ void wxSlider::DoSetSize(int x, int y, int w, int h, int sizeFlags)
             if ( m_macMaximumStatic )
                 w -= maxValWidth / 2;
         }
-
+*/
         // Labels have this control's parent as their parent
         // so if this control is not at 0,0 relative to the parent
         // the labels need to know the position of this control
-- 
2.17.1
```

nohal, 2024-02-08:
Assigning to myself as it needs to be applied to the wxWidgets libraries included in the macOS dependency bundles used for release packaging.

commit 85e5f543 (2024-02-08): Update the macOS dependency bundle generation script. Closes #3639
closed 2024-02-08 by nohal