OpenCPN/OpenCPN #3646: First use configuration wizard

nohal, 2024-02-09:
People often complain about OpenCPN being too difficult to start using for first time users. A basic configuration wizard allowing configuration of a data connection and a chart source was mentioned as a welcome improvement in https://www.cruisersforum.com/forums/f121/best-chart-navigation-software-283446-2.html#post3868928

rgleason, 2024-03-19:
**Goals:** 
1. Getting New Users Started easily and quickly without a manual.
2. Simplify chart installations.

**Describe the solution you'd like**
New Users who have an "empty chart folder" or non-existent chart folder, are prompted:
1. Accept the default location for Charts. Charts folder is created.
2. Browse to desired location to create a new "Charts" folder (provided the location has user permissions).  Once the parent folder is selected the "Charts" folder is created and entered into openpcn.ini.

New Users are then asked to select several ways to get charts.
1. Draw a rectangle for the area desired.
2. Select from the Chart Downloader catalog.
3. If 1 & 2 do not have charts available, download a "sample set of charts" from somewhere, show these charts in opencpn and explain that they are sample charts because there are no free ones available for that area, and then direct the new user to the ocharts website to purchase.

See:
https://www.cruisersforum.com/forums/f134/first-principles-for-a-new-user-284473.html#post3881341
and  Wholybee [wrote](https://www.cruisersforum.com/forums/f134/first-principles-for-a-new-user-284473.html#post3881735)

> Wizards.
> A wizard to install charts. When running the wizard, the chart directory etc. should be hidden from the user. Just create that directory someplace it makes sense for the particular OS. The wizard should at least be able to install NOAA chars  or O-Charts with zero knowledge of how charts work. The user should just draw a rectangle on the screen or choose from a list of locations .
> 

https://www.cruisersforum.com/forums/f134/first-principles-for-a-new-user-284473-4.html#post3880924

and Dave [wrote](https://www.cruisersforum.com/forums/f134/first-principles-for-a-new-user-284473.html#post3881334)

> So, let us get specific.   Regarding your suggestion: "3. Pre-build a base chart file structure. Give some automatic priority so they don't get [lost](https://www.cruisersforum.com/forums/tags/lost.html) if user loads too many chart files."
> 
> I'm serious, now. Would you be interested in proposing a specific chart file structure for new users, and moderating a discussion that may result from same? How exactly would that priority scheme work? Consider the platforms, or for simplicity, just start with Windows, some 65 % of our user base, and the most likely new user platform.

rgleason, 2024-03-19:
The [User Manual Basic shows 3 ways to install charts](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:chart_install#three_excellent_ways_to_install_charts)

Why is this so difficult for people?

rgleason, 2024-03-21:
[GPS Installation comments ](https://www.cruisersforum.com/forums/f121/best-chart-navigation-software-283446-5.html#post3871784)

> ..adding a [GPS](https://www.cruisersforum.com/forums/tags/gps.html) or any type of input device requires a lot more steps and technical proficiency than [commercial](https://www.cruisersforum.com/forums/tags/commercial.html) software. OCPN requires defining [ports](https://www.cruisersforum.com/forums/tags/ports.html) and some troubleshooting. It will not receive all N2K data off my gateway, for example. CE/TZ are much more [plug](https://www.cruisersforum.com/forums/tags/plug.html) and play.
> 
> Another example of OCPN having a feature that is not accessible to many users. It really could be great software if the developer community spent some time with competing products and normal users.

[Supplementary Hardware comments:](https://www.cruisersforum.com/forums/f121/best-chart-navigation-software-283446-5.html#post3871783)

`"It's both. Someone thought it relevant enough to create a document here:"
`
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:gps_setup

```
... but it hasn't been updated in years and is now more misleading than helpful (at least for MacOS).

Edited to add: Strictly speaking, you are correct. However I'm going to venture a guess that most people have not had an occasion to add a GPS receiver to their PC or Macbook prior to using a chartplotter app so getting a GPS receiver set up on their computer is very much a part of getting up and running on OpenCPN.
```

rgleason, 2024-04-01:
First Users  [  GNSS Setup  (User Manual Basic)](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:gps#windows_10_11)  **Please Reveiw.**  I have made improvements to set user expectations, but this is still too complex and must be simplified.

Is it possible to program a small wizard that would help New Users with settting up a GNSS device?

Windows Sensor Compatibility: https://github.com/OpenCPN/OpenCPN/issues/3775

nohal, 2024-04-01:
Please stop cross-posting absolutely everything absolutely everywhere, please.

rgleason, 2024-04-01:
I took that link out.

bdbcat, 2024-04-02:
Just my thoughts, NOT TO BE TAKEN AS _verbatim_ DIRECTION:

Is this intended as Getting Started instructions for neophytes?  If so:
1.  Way too long for one topic.  Intimidating!  Yes, please split into separate topics by OS.

On Windows specifically.  Two points:
1.  Windows Sensor architecture is not very commonly used by our users.  Definitely should not be the first in the list.  Might even want to be in another section entirely.

2. In Windows with GPS puck, there is usually no need to go thru Windows control panel to identify the port name.  The initial solution for most new users should be something like:
a.  Stop OCPN
b.  Plugin in the GPS puck.  Respond to any Windows messages about drivers, etc.
c.  Start OCPN
d.  Settings->Connections->Add connection.
e..  Select "serial-NMEA0183" connection, 4800 baud.  Look in the provided drop-down list (in the OCPN connections dialog) of detected serial ports.  If you recognize an entry for the device you just installed, select it.  And you are done.
f.  If not, proceed as documented using the Windows Control Panel.

3.  MacOS nearly the same.
4. linux will necessarily be more complicated.  That's the way linux is.

bdbcat, 2024-04-02:
More...
1.  Remove specific mention of Windows 7/8.  Ancient. User seeing that will skip over ("I have Win 10").  But they may need those instructions for 10 as well.
2.  Remove specific device model numbers ( e.g. BU354) from basic instructions.  These references quickly become stale.
3. Have a separate section for specific device quirks.
4. Do not recommend any specific devices in a general user manual.  Ever.  This belongs in a App note, or other document.

nohal, 2024-04-02:
This is more about the initial workflow.
1. Localization, set locale to system (Gets rid of the number one complaint in France = "It is in English")
2. Units. Gets rid of "World is metric you dements"
3. A few things that should be default anyway nowadays (Zoom to cursor, smooth zooming). I would actually just make them default.
4. For connections, we can detect a lot based on USB device IDs (eg. NGT-1 has a unique ID and known parameters, so if detected, whole N2K connection can be configured automatically). This gets harder or even impossible with NME0183 GPS pucks, especially the cheap ones, as they use the OEM or outright Chinese fake serial->USB converters. UDP checked for broadcasts on known ports, TCP can be tried to known common ports on default gateway.

nohal, 2024-04-02:
And of course, none of that is about the manual, it is a few steps of a wizard at startup if it is a new installation.

bdbcat, 2024-04-02:
Sorry, you are right.  This is not a manual critique thread.
On the wizard: And are you planning to build this wizard, or are we still at blue-sky stage?

leamas, 2024-04-02:
> linux will necessarily be more complicated. That's the way linux is.

IMHO, not really.  Possible device candidates are easy to detect, and things could basically work as for Windows and MacOS [above](https://github.com/OpenCPN/OpenCPN/issues/3646#issuecomment-2030893516).

The thing disturbing the picture is permissions, general users don't have permissions to serial devices by default. We can handle this the same way as for the chart dongle where we tell the users the exact command to run to fix this. It means copy-paste into a terminal window which is bad, but not that bad.

rgleason, 2024-04-02:
Sorry for muddying up the thread with User Manual stuff.
I have created an [User Manual Basic](https://github.com/OpenCPN/OpenCPN/discussions/3776) discussion here.
Please direct comments there about the manual. Also I could really use some help and review on certain sections.
In particular  Connections,  GNSS Setup  and the individual pages for Linux GNSS, Windows GNSS, MacOS GNSS, etc.

@bdbcat  I have tried to make most of the changes you suggested. I will review all the comment (much appreciated) later.
I will hide this post once we all get using the Discussion [User Manual Basic](https://github.com/OpenCPN/OpenCPN/discussions/3776)   Thanks.

nohal, 2024-04-03:
Here is a mockup of the wizard for discussion before I invest any serious coding and time to it:

[image]
[image]
[image]

After these the last step will simply contain a HTML window with a one-page overview of the GUI elements.

Beside this, I would change a few defaults:

- Menu bar to shown (people keep looking for it and the ones who do are usually unable to enable it in the Toolbox)
- Zoom to cursor to enabled (I have never met anybody who would prefer the zoom to center of the screen)
- Smooth panning/zooming to enabled (I do not remember ever using a system where I would want it not to be enabled)

Also attaching the wxFormBuilder project if you want to play with other ideas.
[init_wiz.zip](https://github.com/OpenCPN/OpenCPN/files/14853074/init_wiz.zip)

bdbcat, 2024-04-03:
Very cool.  Three pages seems about right.  Enough info, but not too much for the impatient new user.
Consider this:  Optionally, if internet is available, how about getting a quick geo-locate, and automatically downloading a NOAA chart for the state, or region where the device appears to be located.

I agree about the defaults, and I will do that today.

Hakansv, 2024-04-03:
"Detected navigation data sources" Looks really nice. I'm eager to see how that will work.

> "I have never met anybody who would prefer the zoom to center" 

My view for the book. No criticism:  I have very seldom met anyone using "zoom to cursor" on board. Including myself.

bdbcat, 2024-04-03:
Hmmm...
After the edit, I'm confused.
Hakan does not prefer ZTC?
FWIW, I definitely prefer ZTC for daily use.

Hakansv, 2024-04-03:
> Really? While using a mouse? 

Yes especially on the rolling boat where i want to use the scroll wheel to zoom and have limited control of where the cursor is. Then it's better it's zoomed around the boat so I keep the follow function. But that's me. And I know where the control is so any default is irrelevant. For me.

nohal, 2024-04-03:
It makes sense only combined with locking the screen to boat position (which could be made to override zooming to cursor as it does override the zooming to the center of the screen). Without locking the screen to the boat, zoom to center is IMHO unusable.

bdbcat, 2024-04-03:
And that is why we have so many UI configuration and control buttons. :)
ZTC defaults ON for new user.

Hakansv, 2024-04-03:
Parts of my note above was accidentally deleted:
I do appreciate the attempt to automatically detect connection(s) with navigational data. 
I use to say Time-Zero is the only real OCPN competitor. If that feature is realized one function I use to honor Time-Zero for would be gone. :)

rgleason, 2024-04-04:
This would be a phenomenal improvement!  Bravo.
I like ZTC but can understand Hakan's point quite well, and my boat computer MS SurfacePro3 is set that way.

rgleason, 2024-04-09:
Wow, thank you Pavel!

Hakansv, 2024-04-09:
First time test on a Win-PC
To test I deleted the config file.
The new nice Dialog was shown at first O-start
- Units works well.
-  Connections is troubled. See below.
-  Charts dialog is working and O was started and the charts listed was installed. But--- The busy spinner is there and won't disappear. 

Connections:
He "found" as of below picture 1.
The SK server 192.168.1.167 has no data and discussed elsewhere.
He didn't found any of the three SK-servers with data.
He didn't found any of three three available TCP N0183 streaming data.
He didn't found N0183 data on a streaming serial-USB COM 5.

When looking at Connections after O start it don't quit correspond to what's found. Pict 2.
For the UDP N2k it's set up as a N0183!

Picture 1
[image]

Picture 2
[image]

Hakansv, 2024-04-09:
Later on a "rescan" he found a N0183 COM 6 but there's no data. The valid COM 5 was not found

nohal, 2024-04-09:
Is any of the TCP data sources on your network it's gateway or running on localhost? If not, there is no way to find it, we simply can't scan all possible network addresses for ports where we think some data might be present, would take too much time.
Signal K we already discussed in the other issue, the mDNS query simply has no notion of "I am not the right Signal K server", it simply says "I'm a Signal K server on this IP and this port". 
Wrong type for the N2K connection is a bug. The question is - Is there something transmitting NMEA2000 on that port in your network?
Is there anything connected to USB6 on your system? If so, what is it? What is connected to COM5 and what data does it transmit?
Run with `-l debug` and see what gets written to the logfile

Hakansv, 2024-04-09:
I admit my Win10 test environment is not representative as a "first time user" system. I've to many things connected.
I will setup a clean PC to test on instead.
So far..
TCP not found.
There are two local streaming TCP N0183 (GNSS and AIS) that would have been found. (192.168.1.73:10110 and 192.168.1.144:10120)

SK-servers
There are four local SK-servers that would have been found.
I agree ....167:3000 is valid although no streaming data.

Serial-COM
COM 6 is a virtual port connected to another application so not available.
COM 5 is an available virtual port where GNSS-data is streaming. See below

```
$GPRMC,,A,5749.5946,N,1153.2809,E,9.700,10.000,230394,,W*4e
$HCHDM,7.000,M*2e
$VDVHW,,T,7.000,M,10.000,N,,K*71
$GPGGA,072951.00,5749.6025,N,1153.2835,E,1,07,1.9,3.1,M,40,M,,*47
$IIXDR,P,1.007,B,BAROMETER,A,20.800,,ROLL,A,-6.400,,PTCH,*a
$HCHDM,7.000,M*2e
$IIMTW,14.7,C*11
```

nohal, 2024-04-09:
TCP will never be found unless it is provided by a network gateway (= the usual scenario where a MFD, multiplexer or a RPi provides the wifi) or locally.
Check with Wireshark if your "good" SK servers reply to the mDNS query.

Hakansv, 2024-04-09:
The TCP 192.168.1.73:10110 as an example is from a RPi via a local router on the same IP segment. Putty Telnet finds it

nohal, 2024-04-09:
Is 172.28.1.73 a gateway in your network (= can you see it as a gateway for the 192.168.1.0/24 network in your routing table)?  Putty telnet does not "find" anything, you give it the IP and port manually. Same as if you manually configure the connection in OpenCPN, it then also "finds" it.
We can't scan all the possible ports on all the possible IP addresses, that would take too long. Unlike for UDP where you simply wait for broadcast packets to arrive, for TCP you must know about the server endpoint, and if you don't there is no way to find it but a brute force trying all possible ports on all possible IPs.

Hakansv, 2024-04-09:
> Is 172.28.1.73 a gateway in your network 

172.28.1.73 ?? What's that? Excuse my ignorance but that address I don't recognize. 

Do I understand the above correct that we won't be able able to lookup TCP connections?

The arp dynamic as of this list:  (192.168.1.73  is there)
```
Interface: 192.168.1.67 --- 0x1c
  Internet Address      Physical Address      Type
  192.168.1.1           6c-ba-b8-d0-41-13     dynamic
  192.168.1.64          a8-9f-ec-8c-17-a2     dynamic
  192.168.1.72          a8-9f-ec-8c-17-33     dynamic
  192.168.1.73          dc-a6-32-2c-b8-74     dynamic
  192.168.1.79          a8-9f-ec-4f-b6-4c     dynamic
  192.168.1.144         28-9c-6e-9e-fb-e2     dynamic
  192.168.1.154         e4-5f-01-4c-2d-39     dynamic
  192.168.1.166         b8-27-eb-e1-87-db     dynamic
  192.168.1.167         b8-27-eb-79-41-ef     dynamic
  192.168.1.178         f0-35-75-7a-ba-8a     dynamic
  192.168.1.180         48-d2-24-f4-43-23     dynamic
  192.168.1.204         a0-8c-fd-06-79-9f     dynamic
  192.168.1.233         74-da-da-3c-c7-d4     dynamic
```

nohal, 2024-04-09:
Sorry, 192.168.1.73 of course. And looking above it does not seem to be your gateway, I suppose 192.168.1.1 and there are are probably no endpoints providing NMEA data on it.

Hakansv, 2024-04-09:
192.168.1.1 is the router yes. And an router would never provide data for us.
So I may start to understand your logic. We would never be able to find a TCP via a router. Only if the device is also a gateway like a True Heading AIS-wifi?
If so TCP case closed.

nohal, 2024-04-09:
Currently only if a device is a gateway for the local network, as the multiplexer, or MFD usually is.

Hakansv, 2024-04-09:
Signal-k servers
I've cleaned another Win to test on.
On WS: There where four mdns response including "signalk-ws". Only one was listed and that had no "self" data like MMSI or vname like the other three.
So maybe we could list only a server providing "self=urn:mrn:imo:mmsi" or "vname" or vmmsi"
And list all of them to the user to select.

Hakansv, 2024-04-09:
Here is the pcap if you want:

[mdns-SK.zip](https://github.com/OpenCPN/OpenCPN/files/14916920/mdns-SK.zip)

Hakansv, 2024-04-09:
Serial COM port.
I've a USB COM 3 providing N0183, GNSS. It was not listed.

nohal, 2024-04-09:
Read the logfile with `-l debug` or, better, step through https://github.com/OpenCPN/OpenCPN/blob/master/gui/src/initwiz/wiz_ui.cpp#L197 to see if the port is found at all and if it is why it is not recognized. I currently do not have any Windows machine to test on.

Hakansv, 2024-04-09:
It has to be log file so far. No build environment on that PC so far.
13:11:02.925   DEBUG wiz_ui.cpp:221 Trying COM3 at 9600
13:11:03.441   DEBUG wiz_ui.cpp:267 IOException COM3 at 9600 IO Exception: Error while reading from the serial port: 997, file C:\Builds\OCPN\OpenCPN\libs\serial\src\impl\win.cc, line 397.

The "whole" connection story here:
[Firstuse_debug.txt](https://github.com/OpenCPN/OpenCPN/files/14917868/Firstuse_debug.txt)

Hakansv, 2024-04-09:
> Serial-COM
> COM 6 is a virtual port connected to another application so not available.
> COM 5 is an available virtual port where GNSS-data is streaming. See below

Sorry the above text from my first Win. It was wrong.  COM 6 is the one we should detect here. I's a bit lost. Maybe more than usual?

The Com 3 issues on the second Win remains though

Hakansv, 2024-04-09:
Github master now:
Win10
The busy spinner is gone. Thanks.
Now he detects also TCP on port 10110 that's not a gateway address. Nice!
N2k UDP now works since PROTO is corrected to N2000. Good. See pict.

The USB COM 3 N0183 is yet not detected. 

No SK connections at all detected with this code.
In WS I couldn't detect any mdns request for signalk. 
The "Looking for Signalk......" was also just a one second glimpse before ended. 

[image]

Hakansv, 2024-04-09:
FYI:
On a RPi5 one SK-server was found. It is on local host.
The GPSd I don't know what it is. Nothing found in Debug window for that

[image]

Hakansv, 2024-04-09:
I tested the last commit "Make the serial detection more ...." on two Win.
Neither detected any serial port.
The serial COM 3 is though well listed by Connection's drop down:
If of any help I may try to debug it tomorrow? Was it at wiz_ui.cpp#L197  ?

[image]

nohal, 2024-04-09:
On my Windows VM it works with USB passthrough to a TTL GPS puck connected via CP1202 TTL->USB adapter. I hove no other Windows machine and no other HW available now.
There should be some relevant info the log with `-l debug`

nohal, 2024-04-09:
@Hakansv I have implemented support for detection of multiple Signal K servers now as well...

[image]

Hakansv, 2024-04-10:
COM ports
All my tests here are on dated not so fast Win-PC
Runtime:
The inbound (virtual)COM 6 is detected valid.
COM 3: YDNU in default N0183 mode is not detected on any of two tested Win.
No SK servers found. (UDP & TCP N0183&N2k are found OK)

Debug mode:
All COM including COM 3 are found. See below
Also one SK-server is found

Quick hypothesis: 
In debug everything is slow. The COM3, YDNU can initially be slow to answer in (default)N0183 mode. 
Should we wait for the buffer data to establish before concluding it's empty?

The same for SK-detection?

Next: I'm off to the boat's PC to test on a more "normal" user system. That may be a more "first time user" test. Although neither SK nor UDP/TCP in sight there.
I'll revert when done.

Following COM3 in Debug
```
		known	false	bool
+		known_usb_devices	{ size=2 }	std::vector<USBDevice,std::allocator<USBDevice>>
+		port	{port="COM3" description="Seriell USB-enhet (COM3)" hardware_id="USB\\VID_0483&PID_A217&REV_0200" }	const serial::PortInfo &
+		port.description	"Seriell USB-enhet (COM3)"	const std::string
+		port.hardware_id	"USB\\VID_0483&PID_A217&REV_0200"	const std::string
+		port.port	"COM3"	const std::string
+		this	0x0021db30 {m_detected_connections={ size=0 } }	FirstUseWizImpl *

row 233:
+		serial::Serial::read returned	"$YDHDG,27.6,,,4.2,E*0A\r\n$YDGLL,5813.5705,N,01145.7546,E,,A*29\r\n$"	std::string
		std::basic_string<char,std::char_traits<char>,std::allocator<char> >::append returned	"$YDHDG,27.6,,,4.2,E*0A\r\n$YDGLL,5813.5705,N,01145.7546,E,,A*29\r\n$"	std::string &
+		this	0x0021db30 {m_detected_connections={ size=0 } }	FirstUseWizImpl *
```

Hakansv, 2024-04-10:
User interface comments:
- When scanning connections it's naturally take some time, no problem with that. But we need some (simple) user info O is working on it. I think the "Rescan" button will else be often used to redo since "nothing" is happening.
- We may ask for desired language in the first page. Not all users want the local lang. since some languages are poor covered.
Select between Unglish(US) and system local?
- I suggest we set up the vector Display category to "All" Nowadays most PC will have enough performance for that. (Maybe not relevant here?)

btw:
SK not detected, mdns querry.
I've seen testing "send to peer" that some routers, like mine, may ignore routing mdns messages to all connected PCs. This can be a explanation for not detecting all SK-servers on my system? (You're the expert on this. I'm just fumble)

nohal, 2024-04-10:
No asking for language, this UI must stay simple. If they are able to handle the OS, they are able to handle the OpenCPN wizard. The vast majority of users do have the OS set to a language they not only understand, but prefer and those are the ones we care about.

Of course on networks where mDNS does not pass the switch, it can't work for SK discovery.

Hakansv, 2024-04-10:
Back from boat.
Win10 also rather dated.
Four COM ports with N0183 data. In general only three detected so something fails also here.
One UDP : 10111  N0183 seldom updated, 0.4 Hz

On a one time occasion all four COM ports was detected but on seven scans only three.
UDP not detected but we're not looking for other ports than 10110 if I'm correct?
Pictures from available ports (note: COM1 not active) and scan result. 
Debug log attached.

[image]

[image]

[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/14931146/opencpn.log)

Hakansv, 2024-04-10:
Yes, understood. I just mentioned the UDP:10111 since it was listed on my Connections picture.
The missing COM-port(s) is the issue

Hakansv, 2024-04-11:
Observations for info:
Missed SK-servers (Win)

Comming from Units page.
No SK servers found. 
In WS no mdns "signalk..." request was seen.
The only mdns was later when O is up and running and the PTR queries are sent.

On the first rescan after the "Units" page mdns query from this host was seen.
After that two of four available SK servers was detected.
We may have to much hurry?
827.25 SK Request
827.25 SK Response RPi4 - listed
827.25 SK Response Openplotter - listed
833.21 SK Response Rpi400 - not listed
833.31 SK Response Rpi5 - not listed
The above was seen several times although not the same time spans.

Running on a RPi I saw this:
pi@RPi400:~/Builds/OpenCPN/build $ ./opencpn -W
Opened 2 sockets for mDNS query
Sending mDNS query : _signalk-ws._tcp.local. PTR
Reading mDNS query replies
Read 10 records
Closed sockets
- All above in one or two seconds. Not much time for SK-servers to respond?

PR #3793 (2024-04-08): Implement basic configuration wizard
Implements a wizard started automatically if there is no configuration file or using `-W` / `--config_wizard` command line option

TODO:

- [ ] Implement detection of more N2K formats
- [ ] Testing in more environments, especially on Windows
- [x] Implement detection of Signal K (Done, waits for https://github.com/OpenCPN/OpenCPN/pull/3758)
- [ ] Content for the page at the end of the wizard

Closes #3646