OpenCPN/OpenCPN #3648: Expand the REST interface

leamas, 2024-02-10:
**Is your feature request related to a problem?**
In a headless context some basic pieces are in place: It is possible to transfer and activate a route. There is also some support for headless pairing.

The still missing pieces in a everyday scenario is to
1. List existing routes
2. Activate a route.
3. Reverse a route.

**Describe the solution you'd like**
Add the functionality above to the REST interface, preferably in 5.10.  Actual local GUI could and probably should be done in a plugin later

**Describe alternatives you've considered**
None, really.

**Additional context**
Headless pairingf #3411
Transfer route + pairing #3121

nohal, 2024-02-11:
When the REST interface is touched, I would like to see an option to relay a general JSON payload originating from a plugin, upon reception simply broadcasted as a plugin message to all listeners.
On the originating side either a plugin message (means asynchronous) or a new synchronous API call, or both, could be implemented.
I have a direct use case for it in the DashboardSK plugin implementing configuration synchronization between devices.

leamas, 2024-02-11:
A first step would be a new point 4.  which just receives an arbitrary JSON message and puts it on the message bus as a fourth kind of message besides n0183, n2000 and signalk.

Additional steps:
  - Ńew primitives in ocpn_plugin.h to recieve  the new message type, similar to existing ones for n0183, n2000, etc.
  - A thin wrapper to curllib to send such a message. Could be implemented in ocpn_plugin.h, but could also be done in a opencpn-libs library.

leamas, 2024-02-11:
This also means this not only about me which certainly raises the priority. I'll start working on it.

PR #3654 (2024-02-15): REST server update (#3648)
Add the missing parts described in #3648

As for @nohal's request for a general JSON relay to plugins this is generalized to a sending an arbitrary string. The message has an id similar to communication messages, so plugins listens to a specific kind  of plugin message.

The plugin message service is yet not exported in the plugin interface. ~Will do before merge~ (done)

Closes: #3648

Besides using a remote client testing can also be done using curl. This requires a working api key. How to set up this is described in the [manual](https://opencpn-manuals.github.io/m [... cut]
closed 2024-02-15 by leamas