OpenCPN/OpenCPN #3650: Highlite track does not work as expected and is confusing

nohal, 2024-02-11:
**Describe the bug**

With `highlite track` enabled, the actual track color is partly or completely overridden by the yellow accent and hardly visible/invisible. How much is depending on the zoom level and angle of the line.
The behavior also is not platform and GL/non-GL consistent.

As minor as this might seem, it turns out to be extremely confusing to many users.

Hakansv, 2024-02-11:
I agree about user's confusion like: "Now I've changed the track's color to blue but it's still yellow!?" 
We may consider to instead have "g_bHighliteTracks" default off. Most new users are not aware of this feature so we better make the use of it an active selection?

nohal, 2024-02-23:
Fixed in #3680