OpenCPN/OpenCPN #3651: Garmin GPSMAP 66 - upload from OpenCPN to device does not work for routes or waypoints

thonglorais, 2024-02-11:
When I go for boat charter I usually do my route planning with OpenCPN upfront. Then, on the boat I'm using a Garmin GPSMAP 66st with my data to navigate (in parallel to the charter boat plotter & hardware) and to log the GPS track. Now what I'd like to do is to directly upload waypoints and routes from OpenCPN to the Garmin GPSMAP 66st, unfortunately that doesn't work.

I have seen issue [#3496](https://github.com/OpenCPN/OpenCPN/issues/3496), but in my impression I'm having a different problem, as I don't even get to "MESSAGE comm_n0183_output.cpp:439". I have also checked the other Garmin-related messages in this board, however none resembles the problem that I have.

Now here is the set-up and some analysis:

Hardware: Raspberry Pi 4 Model B Rev 1.5 4GB with Garmin GPSMAP 66st connected via USB
Software: Debian GNU/Linux 11 (bullseye) 64 bits, Kernel: 6.1.21-v8+, OpenCPN 5.8.4-3 (installed via flatpak)

On Garmin I have selected the following mode:
-Setup -> System -> Garmin Spanner
with the following settings:
-NMEA sentences: On
-Autopilot Mode: Off
-Waypoint IDs: Names
-Precision of Minutes: MM.MMM'
-Baud: 9600
-GSA, GSV: On
-WPL, RTE: On
-GRMN Proprietary: On

For the sake of completeness: I have also tried Garmin Serial and NMEA In/Out, but if I select one of those two Garmin modes the Garmin goes directly into mass storage mode when connected per USB.

Then I plug the Garmin into the RPi USB Port, "Would you like to go to Mass Storage" => No

I takes a couple seconds until the Garmin appears on USB port:

pi@R4-4GB:~/.var/app/org.opencpn.OpenCPN/config/opencpn $ lsusb -t
/: Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/: Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
|__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
|__ Port 1: Dev 12, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
|__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=, 12M
|__ Port 3: Dev 22, If 0, Class=Hub, Driver=hub/4p, 480M
|__ Port 4: Dev 42, If 0, Class=Vendor Specific Class, Driver=garmin_gps, 480M

Then I start OpenCPN 5.8.4-3 and go to Options -> Connections -> Add Connection with the following parameters:
-Serial
-DataPort: /dev/ttyUSB0
-Baudrate: 9600
-Protocol: NMEA0183
-List position: 1
-Control checksum: checked/yes
-Receive Input on this Port: checked/yes
-Use Garmin (GRMN) mode for input: checked/yes
-Output on this port (as autopilot or NMEA repeater): unchecked/no
And close the options/connections window.

Then the GPS position reception works fine with received NMEA0183 RMC sentences. So far so good. Interestingly, if I select the checkbox "Output on this port (as autopilot or NMEA repeater)" the received NMEA sentences are mirrored, one can see them in the NMEA window. Now that all works fine and obviously the connection via USB is working.

Then I open the Route & Mark Manager, select a route, click on Send to GPS, select "Serial:/dev/ttyUSB0". Once I click on "Send" the "Send to GPS..." window stops working. I can then only terminate OpenCPN.

After the crash the log says the following:
pi@R4-4GB:~/.var/app/org.opencpn.OpenCPN/config/opencpn $ grep gps_wx_logging opencpn.log
11:57:07.203 MESSAGE gps_wx_logging.h:30 Unit: GPSMAP 66ST Software Version 10.70
11:57:07.204 MESSAGE gps_wx_logging.h:30 D312
11:57:07.204 MESSAGE gps_wx_logging.h:30 D302
11:57:07.204 MESSAGE gps_wx_logging.h:30 D501
11:57:07.204 MESSAGE gps_wx_logging.h:30 D600
11:57:07.204 MESSAGE gps_wx_logging.h:30 D601
11:57:07.204 MESSAGE gps_wx_logging.h:30 D700
11:57:07.204 MESSAGE gps_wx_logging.h:30 D800
11:57:07.204 MESSAGE gps_wx_logging.h:30 D900
11:57:07.204 MESSAGE gps_wx_logging.h:30 D911
11:57:07.204 MESSAGE gps_wx_logging.h:30 D912
11:57:07.204 MESSAGE gps_wx_logging.h:30 D913
11:57:07.204 MESSAGE gps_wx_logging.h:30 D917
11:57:07.204 MESSAGE gps_wx_logging.h:30 D918
11:57:07.204 MESSAGE gps_wx_logging.h:30 Waypoint: Transfer -1 Type -1
11:57:07.204 MESSAGE gps_wx_logging.h:30 Route: Transfer -1 Header -1 Type -1
11:57:07.204 MESSAGE gps_wx_logging.h:30 Track: Transfer 301 Type 302

If I shut down OpenCPN peacefully, without crashing it through the "Send to GPS..." window also the following messages appear in the log:
13:11:59.189 MESSAGE comm_drv_n0183_serial.cpp:261 Closing NMEA Driver Serial:/dev/ttyUSB0
13:12:00.203 ERROR gps_wx_logging.h:14 A800_Off: Not acknowledged

As the Garmin GPSMAP handheld series all use similar software, I guess this behaviour will be the same for GPSMAP 64, 65, 66, 67, 79 and 86 - just in case somebody has one of these devices. 

How can I get this working to send routes & waypoints to the Garmin GPSMAP 66st? Thanks for any help!

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

bdbcat, 2024-03-21:
Hello...
Maybe you have not set the general Connections panel checkbox 
"Use Garmin GRMN (Host) mode for uploads"

??

thonglorais, 2024-03-22:
Hello bdbcat, surely the flag "Use Garmin GRMN (Host) mode for uploads" was activated. I've just tried it again, below please find the screenshots: 
[image]
[image]
[image]

leamas, 2024-06-13:
@thonglorais:  The beta is out with substantial changes to the garmin upload code. Could you possibly test the beta to check if the problem still persists?

Info on installing the beta [here](
https://www.cruisersforum.com/forums/f134/opencpn-v5-9-1-beta-test-starts-286845.html)

thonglorais, 2024-06-15:
@leamas Thank you very much for looking into this problem!

I have re-tested it on Windows, and it still does not work. However there is a small improvement visible: OpenCPN does not crash anymore but gives an error message. 

Test execution similar to as described above, just on Windows: GPSMAP 66ST connected sucessfully, position received from GPS. Upload of route ends with error. OpenCPN does not crash but gives an error message: 
[image]

Log file is here:
[opencpn_5.9.1-0+b2cbf5a.log](https://github.com/user-attachments/files/15846013/opencpn_5.9.1-0%2Bb2cbf5a.log)

I tried it two times (search for "Sending Routes..."), first time with flag "Output on this port (as autoopilot or NMEA repeater)" not activated, the second time it was activated - no change in result.

Hakansv, 2024-06-15:
I tested again using Beta 5.9.1 and Win10

1. On my GPS 12 device in setup menu Interface I changed to GRMN/GRMN mode. Then he says "Waiting....."
2. I connected the GPS via a serial-USB converter to the PC. (In Device manager I detected COM 3)
3. In OCPN Options->Connect I enabled "Use Garmon host mode for uploads".  See pict 1
4. On a route I selected "Send to new GPS" (Since this connection is new)
5. There I selected COM3 and OK
6. The Garmin device reported "Transfer complete"
7. OCPN reported "Route transmitted" See pict 2

Note1: You don't need to set up a OCPN connection for this transfer. Also if there is a OCPN connection for the same port, my COM 3, these settings are not used for the "Send to GPS" function. Such a connection is automatically disabled during this transfer.
Note2: NMEA Debug window is not used for the "Send to GPS" transfer.

[image]
[image]

Hakansv, 2024-06-15:
Next.
In the case the device don't use GRMN mode to receive a transfer but simply accepts NMEA0183: WPL and RTE you have to set up an output OCPN Connection to the port in question.
Set "Transmit only sentences" to filter "WPL" and "RTE". 
And! Disable "Use Garmon host mode for uploads".

The it looks like this:
```
<BLUE>--> 15:31:54 (Serial:COM3) $ECWPL,5806.279,N,01108.251,E,001*62
<BLUE>--> 15:31:54 (Serial:COM3) $ECWPL,5803.554,N,01108.251,E,002*6C
<BLUE>--> 15:31:55 (Serial:COM3) $ECWPL,5753.809,N,01108.102,E,003*67
<BLUE>--> 15:31:55 (Serial:COM3) $ECWPL,5801.814,N,01101.154,E,004*6E
<BLUE>--> 15:31:56 (Serial:COM3) $ECWPL,5806.910,N,01102.499,E,005*6A
<BLUE>--> 15:31:56 (Serial:COM3) $ECRTE,3,1,c,Test,001,002*11<0x0D><0x0A>

```

thonglorais, 2024-06-17:
I tested again using Beta 5.9.1 and Win11. In my GPSMAP 66ST, there is no "GRMN/GRMN" mode. Therefore I chose Setup -> System -> Garmin Spanner mode. In OCPN Options->Connect I enabled "Use Garmin host mode for uploads". I do not activate a connection. On a route I selected "Send to GPS". Results in an error, picture & log file attached:
[image]

[opencpn_5.9.1-0+b2cbf5a restarted at 2024-06-17.log](https://github.com/user-attachments/files/15877271/opencpn_5.9.1-0%2Bb2cbf5a.restarted.at.2024-06-17.log)

bdbcat, 2024-06-17:
I think I understand the problem here.  Will get to it.  Thanks for detailed reports.
Dave

bdbcat, 2024-06-23:
Please let me know when you are back online.
I have questions.

soggygeek, 2024-06-26:
My Garmin 78sc behaves as this GPSMAP 66 does. I can help test.

bdbcat, 2024-06-26:
I have a Garmin device coming by mail this week, I hope.  Supposed to show the same behavior.  I'll keep you posted as I learn more.
Thanks
Dave

thonglorais, 2024-06-28:
> Please let me know when you are back online. I have questions.

Hi bdbcat, I'm back online

bdbcat, 2024-06-29:
Hi, welcome back.
I have learned much from the Garmin Colorado device I am testing now.  But not quite enough.
Leaving Windows aside for the moment.
I suspect the latest OpenCPN Beta for linux/rpi may actually succeed in transferring routes and waypoints.

Conditions:
First, set up a connection to receive position data from the device.  All this is the same as you have demonstrated for O584, upthread.
1.  Device set for Garmin Spanner mode.
2.  comm port is something like /dev/ttyUSB0, or whatever is found on your system.
3. In Settings->Connections, enable GRMN mode for input on this connection.
4. Also, Enable GRMN Host mode for uploads, globally.
5. Set port to 9600 baud.

With a valid GPS signal, this setup should show position in OCPN.

Now you may try to upload a route, or a single waypoint.  Use the same port, e.g. /dev/ttyUSB0 as is being used for the connection.

The OCPN logfile will contain much useful information.  Please post a copy after you have tested this.

Windows is a somewhat different beast, since the GarminSpanner mode activates a specific device driver in Windows.  Needs a bit of work...

Thanks
Dave

thonglorais, 2024-06-29:
With my Raspberry Pi I'm on flatpak. Using the instruction 
```
flatpak install --user  https://flathub.org/beta-repo/appstream/org.opencpn.OpenCPN.flatpakref
``` 
I only get OpenCPN version 5.9.0-3+5ab4db03b. In this OpenCPN version I do not receive a position from my Garmin GPSMAP 66, therefore further testing doesn't make sense. I've requested to update the flatpak build [here](https://www.cruisersforum.com/forums/f134/opencpn-v5-9-1-beta-test-starts-286845-11.html#post3913082).

bdbcat, 2024-06-29:
There may be USB permission problems with flatpak.  Separate issue, investigating.
Please try the route upload procedure above on Windows, and post the resulting logfile.
Thanks

bdbcat, 2024-06-29:
To clarify, I mean Windows latest Beta 5.9.1

thonglorais, 2024-06-29:
Above (two weeks ago) I tested with 5.9.1-0+b2cbf5a on Windows and it did not work. 
Where do I get a OpenCPN Windows version higher than 5.9.1-0+b2cbf5a?

bdbcat, 2024-06-29:
thor...
I need the Windows OCPN logfile.
And now I see that you have attached with your previous tests.  Sorry, I missed that.
And the logfile gives us what may be bad news.
It appears as though the device, in Garmin-Spanner mode, does not support direct route and waypoint upload.
From the log:
22:58:56.457   ERROR gps_wx_logging.h:14 Send_Route: Unknown route protocol
I see exactly the same thing on Garmin Colorado 400c.
A bit more detail:
When OCPN sets up the USB interface to the device, part of the process is a query by OCPN to enumerate the protocol capabilities of the device.  There is a particular well-known route upload protocol, implemented in OCPN, which has been supported for many previous models of Garmin devices.  But it seems this protocol is not supported when the device is configured as Garmin-Spanner.  At least that is my theory.
Reading the Garmin User Manual, it suggests that the mode we want is "Garmin-serial".
I have tried setting my device to that mode.  When I plugi in USB cable to PC, the device immediately goes into mass-storage mode, with no obvious way to access any serial data stream.  Perhaps you could try this with your device?

More research required, obviously.

bdbcat, 2024-06-29:
From the Garmin 66st manual:

Garmin Serial: Sets the device to use a Garmin proprietary format to exchange waypoint, route, and track data
with a computer.
NMEA In/Out: Sets the device to use standard NMEA 0183 output and input.

It is not at all clear how to access the data in these modes from the USB connection.

bdbcat, 2024-06-29:
I also have a Garmin 60Csx.  This device has a USB connection, as well as the standard Garmin 4-pin serial connector.  It does not support Garmin Spanner mode.
Could it be that the 66st is using the same basic firmware, but dropped the serial connector, assuming that the Spanner mode would be sufficient for most user needs?

thonglorais, 2024-06-29:
I have posted my little knowledge on Garmin devices here: 
https://www.cruisersforum.com/forums/f134/opencpn-v5-9-1-beta-test-starts-286845-11.html#post3913105

thonglorais, 2024-06-29:
So, I set my GPSMAP 66 to "Garmin Serial".  When I plug-in the USB cable to PC, the device immediately goes into mass-storage mode. Same happens when I set it to "NMEA In/Out". Then, from within OpenCPN, the device is not recognized: 

[image]

bdbcat, 2024-06-29:
Thanks for the precis on Garmin device numbers and interfaces.
As I interpret this, said in another way:  "Recent Garmin devices with only USB interface depend on Garmin-Spanner mode for PC positioning input."
On the 60Cszx:  The expected route upload protocol is present and exposed in this device.  Raoute upload from OCPN is possible using the USB connector.

So, in your more thorough understanding of Garmin device history, can you see any way to access "Garmin-serial" mode on the 66?

thonglorais, 2024-06-29:
I think the Garmin 4-pin serial connector was the last "real" serial port, afterwards the serial port has been virtualised.
I don't have any additional knowledge on how to access "Garmin-serial" mode on the virtual USB.

bdbcat, 2024-06-29:
So, it starts to look as if the way to upload OCPN routes to recent USB-only Garmin devices is to:
1.  Export Route in OCPN, as a GPX fil, stored on the PC.
2.  Import that GPX file from the PC to the Garmin device, using the mass-storage mode.

I have not tried this myself.  Would be curious as to results of this workflow.

Any other ideas?

thonglorais, 2024-06-30:
Yes, the above described method works: export route in OCPN as GPX. Put the device into mass storage mode. Move the GPX file to folder D:\Garmin\GPX on the device. The route will appear in the Route Planner (seee attached screenshots). 

This works both under Windows and RasPi-Bookworm. 

[image]
[image]
[image]
[image]
[image]

thonglorais, 2024-06-30:
I had a further look at how Garmin transfers waypoints & routes with its own software. There is a software called [BaseCamp](https://www8.garmin.com/support/download_details.jsp?id=4435) with which certain data can be transferred to Garmin devices. Also this software works only if the device is in Mass Storage mode. To mea this means that also Garmin access the device as a drive, not per serial interface/Garmin Spanner.

thonglorais, 2024-06-30:
Now I tried this also for marks/waypoints and it also works. Also, there is an easier worflow: 
-connect the device in Mass Storage mode and wait until appears as a drive (can take up to 1 minute in Bookworm)
-in OpenCPN use the button "Export selected" and directly store it in folder D:\Garmin\GPX on the device
On the device routes can be accessed in the "Route Planned" and marks/waypoint can be accessed in "Waypoint Manager", see attached screenshots.

[image]
[image]
[image]
[image]
[image]

thonglorais, 2024-06-30:
I checked in which set-up a position can be received from the Garmin device. Turns out that this only works if **not** in Mass Storage mode. Conclusion for me:
-in Mass Storage mode routes and waypoints can be transferred to and from the device, directly from OpenCPN
-in Non Mass Storage mode GPS positions can be received from the device
Both activities at the same time doesn't seem to be possible.

In Mass Storage mode:
[image]

In Non Mass Storage mode:
[image]

thonglorais, 2024-06-30:
It is important to note that these findings only apply to the handheld devices with USB-port (B-mini, B-micro or C). Devices with a real serial port (4-pin round connector) behave differently. 
The 4-pin round connector is obsolete since more than ten years and not implemented anymore in newer Garmin handheld devices. Future devices will get the USB-C port (first device being the GPSMAP67). Therefore I think it's worth to describe this clearly in the OpenCPN documentation. I can offer to write-up a draft version, which you can add to the OpenCPN knowledge base.

Hakansv, 2024-06-30:
> I can offer to write-up a draft version, which you can add to the OpenCPN knowledge base.

I'm sure @rgleason will kindly accept your offer.

rgleason, 2024-06-30:
Yes please, we will add this to write up to the sharing page in the wiki 
If you want to use Word for the writeup and screenshots that will work. Attach it here or to a thread in the forum. Thanks.

thonglorais, 2024-06-30:
> 2\. I connected the GPS via a serial-USB converter to the PC. (In Device manager I detected COM 3)

Hi Hakansv, above you wrote "I connected the GPS via a serial-USB converter to the PC. (In Device manager I detected COM 3)" - can you describe the cable or model that you used here? Does it have the 4-pin round connector on one end, then inbetween a serial-USB converter circuit and then a USB-B connector? 

I'm asking because it was possible to connect the 4-pin round connector directly to a serial port (RS 232/DE-9) on a PC. Now I know most PCs today do not have a serial port anymore, that's probably why the detour via USB is neccessary.

thonglorais, 2024-06-30:
> Yes please, we will add this to write up to the sharing page in the wiki If you want to use Word for the writeup and screenshots that will work. Attach it here or to a thread in the forum. Thanks.

@rgleason is there any current description of Garmin route/WP transfer for OCPN?

Hakansv, 2024-06-30:
> can you describe the cable that you used? Does it have the 4-pin round connector on one end, then inbetween a serial-USB converter circuit and then an USB-B connector?
> I'm asking because it was possible to connect the 4-pin round connector directly to a serial port (RS 232/DE-9) on a PC. Now I know most PCs today do not have a serial port anymore, that's probably why the detour via USB is neccessary.

Yes, correct. It's a dated device. Round 4-pin to the device and a DB9 to the PC thus needs a USB-serial converter these days.

> And - question 2 - can you receive a position and send routes/WPs at the same time? Or are different modes neccessary?

No. For data(WP/RTE) transfer GRMN/GRMN mode is necessary on the device. For position NMEA/NMEA mode is needed so no simultaneous transfer and receive. (Now on my device there is a NMEA/GRMN mode I haven't tried)

thonglorais, 2024-06-30:
> Yes, correct. It's a dated device. Round 4-pin to the device and a DB9 to the PC thus needs a USB-serial converter these days.
>
>
> No. For data(WP/RTE) transfer GRMN/GRMN mode is necessary on the [...]

So the USB-serial sits between DB-9 and an USB-A PC connector?

Hakansv, 2024-06-30:
> So the USB-serial sits between DB-9 and an USB-A PC connector?

Yes, so to say. Like this:
[image]

thonglorais, 2024-07-07:
@rgleason @bdbcat @Hakansv  
Attached please find the draft write-up for the OpenCPN knowledge base. @Hakansv: I have re-used some of your descriptions & pictures as I don't have any Garmin GPS with real serial port anymore. I hope that works with the screenshots - I've not used any manual page breaks to make formatting easier. Feel free to edit as required. If I should add or change something let me know, please. 

[Transferring routes or waypoints from OpenCPN to Garmin handheld devices.docx](https://github.com/user-attachments/files/16118953/Transferring.routes.or.waypoints.from.OpenCPN.to.Garmin.handheld.devices.docx)

closed 2024-07-07 by thonglorais

Hakansv, 2024-07-07:
Well done. Good to have the export method for concerned devices explained. That may clear a lot confusion. ("As simple as that!") I do think @rgleason can use this in the manual?

thonglorais, 2024-07-19:
@rgleason 

I've checked it, but unfortuantely text and pictures have gotten mixed-up. Some pictures are missing, and several ones are not at the right place. Also, the paragraphs are not set-up correctly, so currently it is not easy to read & understand. 

I have saved above Word document as PDF, so the formatting and positioning of pictures can be easier identified:

[Transferring routes or waypoints from OpenCPN to Garmin handheld devices.pdf](https://github.com/user-attachments/files/16309896/Transferring.routes.or.waypoints.from.OpenCPN.to.Garmin.handheld.devices.pdf)