OpenCPN/OpenCPN #3652: Option User Fonts. Changes are not saved to config so they are restored on next OCPN restart.

Hakansv, 2024-02-11:
**Describe the bug**
User font changes are not saved to config file. So on next OCPN restart the previous selections are restored.
**To Reproduce**
Steps to reproduce the behavior:
1. Go to Options->User->Fonts. Select one e.g. AIS Target name.
2. Change size from 12 to 9. "Select" -> "Apply" -> "OK".
3. Now the font size for AIS target names has changed from 12 to 9.
4. Close O and start again.
5. Go to Options->User->Fonts. Select AIS Target name. The font size is now again 12

**Expected behavior**
Any font change should update the config file for next O start.

**Desktop (please complete the following information if applicable):**
 - OS: Win10 and RPi Debian(Raspi)
 - Version OCPN v 5.9 as of Github master today.

**Additional context**
Other user settings than fonts are though saved to config.

Hakansv, 2024-02-11:
Here are strange things going on:
First using en_US:
If I change font and close OCPN the config file has the new font size in [Settings/MSWFonts]
When I start OCPN the (probably) default values are printed to config file during initialization and also used on the Canvas.

If I change language from en_US to sv_SE the font collection for SE are saved to [Settings/MSWFonts]
The same things happens also sv_SE. A changed font is saved to config on O close but changed to default during next start.

AND:
When I change language from en_US to sv_SE the en_US collection is deleted from config file, [Settings/MSWFonts].
And vice versa is the sv_SE collection deleted when having changed to en_US.
With this logic my changes any language fonts will be deleted when I shift language and lost when I later shift back.

Hakansv, 2024-02-12:
It's hard bisect down with all structural changes but this issue is not present after installing a 5.9 from 23 Oct-2023

nohal, 2024-02-12:
Thanks for bisecting it, will have a look at it

closed 2024-02-12 by nohal

Hakansv, 2024-02-12:
It's the little things that make it??
Thanks

nohal, 2024-02-12:
Yes, usually... The `Read` API is somewhat unfortunate and I missed it while replacing the leaked pointer.

commit acd19c3d (2024-02-12): Fix loading of fonts from configuration file. Closes #3652