OpenCPN/OpenCPN #3655: Save

antipole2, 2024-02-13:
**Is your feature request related to a problem? Please describe.**
On too many occasions I have lost hours of planning work because OCPN has crashed unexpectedly.  (I know it shouldn't but it does sometimes happen.)  I believe OCPN should recover through the changes xml file but this has not saved me.  To forestall loss I try to remember to quit OCPN and restart from time to time but that is a nuisance and untidy.

**Describe the solution you'd like**
A Save option in the menus would be much appreciated.  It should do what is done before a clean termination.  Many applications have a Save menu item, so this would be easy for users to comprehend.

nohal, 2024-02-13:
Are you experiencing data loss even with the current code? If you are, https://github.com/OpenCPN/OpenCPN/pull/3585 was not enough, but I would really need some concrete information, especially about the nature of the crashes and `navobj.xml.changes` state after them to do something further.
I consider some further development here, a) making the writing of `navobj.xml.changes` as atomic as possible by writing to another file and then copying it, which should limit the maximum possible data loss to the very last change and b) Actually saving `navobj.xml` with the changes applied more often. I'm afraid manual "Save" command will do nothing for the vast majority of people as is the case in pretty much all the software.

antipole2, 2024-02-13:
Apologies... my bad.

I have not yet got around to[ doing this](https://github.com/OpenCPN/OpenCPN/issues/3114#issuecomment-1935646765).  At present, I cannot import a plugin into OCPN 5.9 as I had a plugin error a while back.  So I revert to v5.8 when I need to import and tend to go on using that for 'production'.  Hence, this issue.
Nothing stops me running v5.9 once the plugin is imported under 5.8.

I have just launched 5.9, created some waypoints and a route and force-quitted it.  They were there on relaunch.  Great - thank you.  This is wonderful! 
I will close this issue when you have had a chance to see this comment.

nohal, 2024-02-13:
If you can't import a plugin to 5.9, and the same works on 5.8, we probably want to see it...

antipole2, 2024-02-13:
I am assuming this is because I have not got around to building my own OCPN with the patch applied [as here] (https://github.com/OpenCPN/OpenCPN/issues/3114#issuecomment-1935646765)?

closed 2024-02-15 by antipole2