OpenCPN/OpenCPN #3657: ref_scale missing from class ViewPort in ocpn_plugins.h

antipole2, 2024-02-13:
**Describe the bug**
Class viewPort  is defined in the viewport.h header.  It includes these attributes:

```
  double chart_scale;  // conventional chart displayed scale
  double
      ref_scale;  //  the nominal scale of the "reference chart" for this view

  int pix_width;
  int pix_height;
```

The class is also defined in ocpn_plugin.h for the convenience of plugins but _ref_scale_ is missing from there

```
  float chart_scale;  // conventional chart displayed scale

  int pix_width;
  int pix_height;
```
Maybe the type for _chart_scale_ should be consistent too.

antipole2, 2024-02-13:
PS. Are the pixels in this class DIP?  (Device Independent Pixels)  I have browsed the code for a while but not found the answer yet.

bdbcat, 2024-02-13:
Once again, if we want to change an existing class (e.g. PlugIn_ViewPort) in the plugin API, we cannot do it directly.
However, we could define  a new PlugIn_ViewPort_Ex, with added fields.
To maintain plugin compatibility with older OCPN versions (e.g. 5.8.x), any plugin wanting to use the new class would need to do a dynamic lookup of the new class in the OCPN core, and use it if found.  Otherwise fall back to the existing class.

bdbcat, 2024-02-29:
Adding this would be a significant ABI breakage for plugins using a prior plugin API. (e.g. API <= 18)

What would be the purpose/use of the member "ref_scale" to a plugin?
If not truly compelling, I'd say NOTABUG, WONTFIX.

antipole2, 2024-02-29:
I have no need or use for this.  I just happened to discover the inconsistency between the plugin and internal versions of opcn_plugin.h and reported it.

Happy for this to be closed with whatever - WONTFIX?

bdbcat, 2024-02-29:
WONTFIX is fine.

antipole2, 2024-02-29:
WONTFIX

closed 2024-02-29 by antipole2