OpenCPN/OpenCPN #3658: Object query broken usning double canvas. (Windows)

Hakansv, 2024-02-13:
**Describe the bug**
When double canvas is active the first double click (or canvas menu) for an object shows related info. Next time object query shows only additional text info.

**To Reproduce**
Steps to reproduce the behavior:
1. Use double canvas and some ENC charts.
2. On left canvas double click an object. See shot 1. Close that and double click the same or another object. Now only "Additional info" is shown. Se shot 2. Subsequent object query on the same Canvas shows only "Additional info"
3. Double click an object on the right Canvas. Now related info is shown. Subsequent Object Queries on the same Canvas shows only "Additional info".

**Expected behavior**
Every call to a Object Query would always show all relevant info.

**Screenshots**
Shot 1
[image]
Shot 2
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Win 10  (Not reproducible on my RPi5)
 - Version  OCPN 5.9 Github master branch "now".

nohal, 2024-02-17:
I can not reproduce the fault on the left canvas, there is always "something" shown in the query window.
I can pretty easily reproduce it in the right canvas - usually the second query in the same location brings just the additional info. But not always.

There seems to be a more serious problem behind this though, I'm pretty consistently able to reproduce a problem where the query is done exactly on eg. a light, many objects are returned, but the very light is completely missing.

bdbcat, 2024-02-17:
There was a commit in 5.8.4, that I thought suspicious, but could not prove it.  Came from a relatively unknown developer.  The idea was to somehow filter or limit the query results.  Details elude me.
I'll try to find it.

bdbcat, 2024-02-17:
Found the one I was thinking of:
#2720

nohal, 2024-02-17:
I'm not sure this has potential to be the cause - seems to me like it pushes everything from `selected_rules` to `ret_ptr` after sorting it, so the total amount of returned objects should be the same as before this patch...

bdbcat, 2024-02-17:
It should be easy to walk the code in debug when selecting exactly one isolated light, and see what happens.
After reading @2720, I definitely remember this....

nohal, 2024-02-17:
It actually seems like the doubleclick results in a wrong location and right-click->object query is good (or better), at least on the Mac I'm using right now.

commit 74620e8a (2024-02-22): Correct S57 ENC object query logic in dual canvas mode. Closes #3658
closed 2024-02-22 by bdbcat