OpenCPN/OpenCPN #3662: AIS target query buttons wrong

nohal, 2024-02-14:
**Describe the bug**

On macOS the AIS query window buttons are wrong

[image]

nohal, 2024-02-16:
Done in a156a908d5af0c28a6cae5612d9722909c343d7d

closed 2024-02-16 by nohal

Hakansv, 2024-02-16:
Pavel..
This change did affect the AIS Meteo Target query.
Before the change a scroll list was not created. Pict 1
After the change a scroll list is always created, Pict 2
A Class A target query is though as it should also after this change. (Strange ?)

Please if you have a change to test, and correct, also the Meteo query. (Included in my test stream as you may know.) 

Pict 1: 
[image]  

Pict 2:
[image]

More maybe not relevant any more:
The change below with "- AdjustBestSize(td.get());" I touched some weeks ago since it solved my issue that time. That was based on a positive test and not knowledge.
(Now it's not a solution to include it again.)

```
@@ -327,9 +324,8 @@ void AISTargetQueryDialog::UpdateText() {
       m_createTrkBtn->Disable();
     else
       m_createTrkBtn->Enable();
 
-    AdjustBestSize(td.get());
     RenderHTMLQuery(td.get());
```

Hakansv, 2024-02-17:
@nohal Thanks. Last changes did it.
One side effect: 
If I have the target query dialog already open and double click another target the dialog gets wider. 
Another target double click while it's still open yet another step wider, and so on.
You have to close the dialog in between each target to avoid growth.

Hakansv, 2024-02-17:
Works like a charm. Thanks!