OpenCPN/OpenCPN #3666: RFE: Add shortcuts help key

leamas, 2024-02-15:
**Is your feature request related to a problem?**
OpenCPN has a multitude of useful shortcut keys. However, the are in most cases not that intuitive and only available in the manual which is not always readily available.

**Describe the solution you'd like**
A new shortcut key (probably '?') which displays the available shortcuts for actual platform

**Describe alternatives you've considered**
None, really

**Additional context**
We have already a lot to do for 5.10, this might have to wait until 5.12

Hakansv, 2024-02-15:
Although not updated we have in the inbound manual:
[image]

leamas, 2024-02-15:
Yes, I'm aware of it and also mentioned it.. However, a more direct shortcut is much more intuitive and also more in line with general application  behaviour. 

Besides, having it in the code base might make the maintenance simpler.

Hakansv, 2024-02-15:
Betcha..

leamas, 2024-02-15:
Had to google that, and still not really sure what you mean ;)

Hakansv, 2024-02-15:
I do agree!

PR #4147 (2024-09-22): Manuals: Use  upcoming manual plugin instead of  bundled html in built-in plugins
This is related to the new manual plugin described in #4145.

The built-in plugins has some html which is used as help. The html is hard to maintain, outdated and not even used everywhere.

This PR aims to replace the bundled html with the "regular" plugin manual. If the manual plugin is installed, this is used without any more dialogs. If not, user is offered to browse the on-line manual instead if she has an internet connection.

EDIT:  added from here
Here is:
  - New Help buttons in the four built-in plugins' [... cut]
closed 2024-09-22 by leamas