OpenCPN/OpenCPN #3673: Long drawing of ENC objects when panning in OpenGL

bc38, 2024-02-20:
When OGL is disabled everything is fine.
also for some reason it took a long time for the program settings window to open

CPN version of latest build master
Win11
Video card nvidia RTX 4070
Previously I had the exact same problem on nvidia 1070
[Screen capture video](https://www.dropbox.com/scl/fi/z58oe45p34446n76x11cv/024-02-20-190104.mp4?rlkey=8nuja8meroerr9r4ai99k360i&dl=0)

bc38, 2024-03-11:
tested on a laptop with AMD Radeon graphics
There is no such long delay in rendering objects when panning. There are minor artifacts in the rendering of depth marks that disappear after a second.
I checked on a laptop with modern Intel IRIS graphics, the problem persists and there are glitches, which leads to opengl being disabled.

bdbcat, 2025-04-26:
Please re-submit with accessible video.

closed 2025-04-26 by bdbcat