OpenCPN/OpenCPN #3675: Chart Downloader disappears when switching languages

bc38, 2024-02-20:
Also disappeared on ENG
[Screen Capture Video](https://www.dropbox.com/scl/fi/xjyzew53b9no3q6ptyep6/224-02-20-194858.mp4?rlkey=n9aguled0hnf45qov7pqr2imm&dl=0)

transmitterdan, 2024-02-20:
I can reproduce the chart downloader disappearance.  It's actually much worse than just disappearing.

The obvious workaround for now is to simply restart OpenCPN after changing language.

nohal, 2024-02-20:
What language and what exactly is the problem? I just tried French and Russian and it seems in both the plugin works fine after restarting OpenCPN. The original problem reported here of course does exist.

transmitterdan, 2024-02-21:
> After reboot it appears! But the language I use is why the plugin then stops working, an unknown dependency. This does not happen with other plugins.

Which plugin stops working?  Describe exactly what is not working.

bc38, 2024-02-21:
> > After reboot it appears! But the language I use is why the plugin then stops working, an unknown dependency. This does not happen with other plugins.
> 
> Which plugin stops working? Describe exactly what is not working.

Chart Downloader plugin stops working
After reboot it works!
BUT why then do other plugins work and not be disabled by changing the language without rebooting the CPN. This is clearly some kind of bug!

transmitterdan, 2024-02-21:
Yes, it is a bug.  Chart Downloader plugin is different from other plugins because it adds a panel to the Options dialog box.  The bug is that when changing languages OpenCPN has to destroy the Options dialog box so it can be reloaded with new language text.  But for some reason, when the Options dialog box is reloaded it does not notify the plugin system so the plugins (like Chart Downloader) that added panels to the Options dialog can reload again.  Chart Downloader thus no longer appears in the Options dialog because it was not told to do that.  I think it is a fairly simple thing to fix. We are looking at it now.

If you change lanugage, simply close and restart Opencpn, then during program startup it correctly installs the downloader panel in Options dialog with the new language.  After that you will have no problems so long as you do not again change the language.

transmitterdan, 2024-02-21:
This bug is fixed in master branch now. See https://github.com/OpenCPN/OpenCPN/pull/3679

transmitterdan, 2024-02-21:
Can we close?

closed 2024-02-22 by bc38