OpenCPN/OpenCPN #3682: Crash when change language

Hakansv, 2024-02-22:
**Describe the bug**
O crash from Dashboard when language is changed. 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to Options->User  Change language. From English to (for me) Svenska. Or the other way.
2. Click on OK  - Crash

**Expected behavior**
No crash

**Desktop (please complete the following information if applicable):**
 - OS: Win 10
 - Version 5.9 Github master as of today

**Additional context**

Crash at "void DashboardWindow::SendSentenceToAllInstruments(DASH_CAP st, double value,  wxString unit) 

<!DOCTYPE html>
<HTML>
<head>
<title>Document</title></head>
<body>

  | Name | Value | Type
-- | -- | -- | --
  | i | 0 | unsigned int
▶ | m_ArrayOfInstrument | {...} | wxArrayOfInstrument
  | st | OCPN_DBP_STC_ALTI (33) | DASH_CAP
▶ | this | 0x003a0043 {m_Container=??? m_binPinch=??? m_binPan=??? ...} | DashboardWindow *
▶ | unit | {m_impl=L"m" m_convertedToChar={m_str=0x00000000 <NULL> m_len=0 } } | wxString
  | value | 33.019999999999996 | double

</body>
</HTML>

Call stack:
```
>	dashboard_pi.dll!DashboardWindow::SendSentenceToAllInstruments(DASH_CAP st, double value, wxString unit) Line 5936	C++
 	dashboard_pi.dll!dashboard_pi::SendSentenceToAllInstruments(DASH_CAP st, double value, wxString unit) Line 962	C++
 	dashboard_pi.dll!dashboard_pi::HandleN2K_129029(ObservedEvt ev) Line 2317	C++
 	[Inline Frame] dashboard_pi.dll!dashboard_pi::Init::__l2::<lambda_427c4cbb2f90bf826fd31bdba6190b56>::operator()(ObservedEvt) Line 656	C++
 	dashboard_pi.dll!wxEventFunctorFunctor<wxEventTypeTag<ObservedEvt>,<lambda_427c4cbb2f90bf826fd31bdba6190b56>>::operator()(wxEvtHandler * __formal, wxEvent & event) Line 547	C++
 	[External Code]	
 	[Frames below may be incorrect and/or missing, no symbols loaded for wxbase32u_vc14x.dll]	
 	opencpn.exe!PluginLoader::UpdatePlugIns() Line 758	C++
 	opencpn.exe!ApplyLocale() Line 8229	C++
 	opencpn.exe!MyFrame::DoOptionsDialog() Line 4152	C++
 	opencpn.exe!MyFrame::DoSettings() Line 2788	C++
 	opencpn.exe!MyFrame::OnToolLeftClick(wxCommandEvent & event) Line 2659	C++
 	[External Code]	
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 725	C++
 	[External Code]	
```

Hakansv, 2024-02-22:
Note..
Now after a while the crash is no longer reproducible.
But, if any Dashboard instrument panel is open while language is changed all Dashboards are closed.
So they have to be restored by the Tool's icon.
It's of course better to close Dashboards than crash O?
If no further hints or thoughts I'll close this issue later on.

transmitterdan, 2024-02-22:
When language changes, O closes down all plugins. Then it changes the 'Locale' and reloads all plugins. Dashboard should restore to the last saved configuration but any open dialogs will not be reopened.

Language change is a pretty rare event for most users. I think that's why we haven't seen bugs related to it. I could imagine there are a few remaining bugs in this area that will be hard to duplicate.

Hakansv, 2024-02-22:
Now tested on more machines.
All survived a language change but strangeness occurred on Dashboard. 
D closed during the change and reopened all previously open panels! But the right click menu wasn't aware. No checks on the three open panels. Closing and reopen D restored everything.
This just for info. No need for further work I think. As said, as long we don't crash we can stand some minor issues for this event.

closed 2024-02-22 by Hakansv