OpenCPN/OpenCPN #3684: 5.9.0 crashes on macOS, macMini M1,

bcn2, 2024-02-22:
**Describe the bug**
Crash on start-up, safe start without PIs does not make a difference

Had to step back until opencpn_590-012148ec4537pkg from two days ago

**Desktop (please complete the following information if applicable):**
 - OS: macOS 14.2.1

nohal, 2024-02-22:
Is the crash report generated? Not happening here, so hard to fix without further info...

bcn2, 2024-02-22:
Interesting,
installed the latest version from today again in order to generate the crash report, now it will not crash.
There is just that annoying warning about assertions and then is starts. 
[image]

bcn2, 2024-02-22:
Here we go...
[image]

nohal, 2024-02-22:
OK, the MeteoVache plugin, something for @Rodemfr :)

bcn2, 2024-02-22:
Disabling the Meteovache PI that one is gone indeed, now the next one (which has been there for some while as well):
[image]

nohal, 2024-02-22:
S63 plugin., yes, there are a few there..

Rodemfr, 2024-02-24:
I noted the issue with MeteoVache. I will investigate...
Thanks for the feedback.

closed 2024-02-25 by nohal