OpenCPN/OpenCPN #3690: manually installed plugin which is not in plugin-catalog is not shown under options->plugins

BerndCirotzki, 2024-02-24:
Installed (new)plugin that is not included in the "plugin-catalog" is not listed in the "Plugins" option menu as not activated.
It is only seen, when it was not manually activated in "opencpn.ini" file before.

Plugin installed to c:\user\MyName\appdata\local\opencpn\plugins\

[PlugIns/NewPlugin_under_construction_pi.dll]
bEnabled=0
... not show

[PlugIns/NewPlugin_under_construction_pi.dll]
bEnabled=1
.... show

This could be important, when for example developing a new plugin, or using a plugin that is not interesting for everybody.

System : Windows 10

bdbcat, 2024-02-24:
Manually adding a plugin without metadata is surely a "developer-only" use case.  Manually editing the config file may allow the plugin to load, with undefined behavior elsewhere due to lack of metadata.  
Many more higher-priority user space issues to work on.
Unsupported, wont fix.

BerndCirotzki, 2024-02-24:
Thanks for the honest answer.
I understand, it's not a bug it's a feature. Not so nice.

OK, it took me a lot of time to figure this out.
 
So, it would be nice, if you would tell me where in the code the metadata is checked, and I can switch "this featue" off.
that would at least save me than the time to find the place myself.

transmitterdan, 2024-02-24:
Bernd,

The shipdriver plugin is a one example of how to create the metadata.

[Rasbats/shipdriver_pi: Almost a simulator (github.com)](https://github.com/Rasbats/shipdriver_pi)

Regards,
Dan

leamas, 2024-02-24:
There is also some docs on how to apply the Shipdriver [here](https://opencpn-manuals.github.io/main/AlternativeWorkflow/index.html)

Would appreciate if you closed this bug, basically this is NOTABUG

nohal, 2024-02-24:
Are we really sure this is not (again) broken? I have just built the o-charts plugin locally, imported the tarball and it is nowhere (until I enable it in the config file). This is really not hoe I want it to work.

nohal, 2024-02-24:
OK, what is broken is the tarball import. The successfully imported plugin does not appear in the list until OpenCPN is restarted.

bdbcat, 2024-02-25:
Yes, I recall this discussion from times past.
I think that is recent breakage?  I seem to remember doing this successfully not too long ago.

Time for some bisect.

PR #3695 (2024-02-27): plugin_manager, plugin_loader: Fix broken imports (#3690)
For some reason the code handling the imports seems to have bitrot, cannot really understand why. Not a point fix, but not that bad.

Closes: #3690
closed 2024-02-27 by bdbcat