OpenCPN/OpenCPN #3691: Eternal loop in quilt composition

nohal, 2024-02-24:
I'm just looking at what seems like an eternal loop when composing the quilt of the IENC charts:

```
...
...
11:29:20.932 MESSAGE chartdb.cpp:1027 Oldest unlocked cache index is 1, delta t is 20
11:29:20.932 MESSAGE chartdb.cpp:246 Removing oldest chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7RH520.000
11:29:20.937 MESSAGE chartdb.cpp:1169 Creating new chart
11:29:20.937 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7RH620.000
11:29:20.937 MESSAGE s57chart.cpp:2714 S57chart::Checking SENC file: /home/nohal/.opencpn/SENC/C1A1B82C5161_1W7RH620.S57
11:29:21.008 MESSAGE chartdb.cpp:1008 Searching chart cache for oldest entry
11:29:21.008 MESSAGE chartdb.cpp:1027 Oldest unlocked cache index is 1, delta t is 20
11:29:21.008 MESSAGE chartdb.cpp:246 Removing oldest chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO080.000
11:29:21.011 MESSAGE chartdb.cpp:1169 Creating new chart
11:29:21.011 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7RH630.000
11:29:21.011 MESSAGE s57chart.cpp:2714 S57chart::Checking SENC file: /home/nohal/.opencpn/SENC/C1A1B82C5161_1W7RH630.S57
11:29:21.082 MESSAGE chartdb.cpp:246 Purging unused chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7RH530.000
11:29:21.085 MESSAGE chartdb.cpp:246 Purging unused chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7RH540.000
11:29:21.088 MESSAGE chartdb.cpp:246 Purging unused chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:29:21.091 MESSAGE chartdb.cpp:246 Purging unused chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO070.000
11:29:21.095 MESSAGE chartdb.cpp:246 Purging unused chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO050.000
11:29:21.099 MESSAGE chartdb.cpp:246 Purging unused chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7RH550.000
11:29:21.102 MESSAGE chartdb.cpp:246 Purging unused chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO040.000
11:29:21.128 MESSAGE chartdb.cpp:1169 Creating new chart
11:29:21.128 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7NE200.000
11:29:21.129 MESSAGE s57chart.cpp:2714 S57chart::Checking SENC file: /home/nohal/.opencpn/SENC/C1A1B82C5161_1W7NE200.S57
11:29:21.296 MESSAGE chartdb.cpp:1169 Creating new chart
11:29:21.296 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7NE190.000
11:29:21.297 MESSAGE s57chart.cpp:2714 S57chart::Checking SENC file: /home/nohal/.opencpn/SENC/C1A1B82C5161_1W7NE190.S57
11:29:21.380 MESSAGE chartdb.cpp:1169 Creating new chart
11:29:21.380 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7RH310.000
11:29:21.381 MESSAGE s57chart.cpp:2714 S57chart::Checking SENC file: /home/nohal/.opencpn/SENC/C1A1B82C5161_1W7RH310.S57
...
```
Following is a grep for one randomly selected cell
```
...
...
11:29:26.884 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:29:28.374 MESSAGE chartdb.cpp:246 Removing oldest chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:29:33.868 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:29:35.421 MESSAGE chartdb.cpp:246 Removing oldest chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:29:40.973 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:29:42.333 MESSAGE chartdb.cpp:246 Purging unused chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:29:48.153 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:29:49.645 MESSAGE chartdb.cpp:246 Removing oldest chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:29:55.177 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:29:56.718 MESSAGE chartdb.cpp:246 Removing oldest chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:30:02.186 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:30:03.512 MESSAGE chartdb.cpp:246 Purging unused chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:30:09.440 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:30:11.011 MESSAGE chartdb.cpp:246 Removing oldest chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:30:16.650 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:30:18.171 MESSAGE chartdb.cpp:246 Removing oldest chart from cache: /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
11:30:23.719 MESSAGE chartdb.cpp:1286 Initializing Chart /home/nohal/Documents/Navigation/Charts/ENC/DE_INLAND/ENC_ROOT/1W7MO060.000
...
...
```

bdbcat, 2024-03-27:
Any clues on how to reproduce this behavior?  Does it hang the app, or get very unresponsive?

nohal, 2024-03-27:
100% CPU. I am not sure if I somehow convinced it to stop or had to kill it, so at least very very very unresponsive.

bdbcat, 2024-04-30:
TBH, I never found a "smoking gun" for this bug.  Nor could I reproduce it.
But I did find an exposure in the quilt cache logic that could result in a loss of synchronization between quilt and chart cache.  That might be at least related to this bug. Corrected that.
Recommend closing this issue.

bdbcat, 2024-04-30:
Additional note:
IENC does stress the chart cache.  Lots of small charts, so the chart cache n_limit is reached easily when zoomed to small scale.  Also, modern "tiled" S57 as published by NOAA has similar characteristics.
Perhaps "n-limit" cache policy need rethink.

nohal, 2024-04-30:
Yes, I'm also unable to reproduce it anymore

closed 2024-04-30 by nohal