OpenCPN/OpenCPN #3694: TCP and serial N2000 does not reconnect after a Windows suspend.

Hakansv, 2024-02-25:
**Describe the bug**
TCP and serial N2000 does not reconnect after a Windows suspend.

These drivers does not reconnect after a suspend:
comm_drv_n2k_net.cpp
drv_n0183_net.cpp
comm_drv_n2k_serial.cpp

This drivers does reconnect after a suspend:
comm_drv_signalk_net.cpp
Touched by this SK patch: https://github.com/OpenCPN/OpenCPN/pull/3640/

See also log entries below

**To Reproduce**
Steps to reproduce the behavior:
1. Use TCP N0183, TCP N2000 and/or Serial N2000 connections in OCPN
2. Let Windows sleep or hibernate for a while. (I'd say > 30 minutes)
3. Let Windows wake up.
4. Connections  listed at 1. does not reconnect. 
Serial N0183 and SignalK connections are though restarted. 

**Expected behavior**
All connections would restart automatically after a Widows suspend without a need for disable-enable in Options->connections. (Or restart O)

**Desktop (please complete the following information if applicable):**
 - OS: Win10
 - Version Github master 5.9

**Additional context**
Log entries for a Win suspend:

```
09:12:45.749 MESSAGE ocpn_frame.cpp:6883 System is going to suspend.
....
10:19:25.658 MESSAGE ocpn_frame.cpp:6895 System resumed from suspend.
10:19:25.658 MESSAGE ocpn_frame.cpp:6901 Restarting streams.
....
10:19:29.114 MESSAGE comm_drv_n2k_net.cpp:382     TCP NetworkDataStream watchdog timeout: TCP:signalk.stupan.se:1455. Reconnection is disabled, waiting another 5 seconds.
>>>> SignalK does receonnect after a suspend.  <<<<<<<
10:19:29.225 MESSAGE comm_drv_signalk_net.cpp:366 Connected to Signal K server version: 2.5.0
10:19:34.059 MESSAGE comm_bridge.cpp:186    ***GPS Watchdog timeout at Lat:58.2262   Lon: 11.7625
10:19:34.209 MESSAGE comm_drv_n2k_net.cpp:382     TCP NetworkDataStream watchdog timeout: TCP:signalk.stupan.se:1455. Reconnection is disabled, waiting another 5 seconds.
10:19:34.209 MESSAGE comm_drv_n0183_net.cpp:348     TCP NetworkDataStream watchdog timeout: TCP:192.168.1.144:10120. Reconnection is disabled, waiting another 10 seconds.
10:19:34.209 MESSAGE comm_drv_n0183_net.cpp:348     TCP NetworkDataStream watchdog timeout: TCP:92.39.188.126:11101. Reconnection is disabled, waiting another 10 seconds.
10:19:39.094 MESSAGE comm_drv_n2k_net.cpp:382     TCP NetworkDataStream watchdog timeout: TCP:signalk.stupan.se:1455. Reconnection is disabled, waiting another 5 seconds.
10:19:44.132 MESSAGE comm_drv_n2k_net.cpp:382     TCP NetworkDataStream watchdog timeout: TCP:signalk.stupan.se:1455. Reconnection is disabled, waiting another 5 seconds.
10:19:44.132 MESSAGE comm_drv_n0183_net.cpp:348     TCP NetworkDataStream watchdog timeout: TCP:192.168.1.144:10120. Reconnection is disabled, waiting another 10 seconds.
10:19:44.132 MESSAGE comm_drv_n0183_net.cpp:348     TCP NetworkDataStream watchdog timeout: TCP:92.39.188.126:11101. Reconnection is disabled, waiting another 10 seconds.

Another occasion using serial N2000 is also not reconected after resume:
No suspend-log for that but this when manualy restarted.
10:16:03.560 MESSAGE comm_drv_n2k_serial.cpp:301 Closing N2K Driver Serial:COM3
```

leamas, 2024-02-25:
Yes, this is just work which is not done. Each and every driver which needs to be patched to resume orderly, even though the overall framework is in place since #3640. Not sure I can find time for this right now.

We can probably list every driver here except the signalK network and serial 0183.

leamas, 2024-03-11:
@Hakansv : @bdbcat pushed commit above supposed to fix this. Could you please test and report back or close?

bdbcat, 2024-03-14:
Latest commit untangles watchdog logic.
Now, the only use for DATA watchdog is if "NoDataReconnect" is true, an unusual configuration.
Windows Resume operation needs more extensive test.
Otherwise, I'm happy now.

Hakansv, 2024-03-14:
You, me and Windows are all happy. Thanks

closed 2024-03-14 by Hakansv