OpenCPN/OpenCPN #3704: No Sound in Flatpack 5.8.4-3

pjc1808, 2024-03-01:
**Describe the bug**
Installed Flatpack OpenCPN 5.8.4-3 on dietpi and connected bluetooth speaker to it. The bluetooth speaker are working well with vlc media player and firefox. But no sound with OpenCPN for Alarms, Bells, etc at all.
I connected headphones and have chosen the BCM2835 Headphones in the sound devices dropdown but no sounds.
Checked the play ship bells - no sound.
I crosschecked it with Watchdog plugin, but no sound.
To get the bluetooth speaker working with dietpi I installed blue-alza and added it as default (added pcm.!default bluealsa
ctl.!default bluealsa to /etc/asound.conf)
To check if sound does work I tried  aplay /usr/share/opencpn/sounds/1beep.wav but it couldn't find the file.
I tried to add a new wav file to OpenCPN and tested it upfront with aplay and it worked with aplay but not with OpenCPN. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to options choose User Interface -> Sounds -> Test several Alarms
2. Click on Test
3. See error - no sound is playing

**Expected behavior**
The sound file should be heard through the connected bluetooth speaker

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: DietPi 
 - Version 9.1.1

**Smartphone/Tablet (please complete the following information if applicable):**
 - Device: RPi 4B 4GB

leamas, 2024-03-01:
It's spelled Flatpak ;)

Before we run into the Flatpak packaging of OpenCPN:  Could you first please verify that you have a working sound setup using some other application? Just anything which plays, the Debian universe has quite a few. A low level tool like aplay is probably not the best option, try something like audacious, rhythmbox  or vcl instead.

If you indeed have a working sound setup, the next step is to check some other Flatpak application to see if the problem is related to the Flatpak sandbox in general. If you check https://flathub.org/ you will find a multitude of sound related applications, just pick any to see if it works.

If you have a working sound setup and other Flatpak applications with working sound we need to look into OpenCPN. But not before that.

pjc1808, 2024-03-01:
> It's spelled Flatpak ;)
> 
> Before we run into the Flatpak packaging of OpenCPN: Could you first please verify that you have a working sound setup using some other application? Just anything which plays, the Debian universe has quite a few. A low level tool like aplay is probably not the best option, try something like audacious, rhythmbox or vlc instead.
> 
> If you indeed have a working sound setup, the next step is to check some other Flatpak application to see if the problem is related to the Flatpak sandbox in general. If you check https://flathub.org/ you will find a multitude of sound related applications, just pick any to see if it works.
> 
> If you have a working sound setup and other Flatpak applications with working sound we need to look into OpenCPN. But not before that.

Hi leamas,
thanks for your input.
As mentioned above, the sound system is working prober with aplay in the cmd line and with VLC Media Player in the GUI. Only OpenCPN does not work right now. 
Let me try a flatpak app to verify if it is linked to the sandbox in general. Will feedback soon.

pjc1808, 2024-03-01:
I tested it with different flatpak tools like Music Gnome or Radio and all are not able to play sound. Looks like an issue between dietpi and flatpak

leamas, 2024-03-01:
Indeed. I suggest that you look for help in the Flatpak and/or the  Dietpi communities. 

I suggest that we close this issue. If you want to keep it alive, we could convert it to a discussion. In any case, this does not seems to be an OpenCPN bug.

pjc1808, 2024-03-01:
> Indeed. I suggest that you look for help in the Flatpak and/or the Dietpi communities.
> 
> I also suggest that we close this issue. If you want to keep it alive, we could convert it to a discussion. In any case, this does not seems to be an OpenCPN bug.

I close it, looks like its not related to OpenCPN - Thanks

closed 2024-03-01 by pjc1808

pjc1808, 2024-03-04:
I was able to solve the issue by the following two actions:
`apt purge bluez-alsa-utils` 
`apt install pulseaudio-module-bluetooth`

Since then it works very nice.
Again thanks for your quick response and the helpful hints.

leamas, 2024-03-04:
Thanks for reporting back, and happy that it works for you.