OpenCPN/OpenCPN #3705: WP properties dialog to small sized

Hakansv, 2024-03-01:
**Describe the bug**
The properties dialog is to small when opened. It's resizable but that size is not saved to next time. 
**To Reproduce**

1. Create a new WP
2. Right click and open properties
3. See the small size and resize it.   Close
4. Open properties again- Still the small size.

**Expected behavior**
First opening would be sized to view all info.
If resized it would be saved to next time.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Win 10
 - Version 5.9.0 from today's master

Hakansv, 2024-03-01:
I think this is about adaption to Windows scaled screen was missed lately. 
My screen scaling is 175%
On Win wo scaled screen and Debian the properties window is sized as usual.

Hakansv, 2024-03-01:
@nohal 

on commit 7689c480c6e
If I take out the added #else conditions below the size is as good as before again. For all three pages.
I don't doubt there was a reason to add it but anyhow? (Not meant for Win?)

```
@@ -869,8 +869,11 @@ void MarkInfoDlg::RecalculateSize(void) {
   SetSize(wxSize(-1, fsize.y));
 
   m_defaultClientSize = GetClientSize();
   Center();
+#else
+  wxSize dsize = GetParent()->GetClientSize();
+  SetSize(wxMin(500, dsize.x / 3), wxMin(600, dsize.y / 1.5));
 #endif
 }
```

nohal, 2024-03-01:
Sizing of this dialog was badly broken already before, at least on every machine I have here, needs to be redone taking into account the actual scaling of the monitors which we now know.

Hakansv, 2024-03-02:
So far this ugly hack solves it on my tested Win.
accept a PR?

```
@@ -869,8 +869,10 @@ void MarkInfoDlg::RecalculateSize(void) {
   SetSize(wxSize(-1, fsize.y));
 
   m_defaultClientSize = GetClientSize();
   Center();
+#elif defined(__WXMSW__)
+  //  Do nothing
 #else
   wxSize dsize = GetParent()->GetClientSize();
   SetSize(wxMin(500, dsize.x / 3), wxMin(600, dsize.y / 1.5));
 #endif
```

Hakansv, 2024-03-04:
FWIW:
After 8c166c89b we are "half" way, see pict, but still need my hack above for normal appearance.

[image]

nohal, 2024-03-04:
This commit does nothing about the sizing of the dialog, I would say your main window just was wider than last time so `wxMin(500, dsize.x / 3)` resulted in a bigger number...

commit a50936d2 (2024-03-04): Improve waypoint properties dialog initial sizing. Closes #3705
closed 2024-03-04 by nohal