OpenCPN/OpenCPN #3706: Hard while() loop in chart downloader plugin causes hang

transmitterdan, 2024-03-02:
**Describe the bug**
There is a hard loop here: https://github.com/OpenCPN/OpenCPN/blob/8fd314ae84acf7d0099f2b6e3332ff7d77e0f829/plugins/chartdldr_pi/src/chartdldr_pi.cpp#L1415

I recently noticed on Windows this loop can cause a hang when downloading a chart file which I thought was done in another thread.  I'm not sure what change created this problem but adding a wxYield() in the loop fixed it.  I'm preparing a PR to do that.  Does anyone know when this became an issue? Or am I the only one seeing it?

Also, it seems downloading is slower than it used to be. I used to max out my ISP (30mbps) but now I only get about 1/2 that rate on NOAA chart downloads.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to download any chart
2. O hangs

**Expected behavior**
O should not hang when downloading a chart.

**Desktop (please complete the following information if applicable):**
 - OS: Windows
 - Version: 11

nohal, 2024-03-02:
Sure, in my latest commit to that file a few days ago when I forgot to put the wxYield cal back after some tests...

nohal, 2024-03-02:
Can you please check how the Grib downloads work for you? I use a slightly different technique to make the event loop do it's job, which to me seems better. But my connection is slow and very high latency so the results are not representative enough.

nohal, 2024-03-02:
FWIW, I would try this

```diff
diff --git a/plugins/chartdldr_pi/src/chartdldr_pi.cpp b/plugins/chartdldr_pi/src/chartdldr_pi.cpp
index e912db03a..84ee013bb 100644
--- a/plugins/chartdldr_pi/src/chartdldr_pi.cpp
+++ b/plugins/chartdldr_pi/src/chartdldr_pi.cpp
@@ -1425,6 +1425,9 @@ After downloading the charts, please extract them to %s"),
 
       // if(g_pi && g_pi->m_dldrpanel)
       // g_pi->m_dldrpanel->Raise();
+      wxTheApp->ProcessPendingEvents();
+      wxYieldIfNeeded();
+      wxMilliSleep(10);
     }
 
     if (cancelled) {
@@ -2462,4 +2465,5 @@ void ChartDldrPanelImpl::onDLEvent(OCPN_downloadEvent &ev) {
     default:
       break;
   }
+  wxYieldIfNeeded();
 }
```

transmitterdan, 2024-03-02:
Download speed on Grib is about same as NOAA ENC (17mbps) but NOAA KAP files download at ISP limit (30mbps).

nohal, 2024-03-02:
The server providing the gribs is in Europe, so that may explain why it is slower than NOAA charts for you. You can compare the speed with direct download of eg. https://grib.bosun.io/grib?model=ecmwf&latmin=-80&latmax=80&lonmin=-80&lonmax=80&length=72 using a browser. There is initial delay as the grib data is generated server-side and nothing is transferred, then it jumps to around 500mbps when I try from another server elsewhere in Europe.

Hakansv, 2024-03-02:
patch to chartdldr_pi tested. Works!

transmitterdan, 2024-03-02:
I could not make the link work in Chrome but there may be some anti-malware at work.  I was able to download a ECMWF grib at 25mbps which is good.

transmitterdan, 2024-03-02:
I agree the patch works.  @nohal will you merge the patch?

commit 0312c1bf (2024-03-02): Fix the chart downloader plugin erroneously broken during recent development. Closes #3706
closed 2024-03-02 by nohal