OpenCPN/OpenCPN #3707: Medium-dark color scheme makes Plugin text disappear

Jon-Hacking, 2024-03-03:
If the user clicks the "Change Color Scheme" button on the left (moon & star) then the text of currently inactive plugins disappears, because the background color becomes the same as the text color.

**To Reproduce**
1. From a full-bright display, click the Change Color Scheme button once to get to the medium-dark display
2. Click Options, & go to the Plugins tab
3. Scroll down to see plugins that are not yet activated
4. Name & descriptive text of inactive plugins is no longer visible

**Expected behavior**
Plugin descriptive text should be visible

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - Updated Win11
 - OpenCPN Version 5.8.4-0

**Additional context**
Found by Harley Stoltes on the FB OpenCPN Users forum

leamas, 2024-03-05:
Situation looks the same at current master. Not a problem on Linux/GTK.

nohal, 2024-03-05:
It is a Windows only problem, Linux and macOS provide real dark mode support at the OS level and we don't need to fight with them...

leamas, 2024-03-05:
The core issue seems to be that the wxWidgets windows theme implementation is inferior to Linux/GTK. Perhaps it's about windows itself.

It would of course be possible to hardcode various font colors to use on windows. However, the overall feeling about this is not great.

An alternative is to avoid the color handling completely, using italics instead of current greyish for not enabled plugin like below.

@bdbcat  @nohal : thoughts? Good enough? 
[image]

nohal, 2024-03-05:
We need to improve `DimeControl` and friends again, our usual routine all the time...

leamas, 2024-03-05:
@nohal : Well, the first step would be toi actually apply DimeControl here, it's not used. I'm on it,  but out of time for today.

nohal, 2024-03-05:
As you wish, I've had more than enough of this fun over time...

leamas, 2024-03-11:
After applying DimeControl also to the PluginPanel. 

> Simply use the same font color for packages as for the other plugins?

The thing is that the PluginPanel uses yet another color, neither background nor foreground, for disabled plugins. To fully support this in different color schemes is IMHO just not worth it, we get the same visual feedback using italics. 

So, this is my proposal. The control checkboxes are far from ideal, seemingly unaware of the color scheme. However, for now I think we should focus on the main problem that the font as of today is unreadable in night mode. To fully support the dusk/night color schemes is perhaps not that important for the plugin management which is not part of daily sailing usage.

[image]

 @Hakansv  @nohal  thoughts?

nohal, 2024-03-11:
The GUI will always look like crap on Windows in dark modes until there is real dark mode support, which will not be the case until wx3.3+ migration, which is not on the table for O 5.10.
What can be tried in general is replacing the palette based approach with one based on decreasing the luminosity of the actual colors used by the widgets with [GetDimedColor](https://github.com/OpenCPN/OpenCPN/blob/master/gui/src/navutil.cpp)

leamas, 2024-03-11:
> What can be tried in general is replacing the palette based approach with one based on decreasing the luminosity of the actual colors used by the widgets with [GetDimedColor](https://github.com/OpenCPN/OpenCPN/blob/master/gui/src/navutil.cpp)

Indeed, I'm on that function, found at least one bug related to a static. I could of course make some such tests. 

However, the core issue here is the "extra" color used by the plugin manager to provide visual feedback about selected vs not selected plugins. Basically, the  not selected plugins uses custom background color. It is of course possible to define this for all color schemes, but again: is it worth it considering that using italics provides a visual feedback which "works" in the same way?

nohal, 2024-03-11:
That some of the elements use extra colors is not a problem when we simply reduce luminosity of it. But yes, using italics instead is OK as well.

bdbcat, 2024-03-27:
Status?

leamas, 2024-03-27:
> Status?

Forgotten. 

#3731is what I have. It does the job, but I assume there are better ways. However, plugins are not a daily task and there are IMHO limits in how much work to spend on it.

The chart bubbles part  of https://github.com/flathub/org.opencpn.OpenCPN/issues/33 looks similar to me

PR #3761 (2024-04-01): Plugin manager: Don't use custom background
Current code uses a specific, custom grey background for  disabled plugins.  This leads to #3707 since these bacgrounds does not follow the overall theme.

Use the standard colour and use an italic font instead as visual feedback for disabled plugins besides the _Enabled_ checkbox.

Closes #3707