OpenCPN/OpenCPN #3711: Frequent crash in s52plib.cpp when using OpenGL

transmitterdan, 2024-03-03:
**Describe the bug**
I have seen quite a number of crashes at this line:
https://github.com/OpenCPN/OpenCPN/blob/b1aa2d53e23a254a42517a91dbf9269a9755587a/libs/s52plib/src/s52plib.cpp#L8310

Always the same address is mentioned as unwritable: 
```Exception thrown at 0x6841E4C9 (atioglxx.dll) in opencpn.exe: 0xC0000005: Access violation writing location 0x0000C1F8.```

**To Reproduce**
Steps to reproduce the behavior:
1. Sometimes it's when ingesting (create database) a bunch of new charts.  Sometimes it is when closing down O.  Not easily repeatable.

**Expected behavior**
Don't crash.

**Screenshots**
[image]
**Desktop (please complete the following information if applicable):**
 - OS: Windoes
 - Version 11

**Additional context**
Have seen this happen in both Debug and Release builds in last 2-3 days. The crash always happens in atioglxx.dll which is called from ```int s52plib::RenderToGLAC_GLSL(ObjRazRules *rzRules, Rules *rules)```
This DLL is apparently supplied by Dell.
Stacktrace:
```
 	atioglxx.dll!6841e4c9()	Unknown
 	[Frames below may be incorrect and/or missing, no symbols loaded for atioglxx.dll]	
 	atioglxx.dll!66552be9()	Unknown
 	atioglxx.dll!66925e94()	Unknown
 	atioglxx.dll!6692598c()	Unknown
 	atioglxx.dll!66aa6ea3()	Unknown
 	opengl32.dll!_glDrawArrays@12()	Unknown
>	opencpn.exe!s52plib::RenderToGLAC_GLSL(_ObjRazRules * rzRules, _Rules * rules) Line 8310	C++
 	opencpn.exe!s52plib::RenderToGLAC(_ObjRazRules * rzRules, _Rules * rules) Line 7976	C++
 	opencpn.exe!s52plib::RenderAreaToGL(const wxGLContext & glcc, _ObjRazRules * rzRules) Line 8745	C++
 	opencpn.exe!s57chart::DoRenderOnGL(const wxGLContext & glc, const ViewPort & VPoint) Line 1745	C++
 	opencpn.exe!s57chart::DoRenderRegionViewOnGL(const wxGLContext & glc, const ViewPort & VPoint, const OCPNRegion & RectRegion, const LLRegion & Region, bool b_overlay) Line 1707	C++
 	opencpn.exe!s57chart::RenderRegionViewOnGLNoText(const wxGLContext & glc, const ViewPort & VPoint, const OCPNRegion & RectRegion, const LLRegion & Region) Line 1574	C++
 	opencpn.exe!glChartCanvas::RenderQuiltViewGL(ViewPort & vp, const OCPNRegion & rect_region) Line 3229	C++
 	opencpn.exe!glChartCanvas::RenderCharts(ocpnDC & dc, const OCPNRegion & rect_region) Line 3536	C++
 	opencpn.exe!glChartCanvas::Render() Line 4258	C++
 	opencpn.exe!glChartCanvas::OnPaint(wxPaintEvent & event) Line 1438	C++
 	wxbase32ud_vc14x.dll!wxAppConsoleBase::HandleEvent(wxEvtHandler * handler, void(wxEvtHandler::*)(wxEvent &) func, wxEvent & event) Line 661	C++
 	wxbase32ud_vc14x.dll!wxAppConsoleBase::CallEventHandler(wxEvtHandler * handler, wxEventFunctor & functor, wxEvent & event) Line 673	C++
 	wxbase32ud_vc14x.dll!wxEvtHandler::ProcessEventIfMatchesId(const wxEventTableEntryBase & entry, wxEvtHandler * handler, wxEvent & event) Line 1431	C++
 	wxbase32ud_vc14x.dll!wxEventHashTable::HandleEvent(wxEvent & event, wxEvtHandler * self) Line 1039	C++
 	wxbase32ud_vc14x.dll!wxEvtHandler::TryHereOnly(wxEvent & event) Line 1628	C++
 	wxbase32ud_vc14x.dll!wxEvtHandler::TryBeforeAndHere(wxEvent & event) Line 4007	C++
 	wxbase32ud_vc14x.dll!wxEvtHandler::ProcessEventLocally(wxEvent & event) Line 1561	C++
 	wxbase32ud_vc14x.dll!wxEvtHandler::ProcessEvent(wxEvent & event) Line 1534	C++
 	wxbase32ud_vc14x.dll!wxEvtHandler::SafelyProcessEvent(wxEvent & event) Line 1650	C++
 	wxmsw32ud_core_vc14x.dll!wxWindowBase::HandleWindowEvent(wxEvent & event) Line 1553	C++
 	wxmsw32ud_core_vc14x.dll!wxWindow::HandlePaint() Line 5373	C++
 	wxmsw32ud_core_vc14x.dll!wxWindow::MSWHandleMessage(long * result, unsigned int message, unsigned int wParam, long lParam) Line 3118	C++
 	wxmsw32ud_core_vc14x.dll!wxWindow::MSWWindowProc(unsigned int message, unsigned int wParam, long lParam) Line 3906	C++
 	wxmsw32ud_core_vc14x.dll!wxWndProc(HWND__ * hWnd, unsigned int message, unsigned int wParam, long lParam) Line 2948	C++
 	user32.dll!__InternalCallWinProc@20()	Unknown
 	user32.dll!UserCallWinProcCheckWow(struct _ACTIVATION_CONTEXT *,void *,struct HWND__ *,enum _WM_VALUE,unsigned int,long,void *,int)	Unknown
 	user32.dll!_CallWindowProcW@20()	Unknown
 	opengl32.dll!_wglWndProc@16()	Unknown
 	user32.dll!__InternalCallWinProc@20()	Unknown
 	user32.dll!UserCallWinProcCheckWow(struct _ACTIVATION_CONTEXT *,void *,struct HWND__ *,enum _WM_VALUE,unsigned int,long,void *,int)	Unknown
 	user32.dll!_DispatchClientMessage@24()	Unknown
 	user32.dll!___fnDWORD@4()	Unknown
 	ntdll.dll!_KiUserCallbackDispatcher@12()	Unknown
 	user32.dll!_DispatchMessageW@4()	Unknown
 	wxmsw32ud_core_vc14x.dll!wxGUIEventLoop::ProcessMessage(tagMSG * msg) Line 164	C++
 	wxmsw32ud_core_vc14x.dll!wxGUIEventLoop::Dispatch() Line 224	C++
 	wxbase32ud_vc14x.dll!wxEventLoopManual::ProcessEvents() Line 234	C++
 	wxbase32ud_vc14x.dll!wxEventLoopManual::DoRun() Line 288	C++
 	wxbase32ud_vc14x.dll!wxEventLoopBase::Run() Line 87	C++
 	wxbase32ud_vc14x.dll!wxAppConsoleBase::MainLoop() Line 381	C++
 	wxbase32ud_vc14x.dll!wxAppConsoleBase::OnRun() Line 303	C++
 	opencpn.exe!MyApp::OnRun() Line 941	C++
 	wxbase32ud_vc14x.dll!wxEntryReal(int & argc, wchar_t * * argv) Line 497	C++
 	wxbase32ud_vc14x.dll!wxEntry(int & argc, wchar_t * * argv) Line 175	C++
 	wxmsw32ud_core_vc14x.dll!wxEntry(HINSTANCE__ * hInstance, HINSTANCE__ * __formal, char * __formal, int nCmdShow) Line 344	C++
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 726	C++
 	opencpn.exe!invoke_main() Line 107	C++
 	opencpn.exe!__scrt_common_main_seh() Line 288	C++
 	opencpn.exe!__scrt_common_main() Line 331	C++
 	opencpn.exe!WinMainCRTStartup(void * __formal) Line 17	C++
 	kernel32.dll!@BaseThreadInitThunk@12()	Unknown
 	ntdll.dll!___RtlUserThreadStart@8()	Unknown
 	ntdll.dll!__RtlUserThreadStart@8()	Unknown
```

nohal, 2024-03-03:
The Windows backtrace is not that useful as we don't see what is happening in `atioglxx.dll`, but I would kind of suspect this might be the same as https://github.com/OpenCPN/OpenCPN/issues/2889
But I have no idea how to prove it or disable multi threaded OpenGL on Windows.

transmitterdan, 2024-03-03:
Could be.  Microsoft says this about OpenGL threading:
https://learn.microsoft.com/en-us/windows/win32/opengl/multithread-opengl-drawing-strategies

transmitterdan, 2024-03-03:
I did some testing this afternoon.  I don't think this is an OpenCPN problem.  I wrote a debugging macro to find out if any thread called glDrawArrays() whilst glDrawArrays() was already running in another thread.  It never happened.  But still I got crashes.

I looked at the atioglxx.dll file history on this computer and found that it was pretty old.  My machine is an old Dell AMD Ryzen so I downloaded the latest AMD Radeon software which updated all the AMD chipset drivers including this dll.  Now, I cannot make O crash doing all the same things I did before which could easily crash it.  So, I am going to close this with the note that probably an old version of the atioglxx.dll was the root of the problem.

closed 2024-03-03 by transmitterdan

bdbcat, 2024-03-03:
These types of errors (old GL drivers) are odd, to me.  Compared to modern game apps, OCPN is actually not very demanding of the GL API.  Although, our datasets can get pretty large...

nohal, 2024-03-03:
The issue was not us calling OpenGL from multiple threads, the "OpenGL threading" is internal to the driver. On Linux it is the new drivers that enabled this by default and we now hard-disable it ourselves no to crash.
But of course on Windows it may be completely different and realy bug in the driver...

transmitterdan, 2024-03-03:
> These types of errors (old GL drivers) are odd, to me. Compared to modern game apps, OCPN is actually not very demanding of the GL API. Although, our datasets can get pretty large...

Agreed, however in this case I found there were quite a few gamers whining in the past about this dll. That's what made me look at the dates and decide to update.  But I am sure these crashes only started in the last week (or maybe 3-4 days).  I can't remember that last time I saw an OpenGL crash.