OpenCPN/OpenCPN #3712: Consistent dialog window placement on multimonitor systems across platforms

nohal, 2024-03-03:
(At least) On macOS all the dialog windows seem to be forced to the first display even if they were previously opened on another display and the main frame is on another display.

We should have a look if something can be done about this to be consistent across platforms...

bc38, 2024-03-07:
on Windows everything looks fine
windows open on the display where they were left before CPN closed

nohal, 2024-03-10:
Does not seem like we can do anything about this macOS feature. The user is able to pin the application to any space/display in Mission Control to make it behave as one would expect, but we can't do it from code without extensive fragile hackery manipulating the OS settings. Not worth the effort AFAICT.

closed 2024-03-10 by nohal