OpenCPN/OpenCPN #3721: Voyage recording

Kitiara, 2024-03-06:
As you know the Maritime Safety Committee (MSC) has  approved a revision of resolution MSC.530 on performance standards for ECDIS introducing new ECDIS functionalities and voyage recording is one of those. There is a VDR plugin but it only records NMEA data therefore is does not comply with the standards. You can see details about the subject below (taken from MSC.530 PAGE 11)

> 11.5 Voyage recording
> 
> 11.5.1 ECDIS should store and be able to reproduce certain minimum elements required to reconstruct the navigation and verify the official database used during the previous 12 hours. The following data should be recorded at one minute intervals: 
> 
> 1. to ensure a record of own ship's past track: time, position, heading, and speed; 
> 
> 2. to ensure a record of official data used: ENC source, edition, date, cell and update history.; and 
> 
> 3. any changes in safety contour, look ahead and route monitoring alert settings. 
> 
> 11.5.2 ECDIS should output the information listed in 11.5.1.2 and 11.5.1.3 to a voyage data recorder 
> 
> 11.5.3 In addition, ECDIS should record the complete track for the entire voyage, with time marks at intervals not exceeding 4 hours. 
> 
> 11.5.4 It should not be possible to manipulate or change the recorded information. 
> 
> 11.5.5 ECDIS should have a capability to preserve the record of the previous 12 hours and the voyage track.

nohal, 2024-03-06:
OpenCPN is not an ECDIS and does not aspire to be one, so what exactly does this mean for us?

Kitiara, 2024-03-07:
> OpenCPN is not an ECDIS and does not aspire to be one, so what exactly does this mean for us?

Then why do you have a certificate about complying the EU requirements to become an Inland ECDIS ? Besides did you know that OpenCPN already meets the most of the requirements to become an ECDIS ? OpenCPN can become a real ECDIS with a little bit of efort, so why stop here ?

nohal, 2024-03-07:
OpenCPN can not become a certified ECDIS "with a llittle bit of effort". I don't see any useful feature for what OpenCPN is and does not already do in this issue, sorry.

bcn2, 2024-03-10:
Inland ECDIS does not require the VDR functionality.

From the functional side of an ECDIS  OpenCPN does lack a route checker as an important point.

The biggest stumblestone however is the homologation of an ECDIS, which includes Hardware and Software. Somebody would need to define and maintain a hardware package plus a "frozen version" of OpenCPN. And homologate that package. Expensive stuff.
If some company would like to go that way, nobody from OpenCPN would oppose. 

Btw.: OpenCPN with the S-63 plug-in and S-63 charts is listed as an IHO ECS.

leamas, 2024-04-01:
Possibly related to #3667