OpenCPN/OpenCPN #3724: Support MacOS ARM natively

hoffmang9, 2024-03-11:
**Is your feature request related to a problem? Please describe.**
I would like to have OpenCPN natively support Mac ARM instead of requiring emulation

**Describe the solution you'd like**
ARM support - but here is the twist:
I'm pretty solid with GitHub Actions and am happy to implement this as a pull request. I'd like to get some guidance about whether the project would prefer a universal binary or just have two Mac installers? Also I'd like to know whether you'd prefer I extend the existing Mac build workflow or pull this out in a separate workflow. This is all much easier now that GitHub has M1 runners. A hiding third consideration is that, if the preference is for a fat installer, that may limit the historical MacOS target and that may mean leaning to having two installers. Two installers has downstream changes to websites and wikis and...

**Describe alternatives you've considered**
Run under emulation - which is what I'm doing on my boat's M1 mini but I'm trying to avoid on my M3 laptop.

**Additional context**
Once I have some thoughts captured, I'll get to building a PR.

nohal, 2024-03-11:
For the master branch we do currently build universal binaries in CI and it is planned to be the primary macOS deliverable for OpenCPN 5.10. The limitation is that the oldest target it can support is 11.0. For that reason we also still do an Intel only build able to run on 10.13+. To maintain that, two deliverables are inevitable.
This change implies that all the 3rd party plugins also need to be built as universal binaries, which is already implemented and fortunately the product is compatible with both architectures in the universal bundle, as well as the Intel-only OpenCPN 5.8.4, at least for all the plugins already migrated to the new workflow producing the universal binaries and on my machines. I actually switched even the production builds of all my own plugins published in the master catalog to this workflow and so far have heard no complaints, whatever value that may have...

The current CI build products from https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/ seem good for me and run natively on Apple Silicon machines, but certainly all my Macs are spoiled by being the ones on which this stuff is developed, so I would really appreciate confirmation that it works as intended.

hoffmang9, 2024-03-11:
Nothing better than hearing my work has already been done. I'll grab the pre-releases, test, and report back.

nohal, 2024-03-11:
I really hope to hear an OK, writing https://github.com/OpenCPN/OpenCPN/blob/master/ci/macos-universal-deps.sh was not exactly the most rewarding job I've ever done :)

rgleason, 2024-03-12:
I don't think it would hurt to have an action script for MacOS universal that could potentially offload CircleCI a little. It could be introduced gradually. Just my 2 cents.

nohal, 2024-03-12:
> I don't think it would hurt to have an action script for MacOS universal that could potentially offload CircleCI a little. It could be introduced gradually. Just my 2 cents.

What do you mean?

rgleason, 2024-03-12:
@hoffman9  offered to build a Github Action script for MacOS Universal, but had some questions that you answered.  I think it would be worthwhile to have such a script.

nohal, 2024-03-12:
Yes, we already do have that and we [do macOS builds using GA as well](https://github.com/OpenCPN/OpenCPN/actions/workflows/MacOS.yaml), what what else do you think we should do?

rgleason, 2024-03-12:
Oh, sorry

hoffmang9, 2024-03-16:
I'm getting a "Must install Rosetta" when I attempt to install the "universal" pkg e.g. https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/detail/opencpn_590-universal8214d7feb3a13-14pkg-xtu/ and the dmg https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/detail/opencpn_590-01325d7feb3admg-cya/ . 

Am I grabbing the wrong artifacts?

nohal, 2024-03-16:
I have never seen anything like this, how exactly does that thing look? And how did you manage to have a macOS system not actually having Rosetta built in?
The artifacts you are using are correct.

hoffmang9, 2024-03-16:
.dmg:
[image]

.pkg:
[image]

I have a fresh laptop and every time I've gotten the "you need to install Rosetta" - I've gone and found the ARM native install version.

I will explore a little further over the weekend.

nohal, 2024-03-16:
```
lipo -archs /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
x86_64 arm64
```
When running `/Applications/OpenCPN.app/Contents/MacOS/OpenCPN`, `ps aux | grep -f <(fuser /usr/libexec/rosetta/runtime 2>/dev/null | xargs -n1)|grep OpenCPN` returns nothing

While when running `arch -arch x86_64 /Applications/OpenCPN.app/Contents/MacOS/OpenCPN`, it returns

```
nohal            61952  18.0  3.8 36711648 314988 s001  S+    9:33PM   0:04.74 /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
```

That is exactly what I expect from a universal binary.

nohal, 2024-03-16:
@hoffmang9 can you please try https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.9.0-universal+8238.1afecc6d0-14.pkg
It explicitly states that the package is universal in the `Distribution` manifest `hostArchitectures` option, so macOS should not insist on installing Rosetta with it.
The information comes from https://www.codejam.info/2022/04/repackage-macos-app-m1-support-without-rosetta.html, respectively https://stackoverflow.com/questions/11487596/making-macos-installer-packages-which-are-developer-id-ready/11487658#11487658

I'm not aware of how to do this with the DMG image, but as we do not distribute it and keep it just out of nostalgy, probably not worth investing much time.

hoffmang9, 2024-03-16:
Tada!
[image]

Installed and works perfectly with no rosetta now.

closed 2024-03-16 by hoffmang9