OpenCPN/OpenCPN #3725: CI: Legacy Intel macOS build not uploaded to Cloudsmith

nohal, 2024-03-11:
The known problem after Homebrew delivering too new Python resulting in PIP not working and `cloudsmith` command not being available: https://app.circleci.com/pipelines/github/OpenCPN/OpenCPN/2917/workflows/8dfbdc1f-5c16-422a-a822-f454ccf0ffb3/jobs/8160?invite=true#step-105-3110_60

leamas, 2024-03-11:
Patch which fixes it below. There are are other work arounds forcing the original install to succeed. However, recent python versions does the right thing when it refuses to mess up the system package manager be it brew, dpkg or rpm.  Using venv is the recommended practise and probably more robust.

 ```
From 51f93f8d1d6f6ce83a21cdb36fdd69340155df65 Mon Sep 17 00:00:00 2001
From: Alec Leamas <leamas.alec@nowhere.net>
Date: Mon, 11 Mar 2024 09:49:33 +0100
Subject: [PATCH] macos: Use a venv to install cloudsmith-cli (#3725)

---
 ci/appveyor-upload.sh       | 1 -
 ci/generic-upload.sh        | 3 +++
 ci/universal-build-macos.sh | 2 +-
 3 files changed, 4 insertions(+), 2 deletions(-)

diff --git a/ci/appveyor-upload.sh b/ci/appveyor-upload.sh
index b67e72809..6ce22dda1 100755
--- a/ci/appveyor-upload.sh
+++ b/ci/appveyor-upload.sh
@@ -22,4 +22,3 @@ else
 fi
 
 # TODO: Upload .lib and .pdb files as well.
-
diff --git a/ci/generic-upload.sh b/ci/generic-upload.sh
index 2d9742a1a..2110d3327 100755
--- a/ci/generic-upload.sh
+++ b/ci/generic-upload.sh
@@ -46,6 +46,9 @@ else
         $SUDO python3 -m pip  install wheel
         $SUDO python3 -m pip install cloudsmith-cli
         pyenv rehash
+    elif test -d $HOME/cs-venv; then
+       $HOME/cs-venv/bin/activate
+       python -m pip install cloudsmith-cli
     else
         # Assuming builders have installed python3 + pip
         $SUDO python3 -m pip install -q cloudsmith-cli
diff --git a/ci/universal-build-macos.sh b/ci/universal-build-macos.sh
index ca695e348..eb6e5a067 100755
--- a/ci/universal-build-macos.sh
+++ b/ci/universal-build-macos.sh
@@ -4,7 +4,7 @@ set -x
 
 # Install the stuff needed for upload to the Cloudsmith repository
 # before messing w /usr/local
-pip3 install --user  -q cloudsmith-cli
+/usr/bin/python3 -m venv $HOME/cs-venv
 
 #
 # Build the OSX artifacts
-- 
2.43.0

```

leamas, 2024-03-11:
Thou shall not do things with your left hand... the patch needs to be updated. Stay tuned.

leamas, 2024-03-11:
Script updated, now builds ok.

Note the hdi-utiil error message which is since long:
```
created: /Users/distiller/project/build/OpenCPN_5.9.0-universal+c959783f4.dmg
hdiutil: internet-enable: verb not recognized
```

Full log [here](https://app.circleci.com/pipelines/github/leamas/OpenCPN/4846/workflows/0e424581-c0f9-42e7-9e3e-1835c374fb1d/jobs/9979)

commit ec58dcd1 (2024-03-12): Update CI scripts for Python 3.12 (venv is now required to install PIP packages) Closes #3725
closed 2024-03-12 by nohal