OpenCPN/OpenCPN #3726: Bug with zoom buttons

bc38, 2024-03-11:
Appears when zoom to cursor, smooth zooming is enabled. Usually on small scales, immediately after scrolling.
[Watch the video]https://www.dropbox.com/scl/fi/81mse1i2aktjx1llcb9xl/2024-03-11-125544.mp4?rlkey=osr0n2h2fqcgnlz3k0f28903t&dl=0(url)

bdbcat, 2024-03-15:
Give us a hint.  What are we looking for?

bc38, 2024-03-15:
> Give us a hint. What are we looking for?

When you press the zoom button (-), the scale does not change and the map pans to the area where the cursor is positioned (to the area of the zoom button)
The bug appears when the settings enable: zoom to cursor, smooth zooming. For example, I scroll the map as close as possible and use the zoom button (-) to zoom out, the above-described bug occurs. Doesn't appear all the time.

Hakansv, 2024-03-15:
To compare I tried this on Windows and RPi, Debian. (v5.9 from today's master)
I enabled "Zoom to cursor"
When zooming using the mouse wheel, would be the only zoom tool relevant for this?, the zoom is related to the cursor. Correct as I sees it.
When zooming using the bottom right +- icons the cursor position is not relevant for zoom so instead the zoom center is the screen center.
When zooming using keyboard (+-/PgUpPGDwn) the zoom center is also screen center and not the cursor. Also relevant as I sees it.

Your observations would be on another system not detecting the Canvas tool?

closed 2024-03-16 by bc38