OpenCPN/OpenCPN #3729: Rpi5 Debian Bookworm X11 crash when drag the border between double canvases

Hakansv, 2024-03-14:
**Describe the bug**
O crashes when the border between double canvas is moved sideways.

**To Reproduce**
Steps to reproduce the behavior:
1. Use two canvas
2. Drag the border sideways using mouse left button down
3. O crashes
4. See gdb log below. Unfortunately no stack collected.

**Expected behavior**
No crash

**Desktop (please complete the following information if applicable):**
 - OS: Debian on RPi5 Bookworm using X11
 - Version Master code as of today.
09:57:02.071 MESSAGE ocpn_app.cpp:1147 OS Version reports as:  6.1
09:57:02.071 MESSAGE ocpn_app.cpp:1149 MemoryStatus:  mem_total: 4042 mb,  mem_initial: 156 mb
09:57:02.071 MESSAGE ocpn_app.cpp:1161 OCPN_OSDetail:  arm64 ; Debian GNU/Linux ; 12 ; debian ; 

**Note**: 
On my RPi400 Bullseye no similar trouble. The canvas border can be moved wo crash.

**Additional context**
(opencpn:386570): Gdk-WARNING **: 10:04:18.578: The program 'opencpn' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadMatch (invalid parameter attributes)'.
  (Details: serial 58577 error_code 8 request_code 151 (GLX) minor_code 11)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
[Thread 0x7fffb992aa80 (LWP 386765) exited]
[Thread 0x7fffbb7eea80 (LWP 386755) exited]
[Thread 0x7fffbbffea80 (LWP 386754) exited]
[Thread 0x7fffccf6ea80 (LWP 386750) exited]
[Thread 0x7fffddc92a80 (LWP 386749) exited]
[Thread 0x7fffc14eea80 (LWP 386748) exited]
[Thread 0x7fffde7bea80 (LWP 386675) exited]
[Thread 0x7fffdefcea80 (LWP 386657) exited]
[Thread 0x7fffe757ea80 (LWP 386656) exited]
[Thread 0x7ffff799c500 (LWP 386570) exited]
[Thread 0x7fffc0cdea80 (LWP 386753) exited]
[New process 386570]
[Inferior 1 (process 386570) exited with code 01]
(gdb) bt
No stack.
(gdb)

leamas, 2024-03-14:
You have the instructions on how to create a stacktrace in the error message above.  You will want to install a lot of debug packages.

leamas, 2024-03-14:
That is:
1. Install relevant debug packages, see link above.
2. start gdb on opencpn from the command line with GDK_SYNCHRONIZE=1 in the environment: https://stackoverflow.com/questions/55593045
3. Add a breakpoint at  gdk_x_error
4. run, trigger crash, 
5. The breakpoint should trigger
6. As usual, `bt`shows the stack.

Hakansv, 2024-03-17:
Problem solved by D. Soon in master

closed 2024-03-17 by Hakansv