OpenCPN/OpenCPN #3732: Can't enable AIS Alert / DSC Alert / SART Alert sound checkbox.

mgrouch, 2024-03-17:
Can't enable AIS Alert / DSC Alert / SART Alert sound checkbox.

OpenCPN 5.8.4, Raspberry Pi5 Bookworm, debian package from PPA on BBN OS

Somehow "Play sound on anchor alarm" checkbox selects fine.

Thanks

TwoCanPlugIn, 2024-03-18:
Did you enable the alerts in the Options->Ships->Ais Targets dialog ? This needs to be done before you can enable the specific AIS, SART & DSC alerts.
[image]

May also be related to [https://github.com/OpenCPN/OpenCPN/issues/3343](url)

mgrouch, 2024-03-18:
Looks like a bug to me then. 
Checkbox is not clickable for no apparent reason. 
How would a user know go go to some other screen to enable it?

TwoCanPlugIn, 2024-03-19:
I agree that it is poor UI design.

bdbcat, 2024-03-19:
Corrected in O5.9 master, some time ago.

bdbcat, 2024-03-20:
Please test github master O59, and close if satisfied.

TwoCanPlugIn, 2024-03-29:
While you may have fixed the problem of not disabling (greying out) the AIS, SART and MOB alarm options on the User->Sounds tab, you still have the visually unrelated Play Sounds checkbox and another "Test AIS Alert" button on the Ships->AIS Targets tab.

A user may have enabled the AIS, SART and MOB alarm, yet still not hear any sounds because they have not enabled the Play Sounds option.

I'm sorry, but this is just poor UI design. If the intent is for the user to easily mute/unmute alerts, put a mute/unmute checkbox on the Chart Options Menu. 

So at the moment, I'd prefer not to close this issue.

bdbcat, 2024-03-29:
Himmm,,,
Seems to me that a global "Play Sounds" and "test" button on AIS config page will be the first place a user goes when she wants to hear AIS alerts.  The UserInterface->Sounds page allows finer control of which audio alerts are produced, adjusting sound files, etc.
If the user should happen to go to UI->Sounds first, and enables any sound option, then the AIS->Play Sounds gets enabled automatically.

"put a mute/unmute checkbox on the Chart Options Menu " makes no sense to me.  Chart Options is about chart display, nothing to do with AIS alerts.

Sorry you don't agree with the logic.

bdbcat, 2024-03-29:
Notwithstanding the above, the idea of a global AIS sound mute button has merit.  But where/how?

Hakansv, 2024-03-29:
The "new" sound control page at User interface is to control also the sound driver if I'm correct.
Could the control for example AIS sound at Ship->AIS page and at ->User page set the same variable? Duplicates so to say. The option is true and both controls gets the same status independent of where it's set On or Off.

transmitterdan, 2024-03-29:
Just and idea, but if a user "turns on" a sound alert then would it be ok to turn off the global mute? A pop-up could alert the user that global sound mute has been turned off. Presumably, a user turning on a sound alert means they want to hear it.

rgleason, 2024-03-29:
What a pain.  What are all the instances where sound needs control?
Maybe the sound settings should be together when used so users can see the hierarchy? or is there a heirarchy?  If possible?

MenuBar > Sound > CPA alarms
Toolbar >Options > Ships >  AIS Targets > CPA/TCPA Alerts > Play sound on TCPA/CPA alerts and DSC/SART emergencies

Toolbar >Options > User > General Options >  Play ships bells
Toolbar >Options > User > General Options >  Sound device selection

Toolbar >Options > User > Sound >  Anchor Alarm > Play,  Audio filename, Select alarm sound,  Test, 
Toolbar >Options > User > Sound >  AIS Alarm > Play,  Audio filename, Select alarm sound,  Test.
Toolbar >Options > User > Sound >  Sart Alert > Play,  Audio filename, Select alarm sound,  Test.
Toolbar >Options > User > Sound >  DSC Notification > Play,  Audio filename, Select alarm sound,  Test.

Don't know where these are located?
"Play sound on anchor alarm" checkbox
AIS Alert / DSC Alert / SART Alert sound checkbox.

Are there any sound controls on popup menus etc?

TwoCanPlugIn, 2024-03-30:
> Notwithstanding the above, the idea of a global AIS sound mute button has merit. But where/how?

I meant a global mute button. Not just AIS, but all sounds.

Perhaps we should start a discussion topic about overall UI improvements for milestone 10.12 ?

Far too many examples of inconsistent location of various options, settings being greyed-out (disabled) with no obvious way to enable them, Settings of no apparent use (SignalK Server autodiscovery comes to mind), etc.

Why are chart options set on both the Chart Panel Options hamburger, (which BTW does not have a tool tip)

[image]

 and under Options->Charts ?

[image]

Sound is another example as the following three dialogs illustrate. If you honestly believe that is the epitome of good UI design, then I think you are delusional.

[image]

[image]

[image]

Here's another example, Reception of NMEA 0183 WPL sentences. 

[image]

What the hell is it doing on the AIS Settings tab ? For any user other than a geek, what is APRS ? Clearly this is a feature requested ages ago by a super geeky OpenCPN HAM radio user. Most users would expect that reception of a NMEA 0183 WPL sentence or NMEA 2000 PGN 130074 message would add a waypoint to the waypoints database.

But if the powers that be would like this issue closed, and that we have reached the pinnacle of usability, then at least I have said my mind.

rgleason, 2024-03-30:
I think this is a good idea:
"Perhaps we should start a discussion topic about overall UI improvements for milestone 10.12"

bdbcat, 2024-03-31:
I agree with some of the points above, and will be implementing some changes.
But the UI notion to keep in mind is this:
OCPN is very, very configurable, allowing use in a wide variety of platforms and system configurations.
To accommodate this flexibility, and avoid super dense configuration dialogs, we use a "layered" approach.  

1. Frequently adjusted factors are accessible with one, two, or three mouse clicks.  (e.g. ENC Text on/off, sounding on/off, waypoint properties, AIS alert sound on/off, display scaling, etc.)
2. Infrequently adjusted factors (e.g. sound file locations, ENC global display properties like Safety Depth, Chart file locations, etc) require a few more clicks.
3. Finally, the few really rarely required items generally used for initial setup (e.g. GNSS connection, plugin configuration, etc) require even more clicks.

Now we could discuss which category a particular UI action belongs in, a valid point.  I welcome that.

Believe it or not, there has been some thought put into the structure here. Hhacking on the UI based on last, loudest user comment is probably not the best approach.
So lets keep talking about improvement.

bdbcat, 2024-03-31:
(Hakan's idea) is good, Is this the way it works now?
Yes, that is the intention in O59.  You might want to experiment a bit and see if it behaves as expected.  That was the point of this (drifting) Issue.

bdbcat, 2024-03-31:
We have no fixed plan for a "Global Mute" button.  Discussion:
1.  Is it important enough to dedicate a (one click) toolbar icon for the job?
2. Should we maybe drop one or more tools from the toolbar default set?  e.g. is "Mute" of generally more interest than "Print Chart" tool?
3. AFAIK, there are only two sound sources:  AIS alerts, and "Ship's Bells".  Any others that I am forgetting in default OCPN load?

Thoughts appreciated.

transmitterdan, 2024-03-31:
Print chart is seldom/never used by most users. But how to get to it when needed? Obviously, could be a context menu item but we have tons now.

I think there are 3 sounds in the core; ship bells, AIS and anchor watch. But lots of plugins use sound. So any global mute would need to be obvious at a glance so the status bar (or maybe associate with the chart orientation icon) would be candidates. A speaker icon with a slash through it for mute and no slash for not mute. Maybe make it "marine orange" when muted? Maybe make it flash if a sound is trying to play when muted?

Not a fan of putting global mute in the toolbar.

TwoCanPlugIn, 2024-03-31:
My original thought was the same as Dan's, a speaker button mute/unmute on the status bar.

> To accommodate this flexibility, and avoid super dense configuration dialogs, we use a "layered" approach

As long as the layers are consistent and intuitive to find. We're all to familiar with the UI butput yourself in the shoes of a novice user, or a user familiar with commercial MFD's, or a user battling 30kt winds, a flogging sail and a dark entrance to a strange harbour.

As I tried to illustrate in the screen shots above, chart options are spread from arsehole to breakfast time. A stressed out user is going to be clicking everywhere for the setting they want to change.

> g. GNSS connection, plugin configuration, etc) require even more clicks.

They shouldn't. For well known USB GNS pucks, or any USB device for that matter (your beloved Actisense, Yacht Devices NMEA 2000 Gateways), users should not have to gut the internals of their OS to determine the COM Port or TTY Device. That's why God invented registries. No TwoCan user has to entet the COM Port or TTY Device, it's automagic.

bdbcat, 2024-04-01:
Please, this thread is about Sounds, specifically AIS alerts.  We are drifting. I am as guilty as anyone.
So let us start another thread discussion for general UI criticism.  Otherwise, we will never end.

bdbcat, 2024-04-01:
Moved sound device config to UI->Sounds.  
f4d68bd2a7c3d73e75ec060a89dd3b70ae997afd

Hakansv, 2024-04-01:
AIS (Alert/SART/DSC) sounds
It has been a complex task since the Users->Sounds page implementation where we can select each individually.
On Sounds page each status can be set but on Ships->AIS target page there is one button seemingly for all three.  That's kind of confusing.
Opportunities:
1. On AIS target page change the button to text referring to the Sounds page. (Since it's not xml I think a link is a trouble?)
2. On AIS target page make three buttons on one row for each of Alert/SART/DSC. Take away the test button. The "header" text could be like: "AIS sounds. Details on User->Sounds". For this we need to synchronize each button events to be reflected on both AIS target and Sounds pages simultaneously.
3. More ??

bdbcat, 2024-04-03:
Well. here is my logic.
1.  AIS sounds need to be controlled in Ships->AIS page.  ON or OFF is sufficient.
2. An easily seen TEST button is necessary.  So many things can go wrong in PC sound configuration.   A speaker cable may be uplugged/bad.  A Bluetooth external speaker may have dead batteries.  I like to get absolute confirmation that an AIS Alert situation really produces a sound, and this is what it sounds like on my hardware setup.  After all, this is an ALERT, not a music CD.
3. The UI->Sounds page is where the individual sound files are selected.
4. While pretty unlikely, it is possible to mute DSC and SART Alerts from this page, also.  
5. The check boxes on the two pages are linked.  Turning on Ships->AIS->Play sounds automatically turns ON all sounds in UI->Sounds.  Turning ON any sound in UI->Sounds automatically turns on Ships->AIS->Sounds.

In short, we should consider the UI->Sounds page to be an advanced layer for fine sound control.  Ships->AIS->Play Sounds is the primary control.

If we want to simplify even more, by removing (4) above, then we can simply remove the checkboxes from UI->Sounds.  Then, that page's only real function is to set the sound files desired for the various Alerts.  I could go along with that.

Hakansv, 2024-04-03:
I do agree on 1-5

TwoCanPlugIn, 2024-04-03:
I'll disagree with point 1.

I know of few other apps where settings on one page control those on another (with the exception of securty related options). How often have users asked the question about the "greyed out" alert sounds settings in the past ?

Here's my quick & dirty mockup. (Could've added a volume slider control adjacent to the  Audio Output Device combobox, and apologies for the lack of alignment)

[image]

bdbcat, 2024-04-04:
Thanks for the comment, and the mockup.
Looks a lot like what is actually implemented in O59, with different (tighter) layout.  I see you have added a general Mute button.
I feel pretty strongly that the most important Audio function we have is solely related to AIS Alerts.  So we will keep the master AIS related sound toggle on the AIS page.
The linkage of AIS sound options as currently implemented in O59 does not ever result in any greyed-out elements.  So, whichever page a user lands on will do a sensible thing.

TwoCanPlugIn, 2024-04-04:
OK, I now understand the logic,
If you want to set the sound for an AIS Alert, set the option here or set the option there,
but if you want to set the sound for another alert, set the option over there.

I think duplicating the UI for settings is confusing for end-users.

As @Leamas said in this [post](https://github.com/OpenCPN/OpenCPN/issues/3700#issuecomment-2035450101)

> Besides I think the general UX, well, too general

My opinion is that the general UX is amateurish. It lacks the polish and usability of commercial software. Just because this is Open Source and developed by volunteers, doesn't mean that it has to look the way it currently does. Perhaps it may also be a result of the limitations of wxWidgets.

At the end of the day it is your software. Do what you will.

Hakansv, 2024-04-04:
@bdbcat 
Agree to your above comment.
One suggestion while on the UI for AIS sound:
On Options->Ships->AIS targets.
If  "_Show CPA/TCPA Alert Dialog"_  is not checked 
do hide(grey out): "_Play Sound on CPA/TCPA Alerts and DSC/SART emergencies."_

Now it's possible to select "m_pCheck_AlertAudio" solely giving a user the impression that the sound will be played also without showing the Alert dialog. But that's not the case:
`if (!g_bAIS_CPA_Alert_Audio) m_bAIS_Audio_Alert_On = false;`

bdbcat, 2024-04-04:
I suppose it might be desired to play the audio alert without showing the popup dialog.
Is this what you are suggesting?
Or, rather, should we disable (grey-out) the audio checkbox if alerts are not enabled?
Thoughts, anyone?

Hakansv, 2024-04-04:
My suggestion was only to to disable the alert audio check box if the popup alert is not enabled since that's the actual program function now.
Change of any program function was not my intention.
The alert audio selection are present in Options->Ships-> AIS target and in the AIS Menu bar.

bdbcat, 2024-04-06:
Ready to close?

rgleason, 2024-04-06:
Does this logic need to go into the Manual?

1. AIS sounds need to be controlled in Ships->AIS page. ON or OFF is sufficient.
2. TEST to  get absolute confirmation that an AIS Alert etc situation really produces a sound
4. The UI->Sounds page is where the individual sound files are selected.
5. While pretty unlikely, it is possible to mute DSC and SART Alerts from this page, also.
6. The check boxes on the two pages are linked. Turning on Ships->AIS->Play sounds automatically turns ON all sounds in UI-> Sounds. 
7.  Turning ON any sound in UI->Sounds automatically turns on Ships->AIS->Sounds.

The UI->Sounds is an advanced layer for fine sound control and Ships->AIS->Play Sounds is the primary control.

Hakansv, 2024-04-06:
I think the "Maybe just" will do for most users.
If the logic is incorporated No 6 may need:  "Turning ON any _AIS related_ sound....." (There are more sound controls)

bdbcat, 2024-04-06:
Rick..
The info needs to be there, but paraphrased in style with the rest of the manual.
I'd rather not see my fast, short-handy, code logic notes from github issues turned into _verbatim_ man pages.  Needs a bit more thought than that.

The raw info is accurate....

rgleason, 2024-04-07:
Please see the following:
[Toolbar > Options > UI > Sounds Tab](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:user_interface:sound)

[Toolbar > Options > Ships > AIS ](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:toolbar:options:ships:ais)  at the bottom

User Manual Advanced > AIS  "Play Sound on CPA/TCPA Alert Dialog"
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_advanced:ais#play_sound_on_cpa_tcpa_alert_dialog

rgleason, 2024-04-07:
When I test this with 5.9 I get the following:
UI - all alarms turned on.   (Also Toolbar>Options>Ship>AIS "Play Sounds CPA TCPA Alerts" is ON)
Successively turning off each of these and leaving the previous ones off, does not turn off  Toolbar>Options>Ship>AIS "Play Sounds CPA TCPA Alerts" 

- Anchor
- AIS
- AIS Sart
- AIS DSC

Is that the way it is supposed to work?  So what is 
Toolbar>Options>Ship>AIS "Play Sounds CPA TCPA Alerts"   ON going to do?
This is unclear.

On the other hand,
With all of UI > Sound  items turned OFF
When Toolbar>Options>Ship>AIS "Play Sounds CPA TCPA Alerts"  is changed from OFF to ON.
Then  Toolbar>Options>UI> Sounds
- Anchor   stays off
- AIS   - turned on
- AIS Sart - turned on
- AIS DSC - turned on.

bdbcat, 2024-04-07:
Hmmm...
First part:  Not quite right, small bug.  Should do: When you turn OFF **all three** AIS related sounds in UI->Sounds, the AIS->Play Sounds should click OFF, too.
Thanks for the catch, I'll fix it.

TwoCanPlugIn, 2024-04-09:
I think we'll have to agree to disagree on the current design for sounds. 

Nonetheless, kind of related, the "Show CPA Alerts" settings (from both the Chart Context Menu and the Options->Ships->AIS Targets) are not syncronized. Only the Context Menu setting appears to alter the Status Bar text. 
[image]

Hakansv, 2024-04-10:
Not related to this issue but a short explanation:
The "CPA alarm ON/OFF" indicates the CPA warning toggle status. 
If ON the overall "Warn if CPA less than (NMI) is enabled. (and the "..TCPA is less than...)
If OFF the above setting are disabled.
Then it's up to the user to have set what kind of alert is desired by a warning.

Without "Show CPA/TCPA Alert dialog" and or "Sound..." the CPA warnings are red targets and shown CPA predictions and no alerts disturbing in a crowded area where maybe ten targets are within the range/time limits stated. That's a good overview for whom to react on.

With any alerts enabled it can fit for more open waters with no CPA targets in view and you can take a small rest.

bdbcat, 2024-04-22:
Done.
Please open another issue with specific bugs, if needed.

closed 2024-04-22 by bdbcat