OpenCPN/OpenCPN #3733: RFE:

mgrouch, 2024-03-17:
One suggestion for touchscreen operations
would be to have a crosshairs in the middle of a chart canvas and a single button in a toolbar or detect an area touch around that crosshairs to display a context menu with options to add a waypoint or lookup what the object underneath crosshairs is, or create a point for a distance measurement. So user can move the chart underneath crosshairs with the object of interest and perform actions on that object from the touchscreen.

AvNav does it.

Thanks

rgleason, 2024-03-19:
mgrouch, I know the interface that you are describing, several android nav programs use it including SailGrib, SailGribWR, Windy, Windfinder, Windfinder Pro and also Predictwind.  I think the description should be made a little clearer for understanding by the programmers.

mgrouch, 2024-03-19:
This request is actually for Linux. Not only android

rgleason, 2024-03-19:
Ok then I think you should describe it more completely, as it is likely to interfere with other features.

bc38, 2024-03-21:
It would be great if when creating a waypoint for a route or measurement, the map window panned to this point in the center of the screen. This way we can see more space around the map and continue creating a route with the map automatically moving. This is how Furuno and TimeZero do it, it’s quite convenient and functional

bc38, 2024-03-21:
https://youtu.be/Hey8I-oBLFA?si=ITMf1vYcoI5mcveO
Here is an excellent video with examples of how touch screen interaction is implemented on different chartplotters.
You can borrow and improve things from here

rogerlittin, 2024-03-22:
The AngelNav app on iOS has a good way to handle touch screen operations.  When adding an object to the chart, there's a touch handle that is offset from the object you're adding so you can move it with your finger using the handle and you can see precisely where you will be placing the object.  VNC on iOS works a similar way.  The mouse pointer is offset from the touch point by a couple of cm, so you can see the pointer clearly instead of trying to guess where it will be under your finger.

Hakansv, 2024-03-22:
roger..
That's how it works on OCPN - Android