OpenCPN/OpenCPN #3735: Debian: Sid: FTBFS

leamas, 2024-03-18:
**Describe the bug**
On Debian sid, build fails

**To Reproduce**
Steps to reproduce the behavior:
1. checkout current master at 0b28b21bc2a
2. mkdir build; cd build
3. cmake ..
4. make
5. See error: 
```
In file included from /usr/include/gtk-2.0/gdk/gdkscreen.h:32,
                 from /usr/include/gtk-2.0/gdk/gdkapplaunchcontext.h:31,
                 from /usr/include/gtk-2.0/gdk/gdk.h:32,
                 from /home/mk/src/OpenCPN/gui/src/displays.cpp:36:
/usr/include/gtk-2.0/gdk/gdktypes.h:114:39: error: conflicting declaration ‘typedef struct _GdkDrawable GdkWindow’
  114 | typedef struct _GdkDrawable           GdkWindow;
      |                                       ^~~~~~~~~
In file included from /usr/include/wx-3.2/wx/log.h:14,
                 from /home/mk/src/OpenCPN/model/include/model/logger.h:33,
                 from /home/mk/src/OpenCPN/gui/src/displays.cpp:29:
/usr/include/wx-3.2/wx/defs.h:3147:31: note: previous declaration as ‘typedef struct _GdkWindow GdkWindow’
 3147 |     typedef struct _GdkWindow GdkWindow;
      |                               ^~~~~~~~~
/home/mk/src/OpenCPN/gui/src/displays.cpp: In function ‘void EnumerateMonitors()’:
/home/mk/src/OpenCPN/gui/src/displays.cpp:128:26: error: ‘gdk_display_get_n_monitors’ was not declared in this scope; did you mean ‘gdk_display_get_pointer’?
  128 |   int gdk_num_monitors = gdk_display_get_n_monitors(gdk_display);
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                          gdk_display_get_pointer
/home/mk/src/OpenCPN/gui/src/displays.cpp:138:7: error: ‘GdkMonitor’ was not declared in this scope; did you mean ‘GdkPoint’?
  138 |       GdkMonitor* monitor = gdk_display_get_monitor(gdk_display, i);
      |       ^~~~~~~~~~
      |       GdkPoint
/home/mk/src/OpenCPN/gui/src/displays.cpp:138:19: error: ‘monitor’ was not declared in this scope
  138 |       GdkMonitor* monitor = gdk_display_get_monitor(gdk_display, i);
      |                   ^~~~~~~
/home/mk/src/OpenCPN/gui/src/displays.cpp:138:29: error: ‘gdk_display_get_monitor’ was not declared in this scope; did you mean ‘gdk_display_get_pointer’?
  138 |       GdkMonitor* monitor = gdk_display_get_monitor(gdk_display, i);
      |                             ^~~~~~~~~~~~~~~~~~~~~~~
      |                             gdk_display_get_pointer
/home/mk/src/OpenCPN/gui/src/displays.cpp:139:15: error: ‘gdk_monitor_get_scale_factor’ was not declared in this scope
  139 |       scale = gdk_monitor_get_scale_factor(monitor) * 100;
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~

``` 

**Expected behavior**
Well..

**Additional context**

This is on sid i. e. upcoming Debian 13/Trixie upgraded as of today.

nohal, 2024-03-18:
Did it switch to gtk/gdk 4? Seems like there are quite significant changes between https://docs.gtk.org/gdk3/class.Display.html and https://docs.gtk.org/gdk4/class.Display.html

leamas, 2024-03-18:
_wx/wx-3.2/defs.h_ contains:
```
#if defined(__WXGTK3__)
    typedef struct _GdkWindow GdkWindow;
    typedef struct _GdkEventSequence GdkEventSequence;
#elif defined(__WXGTK20__)
    typedef struct _GdkDrawable GdkWindow; 
    typedef struct _GdkDrawable GdkPixmap;
#else
    typedef struct _GdkWindow GdkWindow;
    typedef struct _GdkWindow GdkBitmap;
    typedef struct _GdkWindow GdkPixmap;
#endif 
```

The failing GdkWindow matches `__WXGTK20__` whereas the one for ` __WXGTK3__` matches the gtk3 realities. So, it seems like we are for some reason defining `__WXGTK20__ ` instead of `__WXGTK3__`.

But the bigger mystery is why wxWidgets duplicates these definitions instead of using what's available in _gdktypes.h_ which by definition is the correct definition

nohal, 2024-03-18:
It is different in wx master https://github.com/wxWidgets/wxWidgets/blob/master/include/wx/defs.h#L2990-L2996

leamas, 2024-03-18:
Yes. But also that should fail in the same way if we havn't defined  `__WXGTK3__`.

 `-D__WXGTK3__ ` is present on the command line. Why then does _defs.h_ seemingly does not see this.

leamas, 2024-03-18:
No, I got it wrong. Please disregard above, starting from zero.

1. _/usr/include/gtk-2.0/gdk/gdktypes.h_ defines GdkWindow as a `typedef struct _GdkDrawable`
2. _/usr/include/wx-3.2/wx/defs.h_ defines GdkWindow as a `typedef struct _GdkWindow`  when   `__WXGTK3__` is defined, which it is. This triggers  the error.
3.  _/usr/include/wx-3.2/wx/defs.h_ defines  GdkWindow as the correct `typedef struct _GdkDrawable GdkWindow` if   `__WXGTK3__` is not defined while `__WXGTK20__` is.  This is weird.

In the end, wxWidgets headers does not match the gtk ones. Sigh. Smells like a wxW bug.

leamas, 2024-03-18:
Found it, probably. The build uses _/usr/include/gtk-2.0_ instead of the correct _/usr/include/gtk-3.0_ . Question is why. I'm on it,  later.

taruti, 2024-03-19:
This also occurs on Arch Linux on yesterday's master 61375a50d6fb0a67d5a299da29e109ee8b33a41b and there too `-I/usr/include/gtk-2.0` gets picked up...

commit c1dd0486 (2024-03-19): Build: ignore gtk2 unless forced by OCPN_FORCE_GTK2 (#3735)

We probably does not use gtk2 in any meaningful build. Thus,
make the build ignore gtk2 libraries unless forced via the new
option OCPN_FORCE_GTK2.

Closes: #3735
closed 2024-03-19 by leamas