OpenCPN/OpenCPN #3736: Possible Text size improvements

rgleason, 2024-03-18:
**Is your feature request related to a problem? 
[See](https://www.cruisersforum.com/forums/f134/first-principles-for-a-new-user-284473.html#post3881527)
 https://www.cruisersforum.com/forums/f134/first-principles-for-a-new-user-284473.html#post3881527

Please describe.**
New user downloads and installs OpenCPN and screen is "grey and murky" and text sizes are way too small. Even after finding options it is difficult to know how to change text size or even to know what or which text parameters to use because the identifiers are obscure and not clear.

This is a paraphrase of the new user.

**Describe the solution you'd like**
The new user wants the program to install and load properly such that it is completely usable without requiring changes to settings that he is not familiar with and does not understand. He is unwilling to "fiddle around" and figure out the issue and he has no spare time to do this. 

**Describe alternatives you've considered**
1. Is there a simple way during initialization of the software, of doing a one time adjustment of all text sizes to utilize the screen size and pixel density (for new high res screens) to automatically set text sizes.

2. Is there a simpler and clearer way to identify the different names and types of text because these are even difficult for old users to identify. It often takes 2-3 trys to find the text type that needs changing.

3. Is there a way to have a "Reset to original settings" button that would give a new user a clean beginning?

These are a few ideas for solving a part of he current 1st principle issue that seems to be so important to some.

The manual is another issue raised. I have been working on that at various tines.

**Additional context**
Add any other context or screenshots about the feature request here.

nohal, 2024-03-18:
Can you reproduce this problem?

bdbcat, 2024-03-18:
Without a screen shot, I cannot parse "grey and murky" as a new user experience.

Hakansv, 2024-03-18:
Of course is font sizes dependent on different screem scaling. The general setting for a new user is though not a problem what I can see.
I last week gave a class for 12 new users all having installed for more or less the first time. (Eleven Windows one MacOs)
No general discussion about font sizes although I showed how to. Using AIS target name as an example. I think three changed to one step smaller.

nohal, 2024-03-18:
I would probably bet that the guy switched his system to either dusk or night mode. But that is just me.

bdbcat, 2024-03-18:
Or somehow activated "Inland ENC" mode.

nohal, 2024-03-18:
In the toolbox that he "can't find"...

rgleason, 2024-03-19:
Good suggestions we could suggest within his very tight attention span.

I had 3 suggestions for improvement which appear to be dismissed.

bdbcat, 2024-03-19:
Not dismissed.  But not fully digested, either. Time....

bdbcat, 2024-03-19:
The first suggestion to the OP, if he can be signaled, is:
$rm ./opencpn/opencpn.conf

rgleason, 2024-03-19:
Thanks, I've made your suggestion to him. We will see this works and if he is capable of uploading a screenshot.

rgleason, 2024-03-26:
I think this is outdated now.  I am assuming Linux is High Res DPI responsive.  Closing.

closed 2024-03-26 by rgleason