OpenCPN/OpenCPN #3737: Plugin Listing - "First Principles"

rgleason, 2024-03-18:
**Is your feature request related to a problem? New User gets confused by all the plugins, some that work and others that "dont work". 

**Describe the solution you'd like**
We do have many plugins, and some work better than others.

The listing order of plugins seem to change with updates, installation, etc. 

And even I dont really understand how the listing changes each time, however I spend an excessive amount of time finding the plugin in the list each time I install.

Are there some improvements that can be made to this situation?

**Describe alternatives you've considered**

1. Always have the internal pi at the top of the plugin list.
2. Always sort the rest of the plugins alpabetically.
3. Allow the user to tag a plugin to be listed directly after the internal PI.
4. If a user tags multiple PI they will be listed alphabetically.

Next - deal with the number of plugins.
Rather than having all plugins together. Put only the more mature PI in master catalog. Move all others less frequently used and less mature into the Alpha branch.

Next question is with the two catalogues, will users get confused and how interoperable is this?  Is using the alpha catalog an Advanced User skill?

There are PI that tend to be used only ocassionally, for specific purposes, or they are not fully developed. These could be moved to Alpha catalog to make the master catalog shorter and more meaningful.

**Additional context**
Add any other context or screenshots about the feature request here.

rgleason, 2024-03-19:
Possible First Step, Remove from Master Catalog?

- rtlsdr
- Deviation
- Admiralty Tide Tables Calculation
- IACfleet
- Projections
- Testplugin

And put them under the Alpha Catalog?
Any others?

nohal, 2024-03-19:
Based on what is the above list to remove created?

rgleason, 2024-03-19:
The list is too long and the order keeps changing and Plugins are difficult to find.
(I removed projections from the list because it is not the old projections but one that deals with SignalK.)

nohal, 2024-03-19:
But based on what did you specifically chose the 5 mentioned?  What is the metric?

rgleason, 2024-03-19:
(For discussion, which we are having now)
Possible First Step, Remove from Master Catalog?
    rtlsdr  - not used by many people
    Deviation  - not fully implemented and only used occassionally, does not need to be in master.
    Admiralty Tide Tables Calculation - missing data, not used by many people.
    IACfleet - IACfleet - an absolutely great plugin, but Met's don't support it with data files.
    Testplugin - TP testing plugin, not useful for much else other than testing Odraw.

And put them under the Alpha Catalog?
Any others?

I hope this gives you an idea of the "metrics" I used.

nohal, 2024-03-19:
I just tested the IACFleet plugin and both the Fiji and NOAA data seems to be available and working.
From where do we get the "not used by many people" info?

rgleason, 2024-03-19:
Previous download data.

nohal, 2024-03-19:
What does "previous download data" mean?

nohal, 2024-03-19:
I see 500+ downloads of the latest version of the Admiralty Tide Tables plugin on windows only vs. for example 33 for pypilot_pi.
Why is that not included in your list for example?
Where I head is that you could probably leave the decision about whether the plugin that is not abandoned by the developer on that developer. Remember that it is exactly two weird guys on the forum that started all this operation of yours...

rgleason, 2024-03-19:
Pavel, rather than extend this further, if you want this lets just leave it in.

nohal, 2024-03-19:
They do have a point in some of their arguments, but believe me that moving plugins between catalogs, especially  ones you don't maintain, or know, is not a solution for that.

rgleason, 2024-03-19:
OK, so this appears to be the only solution that has a chance of working.

"Simplify actually finding plugins in the very long list"

bdbcat, 2024-03-19:
We could, with some effort, group plugins to the organization found in the OCPN Website.
This is a "functional" grouping, of sorts, not perfect.

nohal, 2024-03-19:
I would probably not complicate it much, text input on the top doing dynamic filtering will do (As we have in waypoint manager) will do just fine. Same thing works fine in QGIS which has hundreds of plugins.

transmitterdan, 2024-03-19:
Here is an idea I thought about but don't know how to implement. Suppose there was a way for a user to visit a web site and chose which plugins they might like to use.  This process creates a user specific custom catalog that they maintain themselves, relieving any developer of that responsibility.  To create a custom catalog, the user visits the O plugin web site, selects the plugins they are interested in and then when they retrieve the plugin catalog within O it retrieves the custom catalog minus all the plugins that user doesn't need or care about.  They can revisit the web site as often as the like to see if there are new plugins they wish to add to their catalog or plugins they might wish to remove.

Then the list of plugins within in O is relevant and mercifully short compared to what we have today.  But with no loss of access to more or different plugins in the future.

leamas, 2024-03-20:
Another point of view is that  the handling of plugins is really two use cases: Installing new plugins and maintaining (enable/disable, update...) already installed  plugins.

As for installing new plugins, what would  the purpose be of  limiting the plugin list? Making it possible to install plugins is great, we should of course offer what we have. This could be improved in all sorts of ways:
  - Add search function(s)
  - Add possibility for users to rate the plugin and show the results in the catalog
  - Add download statistics and show also that in the catalog
  - ... and what not.

When it comes to daily life plugin maintenance, user has already marked the plugins she is interested in by installing them. To be fair it does not make much sense that all uninstalled plugins are visible when doing this, it arguably messes up things.

Low hanging fruits: Plugin search box, a "Hide uninstalled plugins" checkbox.

TwoCanPlugIn, 2024-03-21:
Even lower hanging fruit, consistently order the list alphabetically.

Nice to have, filter or group by: enabled, installed, available, all

This is the list after the install and first run.
[image]

After updating the catalog (which perhaps should be done during the initial install if internet access is available) and installing a couple of plugins.

[image]

Consistent behaviour is not just nice, it should be expected.

leamas, 2024-03-21:
> if it is a mystery to you, who use the application for years and wrote the manual, then it is broken, we do not need to describe that in the manual, we need to fix it.

Agreed.

rgleason, 2024-03-21:
Steve, in [TwoCanPlugin](https://github.com/OpenCPN/OpenCPN/issues/3737#issuecomment-2011095655) shows in the second screenshot, a plugin titled "Windows Location Sensor".   This is another area where "First Principles New User" focused...  the user difficulty getting gps working. 

I am very interested in knowing more about this plugin.

bdbcat, 2024-03-21:
Rick:
First off, a "Green Puzzle" icon means that the plugin has an error related to the declared icon.  Noting related to plugin panel sorting.
Let's try to fix those.  I'll help...

rgleason, 2024-03-21:
@bdbcat   I thought so, why would they appear after I updated the catalog? (well perhaps there were a couple before that.)  - Will use email about this, to avoid polluting thread. Sorry.

rgleason, 2024-03-21:
After getting all those green puzzle icons, I closed OpenCPN and reopened it, and all but 4 green puzzle icons were gone and replaced by Boxes in alphabetical order at the bottom of the list.   Full list shown in order of the list itself.

[image]
[image]
[image]
[image]
[image]

What would a new user have done I wonder?
Next how to fix the 4 green puzzles.

rgleason, 2024-03-21:
Next I closed OpenCPN and reopened and I find that the PI List shows many Green Puzzle Icons again. Bizarre. 
[image]
[image]

Then I closed O again and reopened, and I got all Boxes again, and no Green Puzzles.

rgleason, 2024-03-21:
I've changed the[ first post](https://github.com/OpenCPN/OpenCPN/issues/3737#issue-2193228940) to summarize the suggestions.

rgleason, 2024-03-22:
Dave has fixed the plugin listings in his branch of O5.9.
He may implement a filter or search.
Thanks.
Closing.

bdbcat, 2024-03-22:
Not really closable yet.  We did not actually do anything.
TODO:  Implement (or verify) the sorting as discussed.

1.    Disabled (Installed) - alphabetical at top of list
2.   Enabled (using) - alphabetical at upper middle of list
3.   Box - (available but not installed) - alphabetical at bottom of list.

Hakansv, 2024-03-22:
Or maybe in the order; 2. 1. 3
My reason.
When using O at a satisfying set up the likely reason to look up a plugin and fast find it may be to:
a. Check for updates for installed and enabled plugins. Collection 2.
b. Use Preferences for an enabled plugin. Collection 2.
c.  Enable a installed plugin. Collection 1.
d. Install another plugin. Collection 3.

bdbcat, 2024-03-23:
On reflection, I think I like Hakan's list order better.  (2,1,3).

a. Check for updates for installed and enabled plugins. Collection 2.
b. Use Preferences for an enabled plugin. Collection 2.

These actions would seem to be most common for most users, to me.

Other thoughts, anyone?

nohal, 2024-03-23:
Hakan's order is better. Both sorting and searching has to be case-insensitive. Otherwise not much to add.

bdbcat, 2024-03-23:
Yes, current code is case sensitive.  Very unfriendly.

closed 2024-03-23 by bdbcat

Hakansv, 2024-03-23:
"Yes, current code is case sensitive. Very unfriendly." 
Just a question: Was there any attempt to solve that? On Win a possible change was in any case not implemented?
[image]

bdbcat, 2024-03-23:
It was implemented in https://github.com/OpenCPN/OpenCPN/commit/7a38b0456be2b86cc23e80c63a3818629a7c7a6b.
Works fine for me.

nohal, 2024-03-23:
It seems the case-insensitive sorting is not working for the installed plugins (o-charts is last), it does for the list of available ones though (ais-vd in the right place).

[image]

bdbcat, 2024-03-23:
OK, I see that.  Fixing now.

nohal, 2024-03-23:
There is also one more UX problem - the icon on the right side of the tile showing the status of each plugin does not react to mouse events at all, not even expanding the tile, but it certainly makes the user think it will do something...

bdbcat, 2024-03-23:
Must be a Mac thing.  It is a rollover, with tooltip on MSW and linux.
I would entertain ideas for a better icon, something other than a misleading checkbox.
Ideas?

nohal, 2024-03-23:
It is the same on Linux and Windows, clicking on it does nothing, while clicking next to it makes the tile expand/collapse, misleading.

As for icons the one that is "weird" is the one for built-in plugins, evokes it will show preferences, but does nothing.

bdbcat, 2024-03-23:
So the simple answer is to make clicking on it expand the plugin panel.  At least there is some response, and some info is communicated to user..
On the built-in plugins:  The message to quickly communicate is:  "This is a system plugin, permanently installed and cannot be upgraded"
Anyone have ideas of icon for that message?

nohal, 2024-03-23:
Maybe just the same as for a 3rd party plugin that is up to date? Why should the user care?

bdbcat, 2024-03-23:
Hmmm.  Maybe....Probably the right answer..
Other ideas?

bdbcat, 2024-03-23:
Nohal...
Working on plugin status icon.
Are you reporting that the rollover and click action does noting on Mac?
Without any code changes, it works as expected for me,  Rollover tooltip works, clicking expands the plugin panel.

However, on linux x86, and Windows, the rollover tooltip works, but clicking does not.
Just want to be sure we are chasing the same bug.

nohal, 2024-03-23:
For me
Linux -> tooltip works, click event does not
macOS -> both work
Windows -> tooltip works, click event does not

nohal, 2024-03-24:
With https://github.com/OpenCPN/OpenCPN/commit/c907c5ac0d4c5e1a872de6c00e6e4e3e3fd161bb the behavior is OK for me everywhere.

If we want to pursue anything in this direction further, maybe the plugins that are not installed might have a "download" style icon instead of the default checkbox, like 
[image]

bdbcat, 2024-03-24:
Yes, I like it.

nohal, 2024-03-24:
OK, will do it tomorrow, if you are not faster...

bdbcat, 2024-03-24:
I'm down a rathole tonight...

bdbcat, 2024-03-25:
Nice icon Thanks

Hakansv, 2024-03-25:
Nice collection and enough to cover what's needed.

commit 7a38b045 (2024-03-23): Improve sort order on Plugin panel list.  Closes #3737