OpenCPN/OpenCPN #3739: RFE: UI-New User   ChartDownloader Wizard

rgleason, 2024-03-19:
**Is your feature request related to a problem?
Getting New Users Started easily and quickly without a manual.
Goal: Simplify chart installations.

**Describe the solution you'd like**
New Users with an empty chart folders are prompted to enter a chart area to download?
(If there are no free charts available for that area, suggestion and link to purchase o-charts OR user is prompted to select a different area that has some sample charts).

and to browse to a "user folder" location for the storage of charts.

See:
https://www.cruisersforum.com/forums/f134/first-principles-for-a-new-user-284473.html#post3881341
and  Wholybee
https://www.cruisersforum.com/forums/f134/first-principles-for-a-new-user-284473.html#post3881735
```
Wizards.
A wizard to install charts. When running the wizard, the chart directory etc. should be hidden from the user. Just create that directory someplace it makes sense for the particular OS. The wizard should at least be able to install NOAA chars  or O-Charts with zero knowledge of how charts work. The user should just draw a rectangle on the screen or choose from a list of locations .
```
and 
https://www.cruisersforum.com/forums/f134/first-principles-for-a-new-user-284473-4.html#post3880924
and from Dave ..So, let us get specific.
https://www.cruisersforum.com/forums/f134/first-principles-for-a-new-user-284473.html#post3881334
```
Regarding your suggestion: "3. Pre-build a base chart file structure. Give some automatic priority so they don't get [lost](https://www.cruisersforum.com/forums/tags/lost.html) if user loads too many chart files."

I'm serious, now. Would you be interested in proposing a specific chart file structure for new users, and moderating a discussion that may result from same? How exactly would that priority scheme work? Consider the platforms, or for simplicity, just start with Windows, some 65 % of our user base, and the most likely new user platform.
```

leamas, 2024-03-19:
Related to or duplicate of #3646.

rgleason, 2024-03-19:
Kind of a duplicate.  Will fix.