OpenCPN/OpenCPN #3741: Flatpak build on CCI failing

nohal, 2024-03-20:
Since yesterday the flatpak builds are failing

```
(appstream-compose:2): GLib-GIO-ERROR **: 15:07:55.053: No GSettings schemas are installed on the system
Error: ERROR: appstream-compose failed: Child process exited with code 133
make: *** [../flatpak/Makefile:24: build] Error 1

Exited with code exit status 2
```

Full log: https://app.circleci.com/pipelines/github/OpenCPN/OpenCPN/2942/workflows/9103902c-c4e1-4677-b9ff-5799c5dee45b/jobs/8332?invite=true#step-104-1735699_104

I don't see any code changes that should be causing it...

leamas, 2024-03-21:
https://lists.freedesktop.org/archives/flatpak/2024-March/002337.html

leamas, 2024-03-21:
https://gitlab.com/freedesktop-sdk/freedesktop-sdk/-/issues/1711

leamas, 2024-03-22:
This a an upstream issue, fixed in current 22.08 release branch (see bug above).

We could either wait until upstream makes a new version of the 22.08 sdk available or update our sdk version to 23.08.  I'm trying to avoid the latter hoping that 24.08 will be available before we release opencpn -- updating the sdk means forcing big downloads on users. 

For now, sit back and wait a little.

nohal, 2024-03-22:
The upstream fix confirmed to work

closed 2024-03-22 by nohal