OpenCPN/OpenCPN #3744: RFE: New User -GPS Difficulties

rgleason, 2024-03-21:
**Is your feature request related to a problem? Please describe.**
In the CF Post "First Principles for New User" what appears to be a major hurdle is getting their GPS working.

We have tried 7 ways to sunday to "nail this one on its head" but there are too many examples and places users go astray, when attempting this.  We have tried to fix the documentation, but perhaps it is not working.

**Describe the solution you'd like**
I have no solution to this problem at this point.

**Describe alternatives you've considered**
Deleting that section of the manual.
Having someone familiar with each OS write up an example installation. (Which is what we did essentially.)
There are common elements to each OS which should be covered.

---
Steve, in [TwoCanPlugin](https://github.com/OpenCPN/OpenCPN/issues/3737#issuecomment-2011095655) shows in the second screenshot, a plugin titled "Windows Location Sensor".   This is another area where "First Principles New User" focused...  the user difficulty getting gps working. 

I am very interested in knowing more about this plugin.

https://github.com/OpenCPN/OpenCPN/issues/3737#issuecomment-2011095655

nohal, 2024-03-21:
Enough, please, there is absolutely no actionable item in this issue and it is essentionaly a duplicate of for example https://github.com/OpenCPN/OpenCPN/issues/2967. We need to do something, not spend all our time crawling over issues like this.

TwoCanPlugIn, 2024-03-21:
@Rick,

Windows Sensor plugin uses the built-in GPS sensor available on some Windows Surface PC's. It uses the Windows Location API to obtain a position fix, generates a NMEA 0183 GGA sentences and pushes it to OpenCPN.

I haven't announced/released it, as I personally don't have a Surface laptop on which to test although at least one user has reported that it works correctly.

Tarball can be downloaded from https://dl.cloudsmith.io/public/steven-adler/windows_sensor-beta/raw/names/windows_sensor_pi-1.0.1.0-msvc-x86_64-wx32-10.0.17763-MSVC-tarball/versions/1.0.1.0+14.0ed405e/windows_sensor_pi-1.0.1.0-msvc-x86_64-wx32-10.0.17763-MSVC.tar.gz

rgleason, 2024-03-21:
I have a MS SurfacePro3.  I normally use a GPS puck. Not sure if it has built in GPS but wil check and if it does, I'll try it.

closed 2024-03-21 by rgleason

TwoCanPlugIn, 2024-03-22:
I don't think your model has a built-in GPS chip.

The models that do, have a SIM card slot, and the models are designated "LTE". Eg. Microsoft Surface Pro 9 with 5G LTE

rgleason, 2024-03-22:
Yes, I think that is correct.

Steve, due to some recent eruptions I have added several paragraphs to the GPS section
if you @twocanplugin and @Hakansv  could take a look and advise or edit, I would appreciate.

I am making a clear distinction between getting the Manuf GNNS working in the computer and finding the connection information  with entering this information into OpenCPN Connections.

https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:gps#manufacturer_gps_gnns_devices_and_software

https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:get_started:gps#opencpn_gnns_setup

(I am a little pressed for time right now but will revisit this in several days.)

rgleason, 2024-03-22:
@nohal   Please remove your comment above. Thank you.
We are continuing to work on this problem.
@bdbcat I've tried to make the User completely responsible for finding the necessary connection settings. See the links above.

Then it is a simple matter of entering the correct ones into Toolbar > Options > Connection. 

I have no doubt made some other blunder, so hack away at the documentation if necessary. 

The reason for this change is to **set User Expectations**! 
So we don't have people yelling at us.

TwoCanPlugIn, 2024-03-23:
@Rick,
When time permits, I'll have a look and make suggestions/edits where possible.

However it looks very wordy and I believe in the adage "less is more".

As an aside, the manual very much follows the structure of OpenCPN's menu options, rather than themes. Eg. Getting Started, Installing Charts, Installing hardware (GPS, NMEA0183, NMEA2000), AIS. Radar, Routes. Waypoints. etc.Perhaps look at a commercial vendor's documentation. Eg. B&G Vulcan.

Suggest we take this offline, or create a GitHub Discussion topic.