OpenCPN/OpenCPN #3745: Crash upon startup (macOS)

macluky, 2024-03-21:
**Describe the bug**
Starting the app crashes it.

**To Reproduce**
Steps to reproduce the behavior:
1. remove preferences from ~/Library/preferences/opencpn
2. start app

**Expected behavior**
well....

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: macos 14.3.1

**Smartphone/Tablet (please complete the following information if applicable):**
 - macbook m2
 -
**Additional context**
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OpenCPN [63437]
Path:                  /Users/USER/*/OpenCPN.app/Contents/MacOS/OpenCPN
Identifier:            org.opencpn
Version:               5.2.4+6b314e6 ()
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-03-21 20:29:09.2810 +0100
OS Version:            macOS 14.3.1 (23D60)
Report Version:        12
Anonymous UUID:        547CD083-E7AA-6CBB-B257-A906C6A73C86

Sleep/Wake UUID:       25B6D2C6-6C35-4E26-94C5-1A3C78E4F9E2

Time Awake Since Boot: 680000 seconds
Time Since Wake:       1879 seconds

System Integrity Protection: enabled

Notes:
PC register does not match crashing frame (0x0 vs 0x10D2612A2)

Crashed Thread:        2  Dispatch queue: CA DispatchGroup

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000008
Exception Codes:       0x0000000000000001, 0x0000000000000008

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [63437]

VM Region Info: 0x8 is not in any region.  Bytes before following region: 140723155402744
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      mapped file              7ffca9b08000-7ffcd28b8000 [653.7M] r-x/r-x SM=COW  ...t_id=1c0d2c0d

Error Formulating Crash Report:
PC register does not match crashing frame (0x0 vs 0x10D2612A2)

Thread 0::  Dispatch queue: com.apple.main-thread
0   ???                           	    0x7ff8a4666a78 ???
1   libsystem_kernel.dylib        	    0x7ff8140b25ce __psynch_cvwait + 10
2   libsystem_pthread.dylib       	    0x7ff8140ef76b _pthread_cond_wait + 1211
[... 581 lines cut ...]
  "experiments" : [
    {
      "treatmentId" : "a3f9ec09-b145-49f3-8142-da4a1f338456",
      "experimentId" : "65f21aa774b6f500a45abd7d",
      "deploymentId" : 400000001
    }
  ]
},
  "reportNotes" : [
  "PC register does not match crashing frame (0x0 vs 0x10D2612A2)"
]
}

Model: Mac14,2, BootROM 10151.81.1, proc 8:4:4 processors, 16 GB, SMC 
Graphics: Apple M2, Apple M2, Built-In
Display: Color LCD, 2560 x 1664 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Sep  1 2023 19:46:07 version 20.10.1073.3.8.7.165 FWID 01-b66346c5
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: Laptop, Apple Inc.
Thunderbolt Bus: Laptop, Apple Inc.

nohal, 2024-03-21:
5.2.4 is very ancient. Can you try with https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.9.0-universal+8377.bab4554c6-14.pkg instead (Or at least with the current 5.8.4 from opencpn.org)?

macluky, 2024-03-21:
That is interesting, i did download the latest version from: https://opencpn.org/OpenCPN/info/downloads.html 
Let me try again with the link you provided

nohal, 2024-03-21:
The location where it is installed - `/Users/USER//OpenCPN.app/` is also pretty weird, maybe you have some forgotten old stuff around?

macluky, 2024-03-21:
I'm officially an idiot that managed to install one version in /Applications and an ancient one in ~/username/Applications. The new one works perfectly and thanks for the fast support, you guys/girls rock!

nohal, 2024-03-21:
:) Happens...

closed 2024-03-21 by nohal