OpenCPN/OpenCPN #3747: Language change is not possible on Linux Debian - RPi

Hakansv, 2024-03-23:
**Describe the bug**
A change of language does not work and falls back to English (US)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Options->User->Language
2. Scroll down and select desired language, in my case "Svenska" 
3. Click Ok. 
4. Still on English (US). Also after a O restart.

**Expected behavior**
Language changed after Options->Ok
**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Linux Debian (Rpi)  -  Works on Windows.
 - Version: Current Git master

**Additional context**
Log file after attempt to change language.
```
09:00:09.682 MESSAGE OCPNPlatform.cpp:1089 ChangeLocale: Language load for:  sv
09:00:09.706 WARNING intl.cpp:393 Cannot set locale to language "Swedish".
09:00:09.736 MESSAGE OCPNPlatform.cpp:1114 ChangeLocale:  could not initialize:  sv
09:00:09.737 WARNING intl.cpp:393 Cannot set locale to language "Swedish".
09:00:09.737 MESSAGE OCPNPlatform.cpp:1130 ChangeLocale: Fall back to en_US
```

Hakansv, 2024-03-23:
Observations:
For OCPN core in Crowdin we use %two_letters_code% 
But the function is using %locale_with_underscore% (As for plugins)
The function is trying to change to: "Language load for:  sv_SE" (Instead of "sv")
gdb locals from OCPNPlatform.cpp.
```
1107	    locale->Init(pli->Language, 1);
(gdb) info locals
return_val = {static npos = 18446744073709551615, m_impl = L"", m_convertedToChar = {m_str = 0x0, m_len = 549755805752}}
imsg = {static npos = 18446744073709551615, m_impl = L"ChangeLocale: Language load for:  sv_SE", m_convertedToChar = {m_str = 0x0, m_len = 366515011768}}
__FUNCTION__ = "ChangeLocale"
```

Installed .mo folders and files.
There's a strange mix of %two_letters_code% and %locale_with_underscore%?
```
/usr/local/share/locale/cs/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/da/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/de/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/es/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/fr/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/it/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/nl/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/pl/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/ru/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/sv/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/et/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/pt_PT/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/pt_BR/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/nb_NO/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/tr_TR/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/fi_FI/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/el_GR/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/zh_TW/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/zh_CN/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/hu_HU/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/gl_ES/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/ca_ES/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/ar_SA/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/ja_JP/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/vi_VN/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/he_IL/LC_MESSAGES/opencpn.mo
/usr/local/share/locale/en_GB/LC_MESSAGES/opencpn.mo
```

Hakansv, 2024-03-23:
Of course it's confusing OCPN use %two_letters_code% but plugins %locale_with_underscore% . Likely leading to mismatches here there. But that's how  it is for now and it has worked before.

Hakansv, 2024-03-23:
More observations:
On Windows, where the language change is working, both (%two_letters_code% and %locale_with_underscore% ) folders are present for all languages but obviously not on Linux where some are the first and some the latter?

bdbcat, 2024-03-26:
**THIS SHOULD BE IN USER MANUAL**

Changing UI languages supported on linux requires some  specific actions.

You will need to "generate" the desired locale for your system.

1.  List currently available locales
$ locale -a

2.  See list of locales loaded, not yet generated
$ cat /etc/locale.gen

3.  Edit /etc/locale.gen, un-commenting the language that you want to add

4.  Generate desired locale
$ sudo locale-gen de_DE.UTF-8

5.  Change the desired OpenCPN language in Options->UserInterface->Language.

Hakansv, 2024-03-26:
So, it was the "old" issue that language can't be change to others than your system local and Eng(US). And I happened to have English only on my Rpi. (That's for easier discussion on our forum and more)
It was like that in Windows before but obviously still something similar on Linux. 
When having both SE and EN installed I could change also on the RPi.
Thanks

closed 2024-03-26 by Hakansv