OpenCPN/OpenCPN #3748: OpenGL is no longer supported on my PC

bc38, 2024-03-23:
After the latest changes with OpenGL made by Nohal, CPN when launched gives the error "procedure entry point not found"
CPN starts with OpenGL disabled and not active

I tried to enable OGL in the config, it didn’t help

Windows 11
Video nvidia RTX 4070

[image]
[image]

nohal, 2024-03-23:
Can't reproduce, please do me a favor:
In the Windows Explorer navigate to `C:\Program Files (x86)\OpenCPN` and double click `opencpn-glutil.exe` - Do you get the same error dialog?

Hakansv, 2024-03-23:
@nohal 
For a test I did it as well.
In "C:\Program Files (x86)\OpenCPN\bin\  I ran opencpn-glutil.exe
I collection of  errors occurred:  The first: (Translate:) "The code couldn't run since it can't find": 
But the file searched for is actually present one folder back, in /OpenCPN/
[image]

nohal, 2024-03-23:
Yes, that is why I asked to run the copy in the main directory.

Hakansv, 2024-03-23:
Hmm, yes of course - sorry.
When running from "C:\Program Files (x86)\OpenCPN" no error and no message. Just quietly run it successfully.

nohal, 2024-03-23:
Yes, that is the case on all the Windows machines I have access to as well...

bc38, 2024-03-23:
> Can't reproduce, please do me a favor: In the Windows Explorer navigate to `C:\Program Files (x86)\OpenCPN` and double click `opencpn-glutil.exe` - Do you get the same error dialog?

launched no errors
but strangely no window even appears

nohal, 2024-03-23:
That is what is supposed to happen.

bc38, 2024-03-23:
and if you run the file in the bin folder, you get exactly the same error. with the second file opencpn-cmd.exe the same error

nohal, 2024-03-23:
> and if you run the file in the bin folder, you get exactly the same error. with the second file opencpn-cmd.exe the same error

Exactly the same error as you show above or as Hakan shows?

bc38, 2024-03-23:
error from my screenshot

nohal, 2024-03-23:
Then I am almost sure the problem is you have an ABI incompatible copy of the wxWidgets DLLs somewhere in the folders Windows searches for libraries. Anyway, we need to be able to live with that, working on it.

bc38, 2024-03-23:
found a copy of the library in the windows folder
deleted the error disappeared and GL started working
Thanks everyone for your help
I'm closing

commit d18abd77 (2024-03-23): Harmonize installation location of command line utilities with the main application and DLL libraries used at runtime on Windows. Fixes #3749. Fixes #3748.
closed 2024-03-23 by bc38