OpenCPN/OpenCPN #3752: Options | Connections : wrong ordering

leamas, 2024-03-25:
**Describe the bug**
In the Options | Connections tab, the connections are presented in reversed order when doing  manual ordering

**To Reproduce**
Steps to reproduce the behavior:
1. Go to  Options | Connections
2. Create three connections giving them different _List Position_ 1, 2 and 3.
3. See connections presented in order 3, 2, 1

**Expected behavior**
Well...

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
Tested on Linux/Flatpak, no reason to believe this differs on other platforms.

**Additional context**
No big deal.

leamas, 2024-03-25:
At a closer look: Why clutter the display with the _List Position_ column? The position is obvious from the listing as such, the column is redundant and should be removed. Of course it should still be available when editing the connection, but that's something else.

And the ordering should be 1, 2, 3, ...

Hakansv, 2024-03-25:
Since the "list position" is the former priority control *) where highest number also was given highest priority the same logic was transformed when used as the list position instead.
This is of course a matter of owns mind. One mind could be: "Since 3 is higher than 1 put it as high as possible".
I find the list control very useful. When there are several connections I like to sort the list so the one often dealing with should be as high (up most) as possible.
If "3" is higher than "1" or "1" is the first (up most) and "3" would be below does not matter much. I haven't thought of it since I'm used to the former priority logic.

*) The former priority logic was: If e.g. N0183 sentence xxRMC is received from two or more connections use the one with the highest priority number" (The logic could also have been to use "1" as the first and "2" as next, but it wasn't) :)

leamas, 2024-03-25:
Perhaps this is then just a misleading label. "List Position" is hard interpret in any other way than 1 being first, this is hardly personal.

If the label read "Priority" instead I could follow your reasoning.

Hakansv, 2024-03-25:
No I don't think we should change back to "priority". It has nothing to do with connection priority anymore and could be misunderstood.
But if you think the UI list order 3, 2, 1 is more intuitive for our users I've no personal doubts with that. My habits are changeable.

nohal, 2024-03-25:
If it is not priority anymore, what about removing it completely? Is the list order as important that it warrants the confusion?

Hakansv, 2024-03-25:
Compromise?
Keep the list order control. Make the order as you like, 1 2  3 or 3 2 1.
Take away the header column as of leamas suggestion?

nohal, 2024-03-25:
Just out of curiosity, how many connections do you have configured? I usually have between 5 and 10 for all the different testing data sources I use and never actually feel the need to set the order of them...

nohal, 2024-03-25:
On production boats I usually don't see more than 3 or 4 (1 or 2 data feeds over IP from a plotter and/or a multiplexer and 1 or 2 for USB pucks kept as backup)

Hakansv, 2024-03-25:
I use to have 5 to 10 here a home as you. About 5 on board.
But if this "list position" feature is troubled I can live without it. No big issue.

rgleason, 2024-03-25:
How big a chance is there that users will get confused about "List Position"  and "Priority"?
I would make the determination to keep it or not, based upon the outcome of that discussion.

bdbcat, 2024-03-25:
Enough.
Actions:
1.  Remove "List Position" from plugin list panel.  No information there.
2. Invert the list position sense in the layout logic, to conform to human users instead of coders who think "0" is somehow higher than "1" :)

I got this.

leamas, 2024-03-25:
While on it, it might make sense to make the default position to 99 or so. The idea is that entries which havn't got a _List Position_  configured should be displayed after those actually configured.

commit 90619808 (2024-03-26): Refine connections panel list sorting and edit dialogs  Closes #3752
closed 2024-03-26 by bdbcat