OpenCPN/OpenCPN #3754: Grib plugin painting the area selection on wrong place in multicanvas mode.

BerndCirotzki, 2024-03-25:
@nohal
when "O" is in mulitcanvas mode, by define a grib area "by Hand" with selecting the area on canvas, the selection is ... somewhere on the second canvas shown, but not there where the mouse is ! In the one canvas mode it works without problem.

(One more annotation not to do with this here ... the weatherrouting plugin is showin an notice, that the plugin is not working with the grib plugin (4.2 , 4.1) version ... but it works. comunication between them ?!?)

nohal, 2024-03-25:
Weather routing plugin must be updated to accept the increased version of Grib plugin as compatible, I suppose it is a one line change there.

BerndCirotzki, 2024-03-25:
> BTW, have you tested the direct grib download functionality added in the current master? Any feedback on that?

I have tested the direct download. but i used saildocs for test. sometimes you are somewhere without internet .. on anchorage then I can use winlink eMail.

nohal, 2024-03-25:
Sure, the Saildocs request is not going anywhere, it has it's well justified use case.

BerndCirotzki, 2024-03-25:
OK, but what does that have to do with the overlay painting not working?

nohal, 2024-03-25:
Nothing, this is simple bug that has to be be fixed.

commit 066affbd (2024-03-27): Fix grib area manual selection with multicanvas. Closes #3754
closed 2024-03-27 by bdbcat